FormerLurker/Octolapse #568: Add information about Marlin v2 board to help for printer profiles

gmhowell, 2020-07-12:
#### If this is a feature request describe it here

There should be a line of text or otherwise a more prominent display concerning Marlin 2 boards when you use a printer that is likely to have one. 

I have an Ender 3 Pro running an SKR v2 with Marlin v2. I selected the Ender 3 Pro profile. My first octolapse came out great, but the print had zits and stringing. I looked into it, and in the troubleshooting, it mentions changing 'G90/G91 Influences Extruder' setting to eliminate this when running Marlin v2. Putting that text or suggestion into the description for the Ender 3 Pro (and, presumably, the Ender 3, CR10, etc) could stave off some complaints/support requests.

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

FormerLurker, 2020-07-13:
I have been thinking about adding a firmware and firmware version drop down.  I need to overhaul the profile repository first, but it is something I want to do.

I may also add some autodetection for that setting.  I have some test scripts, but haven't perfected it yet.

gmhowell, 2020-07-13:
Another option could simply be to call the existing profiles 'stock' and have additional listings for firmwares/situations that are a different printer in most meaningful ways (this being one of them).