FormerLurker/Octolapse #569: DSLR fails to take 14 images at from roughly layers 630 to 644 goes for a while longer then starts having issues again. Please make sure there is sufficient quoting around the arguments.

WatchOut1233, 2020-07-20:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Running a debug test print(but this issue has shown up on my last 2 prints at the same exact number of photos(well very close)
2. using camera debug until the issue happened then changed it to full logging
3. basically the issue I am having is that everything works like a charm until i get to the 630 area of photos then i get failed snap shots for about 15 then it continues on. i do start having the issue again in the 700 range . I have exhausted every resource to solve this issue. I will say that originally I was using a 16 gb sd card in my (raspberry pi 4 8 gb) it wasnt enough so I cloned it and flashed it to a 64 gb which I then expanded the root. but I have verified that it is expanded and the original issue i had was failing photos and no recovery then wouldnt render. but this was a space issue. now that its corrected im at a loss for words. Everything renders fine but hangs up as i described. I originally thought it was a ending of the print issue due to it happening to prints i was running that were roughly 650 layers but after my testing it has nothing o do with the end of print and for some odd reason everything to do with a specific file name which starts to happen around 630 snap shots.

#### What should have happened?
The photos should have worked as well as the first 629

#### What happened instead?
at 630 photos in  (env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --auto-detect --capture-image-and-download --filename /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/e897fa99-83f6-49b8-963e-eaa602873f5b/053af367-0bf8-4f60-9c98-1764f7d6406c/timelapsetest000783.jpg
	Please make sure there is sufficient quoting around the arguments.) The amazing thing is it also recovers 15 images later and goes for a bit longer then has issues in the 700's 

#### Operating System running OctoPrint and Octolapse

OS Name: raspbian
Os Version: 10

#### Printer model & used firmware incl. version

Printer Model: cr-10
Printer Firmware Version: I dont actually know(what ever came stock)

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Version 84.0.4147.89 (Official Build) (64-bit)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/WatchOut1233/70f1768736722289c59c5ccf37c32b05

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/WatchOut1233/e296004808fd92d0c2aad93397711af1

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/WatchOut1233/039ee333455e967e28436725c9b57573

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/WatchOut1233/2b9c78a432d9a3500c38f4b88e3d026d

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

WatchOut1233, 2020-07-20:
Hey just wanted to clarify my description a bit. normally i dont ask for help and solve issues my self. I have overcome so many barriers to get the perfect time lapse of outstanding resolution, and I feel im there except for this issue.

basically i am running a canon eos I7(2000d) running the script from your set up and everything was slowly tested and set up from your step by step. after solving some small space issues I had I started printing many files. these files are roughly the same size in height and end between 600 and 700 layers. this is do to the area I print in with the dslr camera and achieving the best depth of field i can for the best results i can get. so i can say that this issue happens on every print in the same area.

630 layers in the camera stops taking snapshots or at least got stuck on the previous snap shot. This kinda recovers it self after 15 layers. this is the main difference I see in the failing snapshots according to the log.

env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --auto-detect --capture-image-and-download --filename /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/e897fa99-83f6-49b8-963e-eaa602873f5b/053af367-0bf8-4f60-9c98-1764f7d6406c/timelapsetest000637.jpg
Please make sure there is sufficient quoting around the arguments

sadly i see no difference in the filename that would make me believe I can fix this issue.
I need help so badly on this one lol

P.S srry for spelling mistakes etc I am tired and spent countless nights for the past 2 weeks solving a plethora of issues for what I am trying to do.

WatchOut1233, 2020-07-21:
Update- Done some more testing and digging and now starting to think its a gphoto issue, since the fault given is I/O in progress. 
although I did find a thread open for an issue that seems to be the same. I noticed the creator of gphoto2 is not as involved as you are when It comes to helping people with these difficult issues. Or at least this is my opinion based on how long the thread was opened with more people running into the issue and it was never closed or resolved in multiple years.

Having said that could you give me a hand in writing a script that will run "gphoto2 --reset" no quotes  if it fails to take a snap shot then retry a couple seconds later. I can always tinker with he delay between the reset. just I do not fully understand the syntax and you do lol. having said that I will try and figure it out and test it to see if it works. But I am ok with a bandaid at this point if gphoto2 is just gona go awall after 600 + photos.

WatchOut1233, 2020-07-21:
Ok I have no clue I added the reset code to the script and just let it reset every image and it still failed in the exact same spot. I am lost :( guess I will be printing only 4 inches tall until another solution presents it self. debugging has forced me to trigger my shutter nearly 7000 times which is 7 percent of my shutters life just testing. got a new printer coming in a couple days probably give it one more go on that and if by then no solution presents it self then ill have to try a mechanical switch.

WatchOut1233, 2020-07-22:
Update-- I took the easy way out and replaced my canon eos t7 with the much better nikon d3500. First stress test went past 700 photos with no hangups except slight delays in the transfer of such large photos. You can close this out if you would like. My final conclusion if this can be used to help someone is.... gphoto2 has some bugs when it comes to using some of the newer canon eos cameras. This bug is odd to say the least and I cannot say if it would happen if i was finger pressing the trigger 630 times or not. But gphoto2 fails to surpass 630 photos without failure of capture. Its not fair for me to blame gphoto2 but this issue has been open for 2 years since the release date of the canon eos cameras that I speak of and no real end in sight for a fix. My recommendations for anyone buying a camera to use with gphoto2 is to stay away from the eos models from canon, this is not a headache you want to endure trust me. Even though I took the easy way out I hope that my public break down of my own issue can help someone. maybe former lurker will close this as a solved issue so people can view and have a area to find this specific issue. I know I received no reply for this topic as its only been 24 hours, but I would like to thank you "former Lurker" for your spectacular software and all you do to help people with there issues. You are a true gem in community's like this and keep it up. I do plan on donating when my current covid money situation changes for the better. I am going to start posting time lapse volumes within a week, and if you would like to watch them. post your email and I will send you the first release. Have a good one and keep lapsing everyone.

FormerLurker, 2020-07-22:
I would like to see your videos @WatchOut1233!  Also, I've noticed the issues with newer eos cameras.  This is sort of expected since there is no genuine standard for communications between these cameras and an external device.  It's possible that there is some problem with the firmware that prevents gphoto2 from working perfectly.

I hope your money situation improves.  Don't worry about sending any my way.  These are very difficult times after all.

Post links to your timelapse here please and I'll take a look, and enjoy the software!

closed 2020-07-22 by FormerLurker

WatchOut1233, 2020-07-23:
Question- You ever had the render of the file cause playback problems with large resolution photos. had a lapse where it took all the photos but the render process ruined it almost unplayable.

FormerLurker, 2020-07-23:
Yes.  It depends which format you are rendering.  For example, H264 videos have a max resolution of 4096 × 2048.  H265 can go higher, but unfortunately my PI doesn't have enough memory to render high res videos of any substantial size in either 264 or 265 format.

If you are using the standard MP4 rendering, it will depend mostly on your video player.  Some can't handle anything above 1080p, and some can't play videos above 4k.  I find that VLC can play most of these videos.  I sometimes use handbrake to convert these videos to something more player friendly, or to down-sample them.  If you don't have handbrake, I recommend you download it since it works pretty well and is free.

WatchOut1233, 2020-07-23:
interesting I am using vlc player. old cam vs new cam both 6k res. old cam worked fine rendering new cam came out like a mess and i can definitely tell its the rendering or the player. excluding the player since it worked on the old cam. only difference is i found the capture quality on my new cam and set it to jpeg fine. which is one step under nef. now i never set that in the cannon so i do not know what default file was saved as. hmm maybe ill youtube upload as unlisted see if the playback changes since they do 4 k. If I un tick timelapse generation will it just have a zip for me to download with all the images?

WatchOut1233, 2020-07-23:
ill look into hand brake sounds promising, just want to hash out some of the basics since im using a new came and ift has way more auto features then the old one so im trying to get back to the quality i once had haha. depth of field is a hard one to have considering there are many trade offs as you increase ur depth of field and much worst trade offs when you try and make a blurry all white b background shadows dont look good blury haha you know what im gona send u something its a old video and its just a test, but maybe u can see the issues i speak of when it comes to the background haze and shadow motion i speak of.

https://www.youtube.com/watch?v=IUEYFW-Jcpk

check that out its old and iv overcome most my issues printer and video wise but maybe u see something i dont

FormerLurker, 2020-07-23:
OK, without seeing the video earlier I couldn't really tell what you meant.  It looks like you need to increase the bitrate in the rendering profile.  If you ever see blocky artifacts, that's most likely the culprit.  If you have enough space available on your PI, I also recommend that you turn on snapshot archiving here:

[image]

Ticking this option will cause Octolapse to zip your images up after rendering so they will show up as an archive you can download or re-render with different settings here:

[image]

[image]

I don't have any archives since I am doing some testing with various OctoPrint/Python versions ATM, but there is a download, delete and render option here.  If you're messing with rendering settings this a great place to experiment since you can render over and over again as many times as you like, or you can download the images and import into any video editor you want for timelapse generation.  I hope that helps!

WatchOut1233, 2020-07-23:
https://youtu.be/ZA_kGNa_pSE    wow it is the player the standard definition upload of 360 p (since it takes an hour for youtube to do the hd version is very smooth. so its not rendering vlc player is giving me issues.  anyways thats the print at question. i like the lapse it self just the shadows throwing me off in the back

WatchOut1233, 2020-07-23:
you think I need to have a bit rate higher then 25M (thats what I am using) for my 24.2 mp camera

WatchOut1233, 2020-07-23:
btw ty thats awesome I didnt know i could render and save snap shots then obviously delete them after i got them all . I got 64 gig i think I should be good space wise.

WatchOut1233, 2020-07-23:
I see some weird plashing in the newest hulk video that wasn't there with my old cam. weird every auto function is off and im using 60 power line freq. guess i got some work to do today lol

FormerLurker, 2020-07-23:
Yes, you probably need a higher bitrate.  Youtube recommends 53-68 Mbps for 4K video (3840 x 2160), and that's for H264 video.  You are recording somewhere around 6,000 x 4000, 3x as many pixels as 4k!  Also 64 GB won't last long if you are using high quality JPGs at that resolution, so be VERY careful with disk space here, and delete all archives and unnecessary videos from your pi ASAP, or at least monitor the disk space.  

Unless you plan to do some insane panning or zooming (which is actually very cool), consider dropping your image resolution to 4k.  Otherwise you should try something like 75-150M bitrates.  Just keep in mind your video will render VERY slowly and will be ABSOLUTELY HUGE.  Still, they will be ultra high quality.  

For testing purposes you can actually upload images into octolapse to test, BUT you are limited to I believe something like 100MB upload (working on this):

[image]

I have never tried rendering 21MP images into a video before, and am frankly surprised it worked, lol!  Well, some brave soul needs to try it first.   I'm not surprised VLC can't play this format, but your video editing may well support it (most support much larger frames than they will actually render as output for panning/zooming, etc...)

If you increase your bitrate, also consider turning up your rendering cores.  I would guess you have a PI3?  If so, set the rendering cores to 3 and your videos will render approx 3x as fast.  4 cores isn't a good idea, as it will severely impact your pi's performance.

Let me know how it goes!

WatchOut1233, 2020-07-23:
ok dont mean to keep making ur email ring lol. But I made some changes, thx for the breakdown up there this is all making sense to me I set the archive up, I would assume it would take the same space. wouldnt this take the same amount up space untill render completion. and is only considered more space because it then dosnt delete the file? now I decided to use h264 instead of mp4  what u think the constant rate factor should be? I use a Pi4 and currently have it set to 4 threads ill back it to 3 i can wait haha. space dont bother me ill go get a tb if i have to haha. (rule of thumb can dumb a video down later but can never dumb it up ;))    I do plan on panning zooming etc which is exactly why I am trying to reach top quality and res so I have that option I also am designing a action moving cam right now with some steppers and my arduinos.

FormerLurker, 2020-07-23:
>  I would assume it would take the same space. wouldnt this take the same amount up space untill render completion.

So, it will take the same amount of space right up until the images are zipped.  Since JPG files are about as compressed as they're going to get, here is a breakdown of the space required:

1.  Space of the images themselves (say 100MB for example)
2.  Space of the rendered Video (say 50MB for example)
3.  After rendering space for the zip file (very close to 100MB).
4.  After zipping the original images are deleted (minus 100MB)
5.  If you choose to render an archive in that screen I posted earlier Octolapse will unzip the images (Add 100MB)

In some cases more space can be used if, for example, you are acquiring raw images.  These are converted before rendering (if possible, Octolapse does it's best.)  Rendering overlays and transposition options will also add to the space budget (worst case the size = size of images).

> now I decided to use h264 instead of mp4 what u think the constant rate factor should be?

This isn't likely to work.  Your PI will run out of memory way before you are able to render videos of the resolution you are talking about.  H265 100% will not work.  At 1080P it might work.  If you want to use H264 I'd recommend downloading the image archive and rendering them on a PC with a lot of mem.

>  I use a Pi4 and currently have it set to 4 threads ill back it to 3 i can wait haha.

If you don't do anything while it is rendering 4 cores is fine.  Your interface will be horribly slow and your pi may get hot and throttle unless you have an excellent cooling fan.  I know the 4s can get pretty toasty, but don't have one yet myself.

> I do plan on panning zooming etc

If this is the case, some editors will allow you to import the frames without rendering to a video.  Check yours to see if this is supported.  If it is, you can disable rendering entirely.  If rendering is disabled Octolapse will automatically archive the images for download.

WatchOut1233, 2020-07-23:
ok so im gona double space temporary for the zip then carry the size it would be if It didnt delete the images after makes sence.

my pc is a power house, pi has 8 gb. if thats not enough then I will go back to mp4 although weird to me mp4 before still gave me a 6 k video when Its in my editor. but my pc ha 64 gig. so maybe rendering my self is best for getting max quality and bumping up my sd size in the pi. 

I use fimora 9 and I think it allows me to do this. think when it comes to render or not render I will run some test to see what i can get away with after the fact but starting to look like just getting the archive and not rendering may be best.

now are you saying I should use raw or jpeg fine?

and one thats bugging me is if I decide not to use H264 what are my losses in quality?

WatchOut1233, 2020-07-23:
side note the pi 4 was a decent price 99 dollers I think. I never owned one before but my wife knows what I can make happen when she gets me tech style gifts so bam she got me one for my birthday haha

WatchOut1233, 2020-07-23:
just for the sake of conversation I am a robotics specialist by trade with a degree in electronics engineering, covid just has me tinkering away like tony stark hahahah

WatchOut1233, 2020-07-23:
I am currently running a go big or go home test haha everything maxed constant rate 0, h264, render and archive 3 threads. just have to see if it can do it you know haha. start at the top and work backwards. im going to run 650 layers about a 5 inch test see what i get out of it. (maybe a fire) haha jk

FormerLurker, 2020-07-23:
> now are you saying I should use raw or jpeg fine?
Octolapse will convert to JPG if you use raw, so best to use JPG.  My guess is your camera's jpg conversion is probably better than Pillow with the default options.  I added raw conversion only because it was a constant issue (why is my rendering failing!), but I wouldn't consider using it for such large images.  I highly doubt you will notice any change to the quality of the rendered video.

> and one thats bugging me is if I decide not to use H264 what are my losses in quality?

Well, it depends on your bitrate.  264 will give you a higher quality per megabyte, so a high quality mp4 using xvid (default mp4 option in Octolapse) will be much larger than a video of similar quality using the 264 codec.  Also, 264 maxes out out at 4096 × 2048, so it's not suitable for 24MP images anyway!  H265 DOES support larger resolutions, up to 7680 x 4320, but it is an absolute memory hog, and there are some bugs in the codec (at least the one I use).

Regarding this comment about your Pi4:

>  she got me one for my birthday haha

That's a keeper right there 👍 

> just for the sake of conversation I am a robotics specialist by trade with a degree in electronics engineering, covid just has me tinkering away like tony stark hahahah

You could probably guess I'm a software engineer.  However, I was involved very briefly in aerial robotics way back when the weight/CPU power ratio was ALMOST good enough to try such things.  I haven't messed with it since (beyond 3d printer stuff, which I suppose is sort of close in some ways), but it was super interesting!

> I am currently running a go big or go home test

That's often the best way to do it.  If rendering fails, it will show up in a giant red 'Failed Renderings' button for you to try again.  Pay close attention to the options there.  Octolapse saves all of the settings you used in the initial rendering, but you can switch to other available profiles for both the 'Camera' and 'Rendering' profiles.  Let me know if you have problems, but DO check out the blue question marks and read the integrated help first.  Hopefully it will answer the basic questions.  If not, let me know and I'll improve the documentation.

WatchOut1233, 2020-07-23:
Ya in a lot of ways I took the turtorial as a final end solution to what I was trying to do. never re visited any of these settings. now understanding your tutorial wasnt exactly for 6k haha. but thx for all your help and I will report back with how things turned out. I am the type of person whos quicker to double my sd card size or completely use a dedicated laptop to solve quality issues then settle for anything less haha. so evern if it takes me some time and more work after the fact I will be doing everything in my power to preserve every last detail the camera produces

WatchOut1233, 2020-07-24:
Ok so I have been up for 24 hours since I am "Extra". and I have solved all my issues thx to you. I want to start by saying thank you If I didnt get your breakdown I would of been 2 weeks behind figuring it all out by my self so again I appropriate it. you cant stop me from donating when I can. lmao If someone else created some of the automotive robotics software that I created in my day I would donate to keep the work alive so I really feel you deserve every bit of help you can get. I can only image the shutter damage alone you have endured just developing and updating/changing based off everyone needs and wants. It truly takes a remarkable person to also want to help people as much as you have shown in all your closed issues and just everything in general. (hats off to ya )

now down to business.  I did start by going big or going home with the pi rendering first it took all of 20 mins to relize like u said it wasnt gona meet my demands. so bam turned generation off. since jpeg fine was good enough for me since i didnt need the extra data to manipulate since iv controlled my variables having a enclosed light box with all reflective white surfaces. I did try dng but the preview got messed up and it really bogged down the system and like i said before i wanted to have it not on the camera as to not touch it. after taking what you said i decided to go on the look out for software and I tell ya light house, tr time lapse, and fimora 9 is all you need to take a bunch of 8k res photos and turn them into a masterpiece. It allowed me to deflicker the photos and prep them after converting them to dng. And all I can say is wow, poof all my issues were gone smooth beautiful lapse rendered at 80Mp. Love it. Send a vid when this hulk is done printing thx again.

WatchOut1233, 2020-07-24:
now realize after saying all that, why did i have to stay up 24 hours haha . had to teach my self how to use the software and got some great toys in. new printer artillery sidewinder x1 v4 baby and a bl touch to go with it. god bless this beautiful world.

FormerLurker, 2020-07-24:
Thank you vert much for your kind words!  I am glad to help, and impressed that you stuck with it.  I expect to see great things in my youtube feed from you, lol.  Keep me posted.  I hope to hear from you again soon!

WatchOut1233, 2020-07-24:
will do no job for the past 6 months, this is all that is keeping me sane. thal shalt tinker lol.

WatchOut1233, 2020-07-28:
https://www.youtube.com/watch?v=W9WZkp9AknI said I would give a clip when I got one finalized. Very happy with the results. Cheers