FormerLurker/Octolapse #570: Extrusion/Retraction issue?

spanzetta, 2020-07-21:
Hello

first of all let me tell you a big thank and big congratulation for the great sw you developed...

Anyway.. after few usage I noticed a small problem caused by Octolapse... that maybe it's caused by some wrong settings (my fault)

If you look the attached screenshot (and the full video at the end of this report) you will notice a small imperfection caused by (not enough?) retraction so there is some extrusion in correspondence of the head position for octolapse pictures (in my case it was Centered/Smart-compatibility)

Is there something that can be done to improve the result by avoiding such small imperfection?

That issue doesn't happen if I disable Octolapse..

Thanks and regards

[image]

#### Version of Octolapse
Octolapse Version: 0.4.0

#### Version of OctoPrint
OctoPrint Version: 1.4.0

#### Operating System running OctoPrint and Octolapse
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

#### Printer model & used firmware incl. version
Printer Model: Ender 3 Pro
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

Pls check the full video at https://www.youtube.com/watch?v=HU7WurIzkGc

Gordontgopher, 2020-07-23:
Just to say I am seeing similar issues on my Prusa Mini running 4.1.0 with Octoprint / Octolapse running on a Pi 4B running RaspOS (raspbian) buster. Octolapse is configured for Centered/Smart-compatibility. First time I've used Octolapse, so it maybe a case of "PEBCAK" :) I will update if I discover anything significant.

spanzetta, 2020-07-25:
@Gordontgopher thanks for the confirmation 

Any comments/news form the author?

Thanks

FormerLurker, 2020-07-29:
Sorry about the delay all!  Been swamped here lately.

@Gordontgopher, I would try using fresh filament.  The prusa mini is known to have very few issues that would create the kinds of oozing artifacts seen in OPs image.  Direct drive greatly reduces the problem.  I do not see this when using my Mk2S UNLESS I am using old PLA or the temps are off, ESPECIALLY when stabilizing in the center.  However, if you are using a DSLR or have a large delay in your camera settings (more than 250MS), all bets are off :)

@spanzetta, I see this a lot for folks using printers similar to yours.  If you watch as Octolapse takes pictures, you will see a tiny bit of oozing filament get brushed off on the same spot over and over.  That's why they appear to grow up like that.  There are so many possible reasons for this that it's hard for me to go through them all.  Allow me to direct you to [this print quality guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues).  If you've already looked there please let me know.

Gordontgopher, 2020-07-29:
Thanks for the advice, the printer and filament (Prusament PLA) were about a week old, so I'd be surprised if this was the problem, but then again my experience only as old as the printer! I'm certainly not blaming Octolapse and realise the issue is probably the printer or the user :) I'll do some more experiments when I get a spare minute and follow up in the forum or discord if necessary.

FormerLurker, 2020-07-29:
Actually, feel free to blame Octolapse!  It can and does cause these problems, though I work hard to minimize it as much as possible.

Actually... Isn't the mini a bowden setup?  Oozing is generally more of a problem on those.  I don't have a mini, but I'm have seen plenty of videos without oozing, so I know it is possible.  If you can run Octolapse with no obvious quality issues you know you have a well tuned and well built machine!

spanzetta, 2020-08-07:
Dear
Nobody is blaming Octolapse..

We all should be (and I am) really grateful to you for your job (done for free/the community).

My intention was only to report you something that I noticed that maybe/eventually could have some solution (or improvement) on your code/sw side.. so the end scope is only to improve further your great sw.. 
Maybe a better Retraction management could solve/reduce the issue? Just a guess 

Thanks again for your great job

FormerLurker, 2020-08-07:
@spanzetta, thanks for your kind words!  Perhaps "blame" was the wrong word, but Octolapse definitely does cause these artifacts in some cases.  Just being honest.

Now, regarding retraction management, do you have any ideas how it can be improved?  I would be very interested in new ideas that would help things.

spanzetta, 2020-08-07:
Dear FromerLuker
Thanks for your reply..

My idea was to add the possibility to edit (into octolapse options)  the retraction command parameters that you add to the code before moving the head to the position to take the lapse picture (in my case it was "center").. 

If we could edit this value we could experiment different value to obtain better results..

I am assuming that before moving the head.. you add a retraction command.. isn't it?

Thanks a lot

FormerLurker, 2020-08-07:
I have thought about adding a retraction adjustment.  Currently it uses the slicer settings for retraction if you are using automatic slicer settings.  Perhaps an override would be a good feature.  Octolapse needs to know the slicer setting too though, in order to detect retraction for other quality reasons.

I have a few bugs to fix too, so I will look into that while I am working on those.

spanzetta, 2020-08-07:
_Octolapse needs to know the slicer setting too though, in order to detect retraction for other quality reasons._
you can start with supporting Cura (I use it and I assume it is the most used one)...
Thanks for considering it..

FormerLurker, 2020-08-07:
Are you currently using the automatic slicer settings detection with cura?

spanzetta, 2020-08-07:
Correct.. and I did follow your guide at https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration#install-the-cura-settings-script

ElCheekytico, 2020-08-27:
I too have these zits/blobs at the point where the extruder moves away for the snapshot to be taken. I am using Cura with automatic configuration and have the gcode added per your guide referenced in the previous comment.

FormerLurker, 2020-08-27:
@ElCheekytico, have you looked at [this quality guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues)?  It may help.  I am working on allowing retraction adjustments, but it will take a while.  In general, most quality issues can be partially solved (perhaps not completely eliminated) by adjusting your slicer settings.

Also, if you post a pic of the issues you are seeing, I may be able to give you more specific advice.  The kinds of artifacts shown by OPs post are usually related to oozing.  Worse issues (giant blobs where the extruder left to take a snapshot) are usually due to configuration issues, most likely the G90/G91 influences extruder setting in the printer profile. 

Let me know

ElCheekytico, 2020-08-31:
> @ElCheekytico, have you looked at [this quality guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues)? It may help. I am working on allowing retraction [...]

My apologies. I didn't realize what my flow rate was. I lowered that to a proper number and that mostly eliminated the zits/blobs.

FormerLurker, 2020-08-31:
That would make a difference.  Glad it is working, and thanks for the update!

spanzetta, 2020-09-08:
I am now testing with "**Smart - snap to print - smooth**" trigger (stabilization Disabled) and it seems to works perfectly!
No blobs/artefacts on printed object..
I think I will always use this method.. which works fine!
Thanks

FormerLurker, 2020-09-09:
@spanzetta, that is AWESOME!  I too love that trigger.  Depending on the model it can make a really cool timelapse, though it can also yield a pretty jerky timelapse in some cases.  If you are interested primarily in quality and speed, that is the one to use :)

spanzetta, 2020-09-10:
@FormerLurker there is only one very small issue.. the first picture is taken when printing the side line (in the middle).. added by Cura  before starting any print..
Any idea why it happens?
Thanks

FormerLurker, 2020-09-10:
Yes I know.  It's most likely because the priming line is on a different layer (height wise) than the actual printed part.  This is extremely common since priming is within the start gcode and is not adjusted on a per-print bases.  There are a few ways to deal with this:

1.  Adjust the priming line height to match the layer height (a pain, can cause squished priming, not recommended)
2.  Set the 'priming height' within your printer profile to exactly match the height of the priming line.  You can discover this height by reviewing your start gcode.
3.  The easy way - Use the new @Octolapse command to tell Octolapse to ignore your start gcode entirely by putting this at the top of the start gcode:

@OCTOLAPSE STOP-SNAPSHOTS

Then again at the bottom of your start gcode add this:

@OCTOLAPSE START-SNAPSHOTS

I would also recommend adding this to the very beginning of your end gcode:

@OCTOLAPSE STOP-SNAPSHOTS

That way your start and end gcode will never affect your timelapse!  Enjoy :)

spanzetta, 2020-09-13:
Thanks a lot!! 
I add the suggested code to start/end g-code .. will test it soon and will let you know
Regards

spanzetta, 2020-09-14:
I can confirm.. it now works perfectly
see attached

[Stringing_Test_20200914110730.zip](https://github.com/FormerLurker/Octolapse/files/5217424/Stringing_Test_20200914110730.zip)

FormerLurker, 2020-09-17:
Excellent!  Shall I close this issue?

Very glad things are working now :)

spanzetta, 2020-09-18:
Sure.. it can be closed

Thanks a lot.

closed 2020-09-18 by FormerLurker