FormerLurker/Octolapse #571: [BUG] G29 parsing causes snapshot triggers during purge

mrseeker, 2020-07-29:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

I recently upgraded my motherboard and replaced the firmware. With this I also set the firmware to use "Unified Bed Levelling", which requires a different type of G29 commands. Files printed with this new gcode command have issues with snapshots going off after the purge.

#### What should have happened?
The purge should happen first, then a snapshot should happen in the first layer being printed.

#### What happened instead?
A snapshot is being created just after the purge, before the first layer.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender-3 Pro
Printer Firmware Version: Marlin 2.0.6 with BLTouch & UBL enabled.

#### Link to the gcode file you were printing when the problem occurred
Link to Gcode File: https://www.dropbox.com/sh/h75ttmjdgx6muuh/AAD1hhEvq0gXV6k1oQX97vt7a?dl=0
Offending Gcode below, since the full files are large.

```
; Ender 3 Custom Start G-code
G90; absolute position
M117 GETTING EVERYTHING UP TO TEMP!
M104 S160; start warming extruder to 160
M140 S60; Set Heat Bed temperature
M190 S60; Wait for Heat Bed temperature
G28; Home all axes
M117 AUTO BED-LEVEL GO!
;G29 ; Auto bed-level (BL-Touch)
G29 L1; Load the mesh stored in slot 1 (from G29 S1)
G29 J; No size specified on the J option tells G29 to probe the specified 3 points and tilt the mesh according to what it finds.
G92 E0; Reset Extruder
G1 X0 Y0 F3000;Back to base!
M117 GETTING THE EXTRUDER UP TO TEMP!
M104 S230; Set Extruder temperature
M109 S230; Wait for Extruder temperature
G1 Z1.0 F3000; move z up little to prevent scratching of surface
G1 X0.1 Y20 Z0.3 F5000.0; move to start-line position
M117 LET THE PURGE BEGIN!
G1 X1.1 Y200.0 Z0.5 F1500.0 E15; draw 1st line
G1 X1.4 Y200.0 Z0.5 F5000.0; move to side a little
G1 X1.4 Y20 Z0.5 F1500.0 E30; draw 2nd line
G92 E0; reset extruder
G1 Z1.0 F3000; move z up little to prevent scratching of surface
M117 AUTOBOTS! ROLL OUT!
; End of custom start GCode
```

#### Link to settings.json

Link to settings.json with all passwords removed: https://www.dropbox.com/sh/h75ttmjdgx6muuh/AAD1hhEvq0gXV6k1oQX97vt7a?dl=0

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://www.dropbox.com/sh/h75ttmjdgx6muuh/AAD1hhEvq0gXV6k1oQX97vt7a?dl=0

#### Link to octoprint.log

Link to octoprint.log:  https://www.dropbox.com/sh/h75ttmjdgx6muuh/AAD1hhEvq0gXV6k1oQX97vt7a?dl=0

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://www.dropbox.com/sh/h75ttmjdgx6muuh/AAD1hhEvq0gXV6k1oQX97vt7a?dl=0

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-07-29:
This is because your priming line is printed at a different layer height than the rest of your print (0.4mm), and is interpreted as the only item on that layer.  There are a few ways to deal with this:

1.  Adjust your priming height setting in your printer profile to 0.4mm:

[image]

2.  This is the preferred option, because it won't require any changes to your printer profile, and is a whole lot more flexible.  Use the new @Octolapse command to tell Octolapse not to take any snapshots within your start or end gcode.  For example, you can place this command at the very top of your start gcode:

```@OCTOLAPSE STOP-SNAPSHOTS```

Then place this command at the very bottom of your start gcode:

```@OCTOLAPSE START-SNAPSHOTS```

Finally, at the very beginning of your end gcode, add this line:

```@OCTOLAPSE STOP-SNAPSHOTS```

Let me know how it goes!

mrseeker, 2020-07-31:
That seemed to have remedied the situation, now it works 👍

closed 2020-07-31 by mrseeker