FormerLurker/Octolapse #572: [Request] Popup text modification for M118 error

trumpton, 2020-07-29:
#### If this is a feature request describe it here

Certainly with the Prusa MK3S printer, when there is an interactive menu open (e.g. Heater stopped due to timeout), the printer does not respond to serial port requests.
This is interpreted by a new remote print request as M118 not supported.

Could the text of the M118 not supported error message be changed to "Printer is busy, or M118 not supported"

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-07-29:
Can you take a screen cap of the popup?  ATM I have no idea why Octolapse would be complaining about that gcode.

trumpton, 2020-07-29:
What I had done was leave the printer with the hot end on such that the automatic timeout on the printer turned it off, and the printer had a message with a 'press Ok to continue'.  Whilst this message was on the printer display, the Octoprint could not talk to the printer (I am going via the GPIO onto a Raspberry Pi 4).
I got the error message ... now, I believe it was M118, however, I've tried to re-create the problem quickly and I did a filament load, but when it came to the "is the filament the right colour: yes/no" question, I did nothing, and tried to submit and cancel a job from Octoprint.
I got the attached: "M114 Not supported" message.  Makes me think I misread the previous message, however my request still stands ... could the program say "printer busy or Mxxx not supported".

I've attached a screenshot ...

[image]

Many Thanks, Steve

FormerLurker, 2020-07-29:
Ok, I see now.  That error is part of the Octolapse startup that should happen before any other gcode commands are sent.  I may be able to extend the timeout if a busy message is received and report this to the client.

Question: this is only happening when you start a print, and not in the middle, right?

trumpton, 2020-07-29:
Yes, it's only on starting the print.

And it's caused by the fact that the user has to press a button on the printer.

For me, the printer is in a different room, so I was scratching my head why re-submitting the same print again would generate this error.

Half an hour later, I looked at the printer and noticed it was waiting for me to press the front panel OK button - once I'd cleared the OK, the "paused for user" would have stopped, and Octoprint woke up.

So, I'm not sure a longer timeout would help - just a reminder message to the stupid user (i.e. me!).

Ta, Steve