FormerLurker/Octolapse #573: Classic GCode Triggers NOT case insensitive

CmdrCody51, 2020-08-01:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: Raspbian
Os Version: Buster

#### Printer model & used firmware incl. version

Printer Model: ANET A8 V1.2 motherboard
Printer Firmware Version: Marlin V2.0.5.3

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/CmdrCody51/2a549ecbb01227df72dca2751beb6079

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/CmdrCody51/222d2b0c142ab6fbf971962a219f0f9a

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://gist.github.com/CmdrCody51/f4901268000bddac849fe8401e5093d0

#### Link to octoprint.log

Link to octoprint.log: https://gist.github.com/CmdrCody51/117c1a9b7dc73afbbdbeca9230611eda

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]
[image]
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-08-01:
Sending gcode from the terminal will not trigger Octolapse, sorry.

CmdrCody51, 2020-08-01:
> Sending gcode from the terminal will not trigger Octolapse, sorry.

Then why are the other two blocked like OctoLapse says it does but not the lower case?

FormerLurker, 2020-08-01:
Hmm, not sure I understand what you mean.  The first command (@Octolapse) will never be sent to the printer.  Octoprint prevents anything starting with @ from being sent.  Snap was sent, but was not recognized by the printer, which is expected since it is not gcode.

What were you expecting?  Also, if you are just testing, use the smart gcode trigger, add your snap or @octolapse commands to your gcode file, and start a print to see the preview.  Octolapse will allow you to continue or to abort the print if you dont like what you see.

Does that make sense and answer your question?  Have you read the getting started guide?

CmdrCody51, 2020-08-02:
> Hmm, not sure I understand what you mean. The first command (@Octolapse) will never be sent to the printer. Octoprint prevents anything starting with @ from being sent. Snap was sent, but was not [...]

If you will examine the image, the first command is "SNAP". The second "@OCTOLAPSE TAKE-SNAPSHOT", The third is a lower case "snap". Quoting your help bubble in Alternative Snapshot Commands --
"Snapshot commands are NOT case sensitive in Octolapse V0.4+. Please note that all comments are stripped from the alternative snapshot command while Octolapse is running."

Now - OctoLapse when running a print (even in test mode) will accept "snap" as a trigger.
And then send it right on to the printer.

 So why does it do that?

I don't know. Sounds like a bug.

Yes I have read the getting starting guide......

FormerLurker, 2020-08-02:
I just didn't understand what it was you were showing with your demonstration.  I will take a look.