FormerLurker/Octolapse #575: Cancel print toggle not working

Uko, 2020-08-06:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4

#### Version of OctoPrint

OctoPrint Version: 1.4

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Disable "Cancel print if startup errors are detected.
2. Configure Trigger: gcode
3. Start a print without any snapshot command in the gcode
 

#### What should have happened?
Running print without Octolapse

#### What happened instead?
The print was canceled
[image]

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/jW8TqWFF

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: There is nothing except multiple groups of:

```
2020-08-06 02:19:36,116 - octolapse.__init__ - INFO - Started logging to file.
2020-08-06 02:19:36,124 - octolapse.settings - INFO - Loading existing settings file from: /home/pi/.octoprint/data/octolapse/settings.json.
2020-08-06 02:19:36,171 - octolapse.settings - INFO - Creating settings from iterable.
2020-08-06 02:19:36,259 - octolapse.settings - INFO - Settings created from iterable.
2020-08-06 02:19:36,261 - octolapse.settings - INFO - Settings file loaded.
```

#### Link to octoprint.log

Link to octoprint.log:  
```
2020-08-06 16:27:13,453 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:27:13,459 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=10200 --speed-y=10200 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:27:13,599 - octoprint.plugins.SlicerSettingsParser - INFO - Analyzing file: Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:27:32,966 - octoprint.plugins.SlicerSettingsParser - INFO - Saved slicer settings metadata for file: Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:27:47,869 - octoprint.printer.standard.job - INFO - Print job selected - origin: local, path: Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode, owner: Uko, user: Uko
2020-08-06 16:27:48,007 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-08-06 16:27:48,047 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode, owner: Uko, user: Uko
2020-08-06 16:27:54,596 - octoprint.plugins.excluderegion - INFO - File selected, resetting internal state
2020-08-06 16:27:58,873 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Cancelling"
2020-08-06 16:27:58,892 - octoprint.util.comm - INFO - Force-sending M108 to the printer
2020-08-06 16:27:58,997 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode, owner: Uko, user: None
2020-08-06 16:27:59,083 - octoprint.util.comm - INFO - Changing monitoring state from "Cancelling" to "Printing"
2020-08-06 16:27:59,242 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Operational"
2020-08-06 16:27:59,252 - octoprint.filemanager.analysis - INFO - Starting analysis of local:Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:27:59,264 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python2 -m octoprint analysis gcode --speed-x=10200 --speed-y=10200 --max-t=10 --throttle=0.0 --throttle-lines=100 /home/pi/.octoprint/uploads/Layer_Test/temp-tower-190-230_02mm_pla_mk3s_3h56m.gcode
2020-08-06 16:28:00,954 - octoprint.plugins.excluderegion - INFO - Printing started
2020-08-06 16:28:00,964 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelling
2020-08-06 16:28:01,007 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintCancelled
2020-08-06 16:28:01,927 - octoprint.plugins.excluderegion - INFO - Printing stopped: event=PrintFailed

```

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-08-06:
Thanks for the report.  I'm assuming this happens only when using the smart gcode trigger right?

Uko, 2020-08-17:
> Thanks for the report. I'm assuming this happens only when using the smart gcode trigger right?

Hey, sorry for the late reply. Yes, you are right, it only happens with the smart gcode trigger

FormerLurker, 2020-08-17:
OK, I will take a look.  The 'prevent cancel' functionality is actually pretty complicated, so I'm not surprised there are a few cases where it fails.

Marking this as a bug.