FormerLurker/Octolapse #576: Installation error: ImportError: No module named octoprint.server

wmparsons, 2020-08-07:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Run: pip install "https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip"

#### What should have happened?
Install succeeded

#### What happened instead?
Install failed:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip
  Using cached https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip
    Complete output from command python setup.py egg_info:
    likely tags: v0.4.0
    picking v0.4.0
    got version from expanded keyword {'error': None, 'date': '2020-06-07T19:06:32-0500', 'version': 'v0.4.0', 'dirty': False, 'full-revisionid': 'aeb9350b3e812589452b1904c20b7f9c38904ea1'}
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-rm8QoT/setup.py", line 57, in <module>
        import octoprint.server
    ImportError: No module named octoprint.server
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-req-build-rm8QoT/

#### Operating System running OctoPrint and Octolapse

OS Name: octopi
Os Version: 0.17.0

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

wmparsons, 2020-08-07:
I've tried rebooting, upgrading to python 3, rebooting again, etc, but it keeps failing the same.

wmparsons, 2020-08-07:
And from the plugin manager:

Installing plugin "Octolapse" from https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip.../home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpzerkngc1/Octolapse-0.4.0.zip --no-cache-dirTraceback (most recent call last):File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main"__main__", mod_spec)File "/usr/lib/python3.7/runpy.py", line 85, in _run_codeexec(code, run_globals)File "/home/pi/oprint/lib/python3.7/site-packages/pip/__main__.py", line 26, in <module>sys.exit(_main())File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/main.py", line 73, in maincommand = create_command(cmd_name, isolated=("--isolated" in cmd_args))File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/commands/__init__.py", line 104, in create_commandmodule = importlib.import_module(module_path)File "/home/pi/oprint/lib/python3.7/importlib/__init__.py", line 127, in import_modulereturn _bootstrap._gcd_import(name[level:], package, level)File "<frozen importlib._bootstrap>", line 1006, in _gcd_importFile "<frozen importlib._bootstrap>", line 983, in _find_and_loadFile "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlockedFile "<frozen importlib._bootstrap>", line 677, in _load_unlockedFile "<frozen importlib._bootstrap_external>", line 728, in exec_moduleFile "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removedFile "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 17, in <module>from pip._internal.cli.req_command import RequirementCommand, with_cleanupFile "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 16, in <module>from pip._internal.index.collector import LinkCollectorFile "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/index/collector.py", line 14, in <module>from pip._vendor import html5lib, requestsFile "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/requests/__init__.py", line 43, in <module>from pip._vendor import urllib3File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/urllib3/__init__.py", line 7, in <module>from .connectionpool import HTTPConnectionPool, HTTPSConnectionPool, connection_from_urlFile "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/urllib3/connectionpool.py", line 29, in <module>from .connection import (File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/urllib3/connection.py", line 41, in <module>from .util.ssl_ import (ValueError: source code string cannot contain null bytesError!Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

FormerLurker, 2020-08-08:
Probably an issue with the latest Octoprint release.  I will investigate this.

wmparsons, 2020-08-08:
Let me know if there's anything else I can provide that might help debug further.

FormerLurker, 2020-08-08:
Ok, will do.  I have found two issues with this release so far.  I was warned about one of them, but was too slow to act :(

zanhecht, 2020-10-20:
I had the same error, but fixed it by using `~/oprint/bin/pip install "https://github.com/FormerLurker/Octolapse/archive/v0.4.0.zip"` instead of just `pip install`.

jstuewe, 2021-05-19:
That worked for me as well to get ArcWelder installed.  Thanks!
It does give a PIP version warning though.  I'm up to date on OctoPrint but I'm one revision back on OctoPi.

------------------------------------------------------------------------------
OctoPrint version : 1.6.1
OctoPi version    : 0.17.0
------------------------------------------------------------------------------
pi@octopi:~ $ ~/oprint/bin/pip install https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc2.zip
....
Successfully installed Arc-Welder-1.1.0rc2
WARNING: You are using pip version 19.3.1; however, version 20.3.4 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

FormerLurker, 2021-06-01:
@jstuewe,

That warning isn't going to cause any problems (yet), though you might consider running

```pip install --upgrade pip``` 

on the appropriate virtual environment.  If you google *Upgrade Pip Octoprint* I bet you can find step-by-step instructions.

ku3mich, 2023-09-28:
try activate venv and use just octoprint to run - works
```
which octoprint
/home/okto/core/venv/bin/octoprint
```
core - folder where octoprint is installed
```
source venv/bin/activate
```
to activate venv

```
virtualenv --python=python3 venv
```
to create venv

maybe it will help in the investigation