FormerLurker/Octolapse #577: TypeError

lukhihardik11, 2020-08-08:
Every time I start the print I get this error.
I checked my script which is ok.
[plugin_octolapse (11).log](https://github.com/FormerLurker/Octolapse/files/5044755/plugin_octolapse.11.log)

Traceback (most recent call last):
  File "/home/hardiklukhi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/__init__.py", line 2423, in on_print_start
    timelapse_settings = self.get_timelapse_settings()
  File "/home/hardiklukhi/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/__init__.py", line 2652, in get_timelapse_settings
    if not os.path.isfile(ffmpeg_path):
  File "/usr/lib/python3.8/genericpath.py", line 30, in isfile
    st = os.stat(path)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType

FormerLurker, 2020-08-18:
Interesting.  Try going into the OctoPrint webcam (NOT octolapse) and see if the FFMPEG path is entered.  If it is not, enter it in, then verify using the button that the path is correct.  Finally save your changes and restart OctoPrint and see if it starts working.

In any case I'll need to add a null check and an error message for this issue.  I've never seen the path be null before, but there are so many different installations out there it doesn't surprise me :)

lukhihardik11, 2020-08-18:
> Interesting. Try going into the OctoPrint webcam setting (NOT octolapse)

I was running linux on TX6 android tv box . i tried all linux version
ubuntu and Debian both . I tried python 2 &3 both but showed me same . i
was trying to run DSLR script .
but  it showed me that error. i don't have web. cam .

FormerLurker, 2020-08-18:
FFMPEG is used to convert the snapshots into a video, and is required.  I will considering altering the requirements check so that it is skipped if rendering is disabled.  For now, install ffmpeg if necessary and enter the ffmpeg/avconv path into this setting (OctoPrint setting, not Octolapse):

[image]

Let me know if this workaround solves your problem for the moment.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker