FormerLurker/Octolapse #578: Octolapse 4 cancels capture but allows print to continue says it is expecting an explicit G21

sr1329, 2020-08-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
 4.0

#### Version of OctoPrint
 
OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1.  Configuration is Ender 3 profile
2.  Started a print
3.  Error appeared as a notification stating that Octolapse is expecting an M21 code. The print will continue but Octolapse will not. 
4. Added M21 to start code
5. Same error upon attempting a print
6. Checked settings and found the dropbox to change from requiring G21 explicitly to assuming millimeters without a G21
7. Same error upon attempting a print

#### What should have happened?
Octolapse should have accepted the G21 code in the start code or not required an explicit G21 when set to assume millimeters.

#### What happened instead?
Octolapse cancelled its operations and allowed the print to continue

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 18.1 beta
Os Version: 18.1

#### Printer model & used firmware incl. version

Printer Model: Ender 3 V2
Printer Firmware Version: Marlin 2.0.6 

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: All settings default and retrieved from server

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-08-15:
It looks like the links to gcode, the log file, and your settings.json failed.  Can you attach those somehow (gist.github.com, dropbox, etc)?  Thank you for the report!

sr1329, 2020-08-15:
Sorry I just got around to it. Octoprint.log is too big for Pastebin. Thank you for looking over all of this.

FormerLurker, 2020-08-17:
No problem, this is a busy time of the year for me anyway, so I've been pretty slow to respond myself (I am usually very prompt).

I will review these logs ASAP, thanks!

GB560, 2020-11-17:
@sr1329 did you ever figure this out? I am having the same issue

FormerLurker, 2020-11-17:
Sorry, no.  It has something to do with the Pi4.  I have one on order.

sr1329, 2020-11-17:
Honestly I haven’t checked. I switched to a 4GB pi 4. It seems due to the 64 bit OS needed by the 8GB model.

FormerLurker, 2020-11-17:
@sr1329 , can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

GB560, 2020-11-18:
@FormerLurker I tried it, worked like a charm! Thank you!

FormerLurker, 2020-11-18:
Thank @tech2077, who figured out what was going on.  Turns out converting bool values to long ints doesn't work well with certain compiler optimizations.  Anyway, thanks for testing it out!  This will go live soon.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker