FormerLurker/Octolapse #579: Octolapse does not take any snapshot but logs an error

Gerporgl, 2020-08-16:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.0 and 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES (default logging level)

#### What were you doing when the problem occurred

1. Printing a test cube with octolapse in test mode

#### What should have happened?
Octolapse should have taken a snapshot instead of showing an error on each layer change (this is my understanding of what octolapse should be able to do)

#### What happened instead?
There was a red popup in octoprint from octolapse saying "Stabilization Error
An unexpected error was encountered while running the timelapse snapshot procedure."
I tried this initially a couple of weeks ago with Octoprint 1.4.0, then retried today with 1.4.2 but getting the same error message.
I also tried with other Stabilization setups but that did not help.

#### Operating System running OctoPrint and Octolapse

OS Name: Armbian Focal for Odroid C2
Os Version: Armbian Focal Fossa (20.04) with Linux 5.6.18-meson64

#### Printer model & used firmware incl. version

Printer Model: Artillery Sidewinder X1
Printer Firmware Version: Marlin 2.0.6 and also tried with 2.0.3 previously

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: I don't think it is needed since I tried with many different gcode files, the error seems to be something else entirely.

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log

```
2020-08-16 02:43:55,536 - octolapse.__init__ - INFO - Started logging to file.
2020-08-16 02:43:55,649 - octolapse.settings - INFO - Creating settings from iterable.
2020-08-16 02:43:56,044 - octolapse.settings - INFO - Settings created from iterable.
2020-08-16 02:43:56,055 - octolapse.settings - INFO - Default settings loaded.
2020-08-16 02:51:33,067 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/timelapse.py", line 407, in _take_timelapse_snapshot_precalculated
    snapshot_gcode = self._gcode.create_gcode_for_snapshot_plan(
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1106, in create_gcode_for_snapshot_plan
    if not self.initialize_for_snapshot_plan_processing(
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/stabilization_gcode.py", line 363, in initialize_for_snapshot_plan_processing
    self.distance_to_lift = snapshot_plan.initial_position.distance_to_zlift(self.z_lift_height)
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/gcode_processor.py", line 515, in distance_to_zlift
    if amount_to_lift < utility.FLOAT_MATH_EQUALITY_RANGE:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
2020-08-16 02:52:37,320 - octolapse.timelapse - ERROR - Failed to take a snapshot for the provided snapshot plan.
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/timelapse.py", line 407, in _take_timelapse_snapshot_precalculated
    snapshot_gcode = self._gcode.create_gcode_for_snapshot_plan(
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/stabilization_gcode.py", line 1106, in create_gcode_for_snapshot_plan
    if not self.initialize_for_snapshot_plan_processing(
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/stabilization_gcode.py", line 363, in initialize_for_snapshot_plan_processing
    self.distance_to_lift = snapshot_plan.initial_position.distance_to_zlift(self.z_lift_height)
  File "/home/octoprint/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octolapse/gcode_processor.py", line 515, in distance_to_zlift
    if amount_to_lift < utility.FLOAT_MATH_EQUALITY_RANGE:
TypeError: '<' not supported between instances of 'NoneType' and 'float'
```

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE
```
2020-08-16 02:50:35,574 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': 'local', 'file': '2662cd4e7fecc597d4fae99e438834877064b005'}
2020-08-16 02:50:55,805 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-08-16 02:50:55,820 - octoprint.util.comm - INFO - Changing monitoring state from "Starting" to "Printing"
2020-08-16 02:50:57,854 - octoprint.util.comm - INFO - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2020-08-16 02:51:09,455 - octoprint.util.comm - INFO - Telling the printer to set the busy interval to our "communicationBusy" timeout - 1s = 2s
2020-08-16 02:53:13,128 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Cancelling"
2020-08-16 02:53:13,163 - octoprint.printer.standard.job - INFO - Print job cancelled - origin: local, path: Cube.gcode
```

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-08-17:
I saw this in another issue.  Looking into it.

FormerLurker, 2020-11-17:
@Gerporgl , can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker