FormerLurker/Octolapse #580: Help with error : The printer profile requires an explicit G21...

sandyprint-prusa, 2020-08-18:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

Octolapse Settings:
Printer – (Imported) Prusa MK3S (profile version 1.3)
Slicer Settings: Automatic Configuration
Stabilization – Back Left (non custom, all defaults)
Trigger – Smart – Compatibility (profile version 1.1) (non custom, all defaults)
Rendering – MP4 – 60 FPS (Profile Version 1.1) (non custom, all defaults)
Cameras – Logitech USB webcam
Logging – Log all Errors
Test Mode enabled
Snapshot Plan Preview – Disabled (with this enabled the octolapse will not pass the print on and will crash)
Snapshot Plan info – Disabled (with this enabled the octolapse will not pass the print on and will crash)
Printer: Prusa i3 MK3S
Octoprint and Octolapse are running on a Raspberry Pi 4B 8GB

When sending a print the printer will do bed leveling, and begin printing the first layer. About halfway through first layer Octolapse will show a stabilization error.
It will finish the layer and move the head to the back left and then act like it is going back to print another layer. Then it will move the head to the back right and start moving up the Z axis. It may go all the way up the z Axis or will eventually have a communication error and Octolapse will disconnect from the printer.
Troubleshooting:
if I change the trigger to a classic trigger like Classic – Gcode I get this error:
 
I have changed the printer profile setting for firmware settings to be default to absolute, default to relative, and setting the default units to require explicit G21.
I have manually entered a G21 at the beginning of the code in prusaslicer and I have put a G4 P1 at the change layer so it should get the gcode on the classic tigger and I still get the same error.
I verified that the models I am using were created using mm, and I created my own model of a tall rectangle and verified it is in mm.

#### What should have happened?
a time lapse should have begun moving the print head to the back after each layer shift.

#### What happened instead?
error Octolapse Timelapse Stopped
The printer profile requires an explicit G21 command before any position altering/setting commands, including any home commands. Stopping timelapse, but continuing the print. was received and print continued but Octolapse did not.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.18.0

#### Printer model & used firmware incl. version

Printer Model: Prusa i3 MK3S
Printer Firmware Version: 3.9.0

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: 
[Colum v0_0.15mm_PLA_MK3S_50m.txt](https://github.com/FormerLurker/Octolapse/files/5087424/Colum.v0_0.15mm_PLA_MK3S_50m.txt)

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  
[octoprint.log](https://github.com/FormerLurker/Octolapse/files/5087427/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 
[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-08-18:
I believe this issue is similar to #578, or at least shares the same root cause.  I've been seeing this since OctoPrint 1.4.2 dropped.  Any way you could try downgrading to 1.4.0 and see if that works?  I know that would be a pain, but if you have the time please consider it.

FormerLurker, 2020-08-24:
Hmm..  Perhaps there is some printer profile error?  Would you mind exporting your settings (gear icon on Octolapse tab, export settings) and posting them somewhere I can find them (like gist.github.com for example)?

If that doesn't work, I may have you create a more detailed log file.

GB560, 2020-11-12:
I am having the same issue.  I have a Raspberry Pi 4 Model B 2019 Quad Core 64, the latest versions of octolapse and octoprint. Following here to see if someone finds an answer.

The error message:
Octolapse Timelapse Stopped The printer profile requires an explicit G21 command before any position altering/setting commands, including any home commands. Stopping timelapse, but continuing the print.

FormerLurker, 2020-11-13:
> The printer profile requires an explicit G21

That error is a red herring.  I've checked many logs and settings, and this is most definitely not the problem.  Has anyone tried the 32 bit OS?

GB560, 2020-11-13:
so install the 32 OS on the 64 Pi? I can try

FormerLurker, 2020-11-13:
@GB560, thanks!  Im trying to set up an emulator, and noticed a lot of people saying the 64bit os had issues, so that's why I am suggesting it.  I should get my pi4 soon.  Anything to narrow this down would be great, especially since I know ppl who had no trouble on the pi4.  There has to be a connection somewhere, just need to find it.

GB560, 2020-11-14:
@FormerLurker 
So this time I just followed the wikis and how-to's without trying to do anything fancy. I am a beginner so please bear with me. Here is what I did:
1. Followed this tutorial: https://www.youtube.com/watch?v=F7blKT3VIYE
2. Downloaded octoprint from here: https://octoprint.org/download/
3. At this point, things working fine with no camera feed (black screen and regular controls rather than the pi cam controls). However, I have access to advanced camera settings and no error. 
4. Since I am using a Pi cam HQ, I followed this: https://github.com/FormerLurker/Octolapse/wiki/Configuring-a-Raspberry-Pi-Camera.
5. Tried my first test print, got this: Octolapse Timelapse Stopped
The printer profile requires an explicit G21 command before any position altering/setting commands, including any home commands. Stopping timelapse, but continuing the print.

GB560, 2020-11-14:
Cannot be an HQ cam issue, this guy made it work: https://www.youtube.com/watch?v=hBc-Ua7lnPk&t=318s

GB560, 2020-11-14:
Correction, I restored to a backup before step 4 above. I can do advanced editing but still getting G21.

FormerLurker, 2020-11-17:
@sandyprint-prusa and @GB560, can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

I-AeroZe-I, 2020-11-28:
@FormerLurker I have this same problem.  I tried installing the link you gave @sandyprint-prusa through the plugin manager, and I got an error with something telling me to try to install using --user switch.  I went to the spot where you could check that box to automatically do that, and it wouldn't install.  I ended up installing it as root on my pi (OctoPi x64 image) using pip3.7.  I can confirm that I just ran a print of a filament spool clip and snapshots were working.  The only issue I've noticed before successfully printing this, is that Octoprint seems to hang when cancelling a print.  I'm not sure what's causing it (I'll check some logs).  It could be unrelated, but I wanted to confirm I had this same exact issue, and installing the zip file from the link you posted works.

I-AeroZe-I, 2020-11-28:
@FormerLurker I am running OctoPrint 1.4.2, OctoPi 0.18.0, and Python 3.7.3.  Oh, I just started another print and cancelled with no issue.

FormerLurker, 2020-11-28:
@I-AeroZe-I, if you cancel at the wrong time (right when a snapshot is initiated), there can be up to a minute of delay.  I thought I had that fixed, but I will review just in case.  Thanks for reporting success!

I-AeroZe-I, 2020-11-29:
> @I-AeroZe-I, if you cancel at the wrong time (right when a snapshot is initiated), there can be up to a minute of delay. I thought I had that fixed, but I will review just in case. Thanks for reporting success!

No problem! Thanks for the great plugin.  I'm new to 3d printing, but not computers.  I have a degree in electrical engineering and work doing software maintenance for the Air Force.  If you ever need me to do any beta stuff or help out, I'll do what I can.  I'm pretty busy in life at the moment, but I do plan on doing some open source development in the near future.

wlam02, 2020-11-29:
@FormerLurker I have the same issue. I also ran Pi 4 2019 x64. I grabbed the build you posted and it fixed it for me. just wanted to say. Thanks!

FormerLurker, 2020-11-29:
@wlam02, thanks, but I actually didn't fix this, just added the fix and committed it :)  Another user submitted it (will look for the username, don't have it on me.

However, I DID fix a ton of stuff that will kill Octolapse if you upgrade to 1.5.  Here is a link the fix for that (hopefully):  https://github.com/FormerLurker/Octolapse/archive/6b53072ddefee1a4121bf893a71d83454fdb9098.zip

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker