FormerLurker/Octolapse #581: Totally useless after update

Rad3d21, 2020-08-18:
can't get the new version working at all.
internal server error. octoprint stops working. not able to detect location. It's a mess.
I'm trying to get the slicer settings correct via manual. Tried with or without autodetect. With or without relative.
It starts processing lines and at the end it's not able to determine position.
Slicer is Ideamaker. It does a G28 so octolapse should detect where the location starts. 
Really disappointed at the moment.

FormerLurker, 2020-08-18:
If you would fill out the template that was provided when you created I might be able to figure out what is going on with your installation.  Please keep in mind that Octolapse is quite complex (multi platform, multiple python version support, custom c++ extensions, backwards compatibility with older versions OctoPrint), so it is extra important that I receive information about where you are running OctoPrint (OctoPi version, linux, windows/mac installs), what browser you are using and on which OS it is running, what printer you have and what firmware it's running, the Octolapse settings.json file, plugin_octolapse.log, version of OctoPrint you are running, and possibly more.

Also, please realize that issues take a lot of time to research and resolve (hours or days), and that this is all FREE, so consider spending some extra time to update this issue so that I can help get it solved.  Venting is fine, but not very constructive.

FormerLurker, 2020-08-18:
Since the template is now gone, perhaps it would be easier for you to simply create a new issue?  That way we can start from scratch.

Rad3d21, 2020-08-20:
Sorry for my frustration. I was going to print a large piece and wanted to make a nice timelapse and found out that after updating the octoprint and octolapse it did not work. Tried several things to resolve this but that didn't work out. I had to uninstall the octolapse to be able to start printing at all. It is still printing at the moment.
My config is Raspberry Pi 4 running buster (pi@octopi:~ $ uname -a
Linux octopi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux) with Octoprint 1.4.2 and Octolapse 0.4.
The octolapse.log is very large. I cannot upload it.

Rad3d21, 2020-08-20:
My slicer is Ideamaker and not in the supported slicers so manual config needed. My printer is a sidewinder X1.
No config because I uninstalled octolapse. The plugin_octolapse.log is 1GB. I will delete it before installing octolapse again. (after my print has finished)
The last lines in the plugin_octolapse.log are:
2020-08-18 17:17:01,743 - octolapse.gcode_position - ERROR - GcodePosition.update_position: Cannot update X because the XYZ axis mode is relative and X is null.
2020-08-18 17:17:01,744 - octolapse.gcode_position - ERROR - GcodePosition.update_position: Cannot update Y because the XYZ axis mode is relative and Y is null.
2020-08-18 17:17:01,744 - octolapse.gcode_position - ERROR - GcodePosition.update_position: Cannot update X because the XYZ axis mode is relative and X is null.

FormerLurker, 2020-11-17:
@Rad3d21 , can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker

Rad3d21, 2020-12-05:
Thanks! I will give it a try.