FormerLurker/Octolapse #584: Printer's head stay on top on impression at the end

Jormundos, 2020-08-28:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

At the end of the print with Octolapse enabled, the printer's head stay right on top of the printed pied with the nozzle in contact with the print, it's seems like the end code in the slicer is not execuded. Without Octolapse, the printer make perfectly the end Gcode.

#### What should have happened?
Do the end Gcode after the printing.

#### What happened instead?
Nothing

#### Operating System running OctoPrint and Octolapse

OS Name: _OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 5 plus
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://gist.github.com/Jormundos/39c1f365c2b86dca552678a55d6eefe3

#### Link to settings.json

Link to settings.json with all passwords removed: https://gist.github.com/Jormundos/39c1f365c2b86dca552678a55d6eefe3

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  https://gist.github.com/Jormundos/39c1f365c2b86dca552678a55d6eefe3

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Jormundos, 2020-11-04:
> Can you post a video showing this problem? I'm not 100% sure I understand
> what's going on.

Hello,

I made a complete reinstallation of Octoprint and Octolapse and the problem
has disappeared.

I don't know how to delete my post. Sorry for the inconvenience.

FormerLurker, 2020-11-04:
Hey, no problem! Thanks for letting me know, and happy printing!!

closed 2020-11-04 by FormerLurker