FormerLurker/Octolapse #585: [Feature Request] Timestamp embedded into timelapse video.

Simrien, 2020-08-29:
Firstly, Octolapse is a top notch time lapse feature. Kudos...

Next, I saw lots of people placing analogue clocks inside their enclosures to give a sense of time passing during the capture.
I did this myself with a kitchen timer and a digital clock.

Then figured, this is high end tech... why not have the current date/time stamped on the frames (circa 1995 style)?
That would be very cool with either an analogue (or digital) clock spinning in one of the corners of the video.
Presumably the timestamp can be read from the frame capture...

Has anyone suggested this before?

FormerLurker, 2020-08-29:
You are in luck, as this feature already exists :)

Check the rendering profile and look for the text overlay section.  There are lots of options there, like font and size selection, alignment, position, and myriad replacement tokens.  You can even preview the overlay before you print.  Let me know how it goes!

Simrien, 2020-08-29:
I LOVE YOU... going to look now :0)

Simrien, 2020-08-30:
It worked perfectly, recommend 72 point text for a 1920x1080 timelapse, thank you :0)

FormerLurker, 2020-08-30:
Woot!  Fyi, if you are using a raspberry pi, install fontconfig for access to all of the installed fonts.  Or you can add your own and have access to whatever you can find online.

Simrien, 2020-08-31:
You're a star, is fontconfig an octopi add-in? I was thinking it needed an 80s style digital clock font.
Next question, how do I give you a donation?

FormerLurker, 2020-08-31:
[FontConfig](https://en.wikipedia.org/wiki/Fontconfig) can be installed via apt-get like so: `sudo apt-get install fontconfig`.  It is not limited to OctoPi, and will run on any (most??) Linux distro.  You can [install your own fonts](https://medium.com/source-words/how-to-manually-install-update-and-uninstall-fonts-on-linux-a8d09a3853b0), reboot (I think this is necessary) and they will show up in the Octolapse font list.

Regarding donations, thank you very much!  You will find links on the Octolapse tab to Patreon, Paypal and Github (I'm on Github Sponsors too):  

[image]

Donations are not required, but are much appreciated!

Simrien, 2020-08-31:
Sent some spons via paypal, thank you, well deserved, top product and top support.

It looks like fontconfig is already installed and up to date so all I need to do is get some TTF files over to the pi.

xx@xxxxx:~ $ sudo apt install fontconfig
[sudo] password for pi:
Reading package lists... Done
Building dependency tree
Reading state information... Done
fontconfig is already the newest version (2.13.1-2).
fontconfig set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 16 not upgraded.

FormerLurker, 2020-08-31:
Thanks!

Also, fontconfig must now be installed by default, which is awesome.  It used to not be.  Let me know if you find a good digital font!

Simrien, 2020-08-31:
You're most welcome.
I got a digital font from https://www.1001fonts.com/digital-fonts.html
Top one, downloaded and unzipped that to my local drive.
Installed the FileZilla 64 bit client.
Be warned, McAfee detects spyware in the official download. I kicked off the install and let McAfee deal with the potential spyware.
Connected to the IP address of the Raspberry PI on port 22 via FileZilla using my Pi username and password.
The official path to install fonts is /usr/share/fonts
However, I kept getting an access denied so did some more research.
The information I found said to create a .fonts folder in the home directory /home/pi/.fonts
So I created the .fonts folder and uploaded the digital-7 folder to that location.
Then using Putty ran the following command... fc-cache -v -f
This picked up the 4 new TTF fonts and were immediately available for use in Octolapse.... ta-da... :0)

FormerLurker, 2021-01-05:
Hey, I added more formatting options to the 'time elapsed' token.  Watch for the next release.

Closing this now, thanks for the submission!

bheiland, 2021-07-20:
I'd be interested in an analog clock feature.  I have seen code for creating an analog clock in python.  The code to make the clock does not seem very complicated.  Is there a way that a analog clock could be drawn on the frames.   The sweep of the minute and hour hand has a nice look to it when a timelapse is playing.  I currently sit a small clock within the view of the camera but wish I could experiment with a drawn clock.  Is there an existing point I could plug code into to draw on the frames if I wanted to experiment with the idea?

FormerLurker, 2021-07-20:
Let me think about this @bheiland.  There may be a way to generalize it, which would be cool 😎

bheiland, 2021-07-22:
I am testing some code outside of the application where I draw a clock on an image.   It is very basic at this point, but it is a starting point.   If I could just squeeze the code into the right section it would draw the analog clock at the correct spot.

Here is the code I came up with for drawing clock on  an image.  I originally looked at code for drawing a clock for turtle, but wanted to used PIL for compatibility with octolapse.   

Please check it out.  It should easily be able to be plugged into octolapse.   There are a few things that are hard coded ( offsets, colors ) etc.  These could be interface selectable .  

Eventually Id like to take the clock face and once drawn, transform to match the bed angle to place clock on bed so it appears that the clock is being projected onto the bed in the image.

Here is the code.

from PIL import Image, ImageFilter, ImageFont, ImageDraw, ImageEnhance
import math
import time

def lineAngleLengthEndpoint(x0,y0,angleDegrees,length):  # calculate x/y coordinates of a line at given length and angle 
    angleRadians = (math.pi /180.0) * angleDegrees  # work in radians
    x1 = x0 + math.cos(angleRadians)*length  
    y1 = y0 + math.sin(angleRadians)*length
    return x1,y1;
try:
    original = Image.open("image.jpg").convert("RGBA")
    #print("The size of the image is: ")
    print(original.format, original.size, original.mode)

    draw = ImageDraw.Draw(original)
    x = 100 #offset center of clock
    y = 200 # offset center of clock
    lengthBlank = 50 # used for drawing tic marks from center outward
    lengthNotBlank =10 # length of tick marks from center plus lengthBlank
    circleRadius = lengthBlank+lengthNotBlank  # outer radius of clock
    
    draw.ellipse((x-circleRadius, y-circleRadius,x+circleRadius, y+circleRadius),outline ="black") # draw outer circle of clock
    for angle in range(0, 360,30):   # make a tick mark every 30 degreez (each hour)
        angleRadians = (math.pi /180.0) * angle;   # need to work in radians
        startX,startY = lineAngleLengthEndpoint(x,y,angle,lengthBlank)  # find x-y value for inside of tick mark
        endX,endY = lineAngleLengthEndpoint(x,y,angle,lengthBlank+lengthNotBlank) #find x-y value for outside of tick mark
        #print(angle , angleRadians,startX,startY,endX,endY);  # print to screen values for testing
        draw.line((startX,startY,endX,endY), fill=128)  #draw tick marks

    hands= [("white",35,12,5),("Blue",50,60,4),("red",55,60,1)] # define hands Houre, Min and sec ("color", length,number of units on face, width)
    
    
    sec = int(time.strftime("%S"))  #  current seconds
    mn = int(time.strftime("%M"))+sec/60 # current minutes, plus partial sec/60 (makes minute ha=nd smoother)
    hr = int(time.strftime("%I"))+mn/60 # current Hour in 12 hour format + partial minutes / 60  (makes hour hand smoother)
    time_set=(hr,mn,sec)   

    for hand in hands:   # step through hands
        time_part = time_set[hands.index(hand)] # 
        angle = ((time_part/hand[2])*360)-90 # example 3.5 hours /12 * 360 will be the number of degrees to point hour hand. Subtract 90 because 0 = 3 oclock position
        endX,endY = lineAngleLengthEndpoint(x,y,angle,hand[1])  #Find where the end point from center wil be located
        draw.line((x,y,endX,endY), width = hand[3], fill = hand[0]) # draw line to represent type of hand , color, and width of hand
        
    original.show()  # show image.
except:
    print ("unable to load image")

FormerLurker, 2021-07-22:
I'm thinking about adding an event to Octolapse, or something similar.  Could be used for any effect.  I can provide the images and metadata necessary.  I would do this before rendering the video in the preprocessing phase.

bheiland, 2021-07-23:
[image]

I placed code for drawing a basic clock in the section of code that generates the preview just to see how it would look .  Trying to determine how to make the clock update the time from the image metadata and draw on the actual time-lapse.

bheiland, 2021-07-26:
video of analog clock and graphic of analog clock.
https://www.youtube.com/watch?v=K898IISu-TA

FormerLurker, 2021-07-26:
Really nice!  Hoping I have some time to work on a universal integration for your code!

bheiland, 2021-07-26:
this is just a proof of concept.  I sorta shoe horned it into your code.  Your code is a bit more advanced than my understanding of it all so I struggled getting it in.   Didn't work the first few ways I tried to cleanly put it in.