FormerLurker/Octolapse #587: [Feature Request]: Disable "Snapshot Plan preview popup" also if you use a "Smart Triggers"

Metus88, 2020-09-01:
#### If this is a feature request describe it here

I would like to have a check box in the setting of Octolapse to disable the popup when you start a printjob that show you "Snapshot Plan preview". I read that it show itself only if you use a "Smart Triggers".
I ask it because I would use other plugin that they run print job and I'm afraid that this popup can interferate. 

#### Version of Octolapse
 
Octolapse Version: ___REPLACE_THIS__OCTOLAPSE_VERSION_GOES_HERE

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-09-02:
You can already do this.  Check the main settings (gear icom on the octolapse tab).  You can even set an auto dismiss timer.

Metus88, 2020-09-02:
> You can already do this. Check the main settings (gear icom on the octolapse tab). You can even set an auto dismiss timer.

You are right! In the main setting there is "Snapshot Plan Preview Preview Snapshot Plans Enabled "

Sorry I did not see!
My mistake.
Thanks again

closed 2020-09-02 by Metus88

FormerLurker, 2020-09-02:
No problem, glad it is doing what you need!