FormerLurker/Octolapse #588: Plugin creates a stack trace on CLI backup

cp2004, 2020-09-11:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: Unsure, likely latest

#### Version of OctoPrint

OctoPrint Version: 1.4.0+

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

Users are running my Python 3 Upgrade script and (*mildly*) panicking when the see a stack trace.

It has **no functional impact and the backup appears to be intact & work fine.**

See also https://github.com/cp2004/Octoprint-Upgrade-To-Py3/issues/6 
https://community.octoprint.org/t/upgrade-your-octoprint-install-to-python-3/23973/36?u=charlie_powell

The stack trace appears to be:
```python
2020-09-10 13:11:47,024 - octolapse.__init__ - INFO - Release mode detected.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1676, in shutdown
    h.close()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/logging/handlers.py", line 32, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
```

Hopefully this helps.

#### What should have happened?
No stack trace, no panic

#### What happened instead?
Stack trace (see above)

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi, (and others)
Os Version: 0.17

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  Don't have it

#### Link to octoprint.log

Link to octoprint.log:  Don't have one

.

On a side note, that template was rather hard to fill in with such a specific issue, especially when it is something reported to me that I'm passing on. Anyway, if you want to discuss, then I will be happy to explain further.

FormerLurker, 2020-09-11:
Interesting that it is the console logger that is failing.  I will look into it.

cp2004, 2020-09-11:
I forgot to include an easier way to reproduce, its been a hectic 2 days and I'm starting to fall apart!

You can run `octoprint plugins backup:backup --exculde uploads --exclude timelapses`

FormerLurker, 2020-09-11:
I saw that post on the octoprint aite yesterday, but haven't yet had a chance to provide.  Thanks for giving me a way to reproduce, that will be super helpful.

wlhlm, 2020-12-10:
I'm hitting the same exception. Easiest way to reproduce is to run the `octoprint plugins --help` command:
```
$ /opt/octoprint-venv/bin/python -m octoprint plugins --help
Initializing settings & plugin subsystem...
Initializing GcodePositionProcessor V1.0.1 - Copyright (C) 2019  Brad Hochgesang...Python 3+ Detected...complete
2020-12-10 07:45:16,054 - octolapse.__init__ - INFO - Release mode detected.
Usage: octoprint plugins [OPTIONS] COMMAND [ARGS]...

  Additional commands provided by plugins.

Options:
  --help  Show this message and exit.

Commands:
  backup:backup          Creates a new backup.
  backup:restore         Restores an existing backup from the backup zip...
  softwareupdate:check   Check for updates.
  softwareupdate:update  Apply updates.
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.8/logging/__init__.py", line 2115, in shutdown
    h.close()
  File "/opt/octoprint-venv/lib/python3.8/site-packages/OctoPrint-1.5.1-py3.8.egg/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
```

- python 3.8.6
- octoprint 1.5.1
- octolapse 0.4.1

Pa1nstar, 2021-05-18:
Hi,

i've got the same error when i'm trying to `octoprint plugins backup:backup --path /tmp/backup --exclude timelapse`.

This is the Errormessage:
```
root@octopi:/home/pi/oprint/bin# ./octoprint plugins backup:backup --path /tmp/backup --exclude timelapse
#Initializing settings & plugin subsystem...
Initializing GcodePositionProcessor V1.0.1 - Copyright (C) 2019  Brad Hochgesang...Python 3+ Detected...complete
2021-05-18 14:13:07,860 - octolapse.__init__ - INFO - Release mode detected.
Creating backup at backup, please wait...
Done.
Backup located at /tmp/backup
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3.7/logging/__init__.py", line 2039, in shutdown
    h.close()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/logging/handlers.py", line 31, in close
    self._executor.shutdown(wait=True)
AttributeError: 'OctolapseConsoleHandler' object has no attribute '_executor'
```

Any news on the problem itself? Are you working on possible Hotfix? :) 👍 

Thanks for your support! :)

cp2004, 2021-05-18:
This particular error doesn't actually make any difference to the backup itself. The issue is because OctoPrint doesn't fire up the entire system to create a backup, and in this case there's a reference to something that's not been created. Nothing significant to worry about so a hotfix is not really necessary in this case.

Pa1nstar, 2021-05-18:
Hi,

thank you for the fast answer and the explanation. I'm really sorry, i hadn't looked in the Backup Path, you're right, the file ist there... 🙈 

I wanted to automate my backup via cron and stumbled across the error. Sorry again. :)