FormerLurker/Octolapse #589: Failed to Start

cru001, 2020-09-11:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Installed Octolapse plugin.
2. Correctly followed the installation guide/youtube video.
3. Started print from local

#### What should have happened?
Test print should have started

#### What happened instead?
Nothing happened. Terminal tabs shows lines I do not understand.

#### Operating System running OctoPrint and Octolapse

OS Name: MacOS
Os Version: Catalina 10.15.6

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Marlin Bugfix 2.0

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: MacOS

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: 
[xyzCalibration_cube.gcode.zip](https://github.com/FormerLurker/Octolapse/files/5210946/xyzCalibration_cube.gcode.zip)

FYI, I printed this same Gcode with Octolapse disabled and it printed fine. 

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.json.zip](https://github.com/FormerLurker/Octolapse/files/5210950/settings.json.zip)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log: 
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/5210952/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  [octoprint.log](https://github.com/FormerLurker/Octolapse/files/5210958/octoprint.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-08:
Sorry I didn't' get around to this earlier!

I'd like to confirm that you are running OctoPrint from MacOs (not a raspberry pi).  If so, this is a known issue, and has something to do with compiling the snapshot plan c++ for Macs.  I do not have access to a Mac, unfortunately, but have been working on figuring out what's going on here.  For now I recommend that you use one of the 'classic' triggers.  They should work (let me know if they don't).

FormerLurker, 2020-11-17:
@cru001 , can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped.  Might take a while to show up in the plugin manager.  Reopen a new issue if you continue to have problems.  Thanks!

closed 2020-12-05 by FormerLurker