FormerLurker/Octolapse #591: [Request] add Date and Time token values

spanzetta, 2020-09-14:
Currently the token values for Overlay Text Template includes a "single" {current_time} value which formats in this way
yyyy-mm-dd hh:mm:ss

I kindly ask you to create a separated **Date** and **Time** token .. maybe with various formatting:

{current_time_1} = hh(12):mm:ss
{current_time_2} = hh(24):mm:ss
{current_date_1} = dd-mm-yyyy
{current_date_2} = mm-dd-yyyy
{current_date_3} = yyyy-mm-dd

What do you think about?

Thanks a lot

FormerLurker, 2020-09-17:
I've thought about doing this actually, perhaps adding a format string to the token instead of creating several new ones?  Maybe something like this:

{current_time:"%m/%d/%Y, %H:%M:%S"}

would yield something like this:

12/24/2018, 04:59:31

[Here is a reference](https://strftime.org/) for the strftime replacement tokens.  This will be a bit tricky to parse giving the existing code, but not too bad.

spanzetta, 2020-09-18:
It's fine.. once we have all variables available we can compose as needed.. 
Documentation is clear.. so no worry about complexity 
Pls advise when I could do some test on it (beta to test?)
Thanks

spanzetta, 2020-09-26:
Hi @FormerLurker 
do you have some news on this?
Thanks in advance

FormerLurker, 2020-09-28:
Not sure when I'll get to this, sorry.  I will keep this open and will post here when I've pushed to the dev branch.

The last/next few months have been just crazy, and I'm about to release a new plugin.  I expect to be pretty much confined to my home throughout the winter, so I should start to have some time for the next version of Octolapse.

spanzetta, 2020-09-29:
Thanks a lot of the update!
Have a nice day
Bye

spanzetta, 2020-10-18:
Any progress about this?
Thanks in advance

FormerLurker, 2020-10-18:
I'm embarrassed to say I have yet to make any progress on this.  I'm still finishing up ArcWelder, but am getting very close to a release.  Thank you for your patience!

spanzetta, 2020-10-18:
sorry... I didn't want to push.. 
I understand perfectly that you may have so many things to do..
I was just asking..
thanks

FormerLurker, 2020-10-18:
lol, feel free to keep pushing.  That's how things get done :)

carcamerarig, 2020-10-18:
> Not sure when I'll get to this, sorry.  I will keep this open and will post here when I've pushed to the dev branch.
>
>
>
> The last/next few months have been just crazy, and I'm about to release a new [...]

Anything interesting, I'm intrigued

FormerLurker, 2020-10-18:
@carcamerarig, it's a new plugin called Arc Welder that compresses G1 and G2 calls to G2/G3 calls (arcs).  It's designed to reduce serial communication, thus reducing stuttering, with a side benefit of having smaller gcode files (usually between 30-80% compression).

You can check it out [here](https://github.com/FormerLurker/arcwelderplugin).

There is also a console version, a C++ library, an inverse processor (convert g2/g3 to g0/g1), and a python library available [here](https://github.com/FormerLurker/arcwelderlib).

FormerLurker, 2021-01-05:
@carcamerarig and @spanzetta, I wanted you to know I'm working on this now.  Better late than never?  lol

spanzetta, 2021-01-05:
Great!..
Waiting your news..
;-)

FormerLurker, 2021-01-05:
This was tricky, but I got the date and elapsed time both working, including custom formatting tokens.  You can install via the plugin manager using this URL in the *from URL...* box:

```https://github.com/FormerLurker/Octolapse/archive/07a19ee284977ff0b4ac27432f731bbb3da9ad79.zip```

Please click on the help icons next to the new tokens for usage instructions, and be sure to let me know if the docs need work (specific examples would be welcome).  I'd also recommend running a couple test mode prints using as many variants of the formats as you can think of.  This will help me test a lot.  If you would, let me know what version of python and OctoPrint you are using (I'm currently testing on OctoPrint 1.5.2 Python 3.7.3 OctoPi 0.17.0).

Thanks!

FormerLurker, 2021-01-05:
FYI, here is where you can find the popup help:

[image]

It might not be visible enough.  I will think about this.  Suggestions are welcome.