FormerLurker/Octolapse #592: Unable To Enable Custom Preferences

calebmah, 2020-09-14:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1. Followed the [documentation here](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera) about setting up a raspberry pi camera up to step 7
2. In step 8 I tried to enable custom image preferences by ticking the checkbox

Hi, I set up a new camera profile for a raspberry pi camera v2.1 that I just installed, but was unable to enable the custom image preferences. The snapshot and test buttons work fine, its just enabling of the custom preferences. What confuses me is that this definitely worked before on a usb webcam that I was using. On the first camera profile with the usb camera, I enabled custom image preferences and was able to change the settings. Now, I am unable to enable the custom image preferences for the new raspberry pi camera. The original camera profile still has it enabled, but when I click the pencil icon to edit that profile I get the same error as the pi camera profile, so I am not able to adjust the custom image preferences for any profile now. Appreciate any help with this error, thanks!

#### What should have happened?
Custom image preferences for this profile should have been enabled, showing the 3 options when they should be applied

#### What happened instead?
Nothing happened, and after a few seconds an error popup in the top right corner appeared with
```
Unable To Enable Custom Preferences
An unexpected error was raised while testing custom image preferences for the 'Raspberry Pi Camera' camera profile.
```

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: None

#### Link to settings.json

Link to settings.json with all passwords removed: https://pastebin.com/2ajnkXgt

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/cYUmqWab
```
2020-09-14 19:38:58,911 - octolapse.camera - ERROR - An unexpected error was raised while testing custom image preferences for the 'Raspberry Pi Camera' camera profile.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/camera.py", line 426, in _test_mjpg_streamer_control
    r = utility.requests_retry_session(session=s).request("GET", url, timeout=timeout_seconds)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
ConnectionError: HTTPConnectionPool(host='127.0.0.1', port=8080): Max retries exceeded with url: /?action=command&id=-1 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x64801d50>: Failed to establish a new connection: [Errno 111] Connection refused',))
```

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/X1aJKcHq

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: 

[image]

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-09-17:
Are you running octoprint anywhere or the spaghetti detective my chance?  If not, can you look at:

http://{ip_of_pi}:8080/control.htm

and see if anything loads (send a screenshot of what comes up).  If that doesn't work, try this url:

https://{ip_of_pi}/webcam/control.htm

Let me know what you find!  Looking forward to solving this.

calebmah, 2020-09-18:
So it turns out that this was entirely my fault in setting up the camera incorrectly. For the benefit of anyone who may face the same issue as me, I had incorrectly set `camera="raspi"` and used `camera_raspi_options` instead of `camera_usb_options`. When I followed [the documentation](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera), I wrongly assumed that there was a typo in using `camera="usb"` since it looked exactly the same as the wiki page for setting up a usb webcam and I thought it was accidentally copied over. After all, it is a raspi camera and this setting worked for the normal webcam, just not for enabling custom image preferences. Don't be like me everyone, follow the instructions!

After watching your youtube video explaining the process I realized where I had gone and fixed it, and everything is working fine now! It didn't help that there were a couple of broken links in the wiki, like this [troubleshooting guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting.md#i-cant-enable-custom-image-preferences) that was supposed to help with exactly this issue. But anyhow, thank you so much for trying to troubleshoot and of course for developing this, its really helpful and much appreciated!

closed 2020-09-18 by calebmah

FormerLurker, 2020-09-18:
Hey, can you take a screen cap and highlight the broken link for me?  I'd like to get that fixed.  I wrote the entire wiki for 0.4 (including separating out the old docs) in a couple day mad rush, so I'm not surprised there are issues.  The markdown editor I use has a different format for relative links than github uses, so it's been quite a pain to get the links right :)

Glad you got things working!  The usb thing is pretty odd, and is just due to the nomenclature used by OctoPrint.  USB really means NOT raspicam since the driver I am using didn't used to work for this sort of thing.  I totally understand how you could have been confused.  I will add a big bold message to the wiki about this if I can remember to do so.  If you want to create an issue for that and the broken link I would appreciate it very much.  I am pretty much limited to mobile today since I threw my back out :(

calebmah, 2020-09-28:
I created a new issue to track and document the missing links, hope it helps you! https://github.com/FormerLurker/Octolapse/issues/602