FormerLurker/Octolapse #595: "Building wheel for Octolapse (setup.py): still running..." Eternal Loop!

ProgramFreakHD, 2020-09-19:
I just love how this piece of shit plugin dosen't want to download! after 7 days of just trying to google and experimenting for it to download I lost all hope. I,m running the latest 1.4.2 on a PI0W, no luck, keep getting "Building wheel for Octolapse (setup.py): still running...", I tried on a RPI B+, guess what... same garbage eternal loop of "Building wheel for Octolapse (setup.py): still running...". i tried to downloade to 1.4.1 and 1.4.0, no luck either.
I want to know why dose everyone else except me can just download this plugin with no errors but I can't. I watched a million videos on HOW TO DOWNLOAD OCTOLAPSE!, and in their videos, It's easy as one, two, three, BOOM, NOW YOU HAVE OCTOLAPSE!
[image]

FormerLurker, 2020-09-19:
Not sure what is going on because I have never seen this before.  You should fill out the issue template if you want to give me a fighting chance to debugt your issue.  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue) for more info on how to report an error.

ProgramFreakHD, 2020-09-27:
Ok. Sorry for the long delay of responding, I had no time filling your issue template frankly so I'm just going to share my experimentations i did with minely success. I've managed to install octolapse on PI3B+ and PI0W. First of all the pip that is pre-installed on octopie dosen't exsis, secondly the python that is also pre-installed is outdated.

Here is step by step how i managed to install it.
First: I made a fresh octopie image on my sd and installed it on my raspberry pi, the defult version is **Octoprint 1.3.12**, once you have your pi settup and running **Octoprint 1.3.12** you have to go to plug-in manager and download octolapse, for me I had to wait 10min and at the end it gave me an error.
second: I opened pytty and SSH into the pi, in the terminal I had to upgrade to version 1.4.1, to do that I used **~/oprint/bin/pip install https://github.com/OctoPrint/OctoPrint/archive/1.4.1.zip**, once it's downloaded i had to install pip and python next, to do that I used 
**curl https://bootstrap.pypa.io/get-pip.py -o get-pip.py**,
**python get-pip.py**,
**python -m pip install -U pip**,
**python -m pip install downloads/pip-20.2.3-py2.py3-none-any.whl**
those commands I had to run one after another. After I executed and installed all of those i had to restart the octoprint server, to do that i typed **sudo service octoprint restart**. after it installed with some luck I had the octolapse installed and running.