FormerLurker/Octolapse #596: Unable to save printer profile when PrettyGCode is installed

Trevor-Miller, 2020-09-21:
#### If this is a feature request describe it here

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred

1. Install [PrettyGCode Plugin](https://github.com/Kragrathea/OctoPrint-PrettyGCode)
2. Restart server so PrettyGCode is compiled and running
3. Install Octolapse Plugin (entirely fresh install)
4. Add a Printer Profile and Save

#### What should have happened?
The Printer Profile should save, the modal should close and the printer profile should be visible in the Printer Profiles list. 
**IMPORTANT NOTE:** This issue does not occur if PrettyGCode is removed.

#### What happened instead?
UI did not respond to save button press, and console logs the following error:
`Uncaught TypeError: this.charCodeAt is not a function.  Exception occurred when checking element octolapse_printer_snapshot_command, check the 'octolapsePrinterSnapshotCommand' method.
    at PrettyGCodeViewModel.String.hashCode (VM3996 packed_plugins.js:20149)
    at i (VM3998 packed_libs.js:1)
    at jt (VM3998 packed_libs.js:1)
    at Function.w.param (VM3998 packed_libs.js:1)
    at a.validator.remote (VM3996 packed_plugins.js:6129)
    at a.validator.<anonymous> (VM3996 packed_plugins.js:6883)
    at a.validator.check (VM3996 packed_plugins.js:6129)
    at a.validator.checkForm (VM3996 packed_plugins.js:6129)
    at a.validator.form (VM3996 packed_plugins.js:6129)
    at w.fn.init.valid (VM3996 packed_plugins.js:6129)`

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Ender 3 Pro
Printer Firmware Version: Marlin V 0.0.6

#### Browser and version of browser, operating system running browser

Browser: Google Chrome Version 79.0.3945.130 (Official Build) (64-bit)
Browser OS: Windows 10 (64-Bit)

#### Link to the gcode file you were printing when the problem occurred

#### Link to settings.json

#### Link to plugin_octolapse.log

#### Link to octoprint.log

#### Link to contents of Javascript console in the browser

Link to javascript console output: https://pastebin.com/9NZSjixL

#### Screenshots and/or videos of the problem:

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-09-23:
Hmm..  The error looks to be coming from the PrettyGCodeViewModel.String.hashCode method, but I will review my code as well.  That's just a 'first glance' idea though, and could be wrong.  Thanks for reporting this!

apbarratt, 2021-05-12:
I've just had this same issue.  I searched for the error in the console and this page came up.  After disabling prettyGcode and restarting, I was able to add a profile to Octolapse.  This is my first time setting up octolapse.

joshwegener, 2022-04-27:
Having the same issue.

FormerLurker, 2022-04-27:
I have added PrettyGcode to my local dev environment, and will take a closer look today.  Will post back as soon as I have a definitive solution.  I love that viewer, and want it to be compatible!