FormerLurker/Octolapse #597: Add ffmpeg custom parameter for better Apple devices / Quicktime compatibility

tomav, 2020-09-21:
The rendered video does not play smoothly on Quicktime / Apple devices (iMac).
There might be a missing ffmpeg parameter like `-pix_fmt yuv420p`. (already had this kind of issue on another ffmpeg related project).

More information from ffmpeg:
https://trac.ffmpeg.org/wiki/Encode/H.264#Encodingfordumbplayers

I can help on this but I was wondering if there a way to add/inject an extra parameter to a rendering profile using the UI or in the JSON template. This could be useful for this kind of issue or anything related to ffmpeg/encoding.

#### Version of Octolapse
Octolapse Version: 0.4.0

#### Version of OctoPrint
OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: No

#### What were you doing when the problem occurred
Using profile "MP4 - 30fps", the rendered file does not play smooth on a Mac using Quicktime.

#### What should have happened?
Video should run smooth.

#### What happened instead?
Video was not smooth.

#### Operating System running OctoPrint and Octolapse
OS Name: Linux octopi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
Os Version: Linux octopi 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux

#### Browser and version of browser, operating system running browser
Browser: Google Chrome
Browser OS: Version 85.0.4183.83 (Official Build) (64-bit)

#### Screenshots and/or videos of the problem:
Screenshot/Video Links: https://drive.google.com/drive/folders/1C6gUcAOxsLjQ_yRaAWEidq9AFatxy67o?usp=sharing
2 videos:
- Rendered video by Octolaps: DM_Parts_Tray_Drawer_4_20200921101406.mp4
- Fixed version: DM_Parts_Tray_Drawer_4_20200921101406_fixed.mp4

To be read on a Mac with Quicktime.

tomav, 2020-09-21:
Looking at the code, it seems `-pix_fmt yuv420p` is already added (will use debug log to check if there's any confirmation with the ffmpeg command played). Could be related to another parameter like the codec MP4/libxvid vs. H264.

Will update later this week but let me know if you have any clue.

FormerLurker, 2020-09-23:
Are you already using H264?  If so, you may want to make sure you are using ffmpeg and that it is up-to-date.  There are some issues with 264 ATM, and 265 won't work on the pi (not enough memory to encode).

Also, both images play just fine on my PC (even in the browser).  I wonder why it's not working on Mac/IPhone.  I figured Apple had good QC for this kind of thing.  I do NOT own any apple devices yet (too $$$ for me ATM), so it's extremely difficult to debug/fix these sorts of things.  However, I've managed it in the past, so it's possible I can figure this out too.

Regarding the custom parameters, I'd LOVE to add those, but it renders many of the rest of the rendering profile settings moot.  I have thought about this and would love to implement it somehow without making it too difficult to configure (the parameters are quite messy and all interconnected).  

You could help me debug if you can figure out what is going wrong with the ffmpeg/avconv parameters.  You can enable rendering logging and see the full command parameter set in the octolapse log file (see the logging profile and add the rendering logger set to verbose output).  That may be useful since you could also enable snapshot archiving like so:

[image]

This will give you access to a zip of all the images after the print and rendering complete:

[image]

Next, you can run ffmpeg manually and tweak the parameters to see if you can figure out what's going on.  I can easily add mac specific code too, in case the parameters are different.  Let me know if you can try this and what you find.