FormerLurker/Octolapse #598: Snapshot triggers when layer is not completely finished

buildcomics, 2020-09-23:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Attached a DSLR through a rest API
2. Created scripts etcetera to take pictures
3. Created test prints on prusa mini with firmware 4.2.1-final+2072
4. Test prints got very stringy
5. Changed retraction length all the way up to 10, z-hop to 0.5. 
6. Added custom G-code to layer change (G4 P1) in prusa slicer
7. Set this up in octolapse settings 

#### What should have happened?
A picture at every full layer change

#### What happened instead?
Pictures at half/partially done layers. And also a lot of stringing still

#### Operating System running OctoPrint and Octolapse

OS Name: Arch LInux
Os Version: 5.5.13-arch1-1

#### Printer model & used firmware incl. version

Printer Model: ___Prusa Mini
Printer Firmware Version: ___4.2.1-final+2072

#### Browser and version of browser, operating system running browser

Browser: Firefox
Browser OS: Manjaro Linux

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [https://gist.github.com/buildcomics/c5416c0dc4b9568dfeb55b27ce373e08#file-cube_0-2mm_pla_mini_38m-gcode](url)

#### Link to settings.json

Link to settings.json with all passwords removed: [https://gist.github.com/buildcomics/c5416c0dc4b9568dfeb55b27ce373e08#file-settings-json](url)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [https://gist.github.com/buildcomics/c5416c0dc4b9568dfeb55b27ce373e08#file-plugin_octolapse-log](url)
Note: I had to cancel the print early because the cube came off the buildplate, I hadn't set the z-height correctly for this one.
Note2: The camera didn't start for a couple of layers, because the battery was empty. I replaced it after the first couple of layers, but it didn't seem to bother octoprint it only got the same image all the time :-).

#### Link to octoprint.log

Link to octoprint.log:  [https://gist.github.com/buildcomics/c5416c0dc4b9568dfeb55b27ce373e08#file-octoprint-log](url)
Note: I had to cancel the print early because the cube came off the buildplate, I hadn't set the z-height correctly for this one.

#### Link to contents of Javascript console in the browser

Link to javascript console output: Not Applicable

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: Here is a screengrab of an earlier video I took, which explained the 
[image]
This is a screenshot from an earlier video, showing clearly that the first layer wasn't finished when the screenshot was taken. 
Here is a video of a couple of layers later: https://youtu.be/lrTdWTZrWNI

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

I do like this project, I send you some beer money on Paypal just now ;-).

FormerLurker, 2020-09-23:
This is by design for the smart triggers, which are designed to minimize travel distance (and print time) and maximize print quality.  This was the #1 complaint in the old version of Octolapse, so it's a super important feature.  

If you'd prefer the old method of taking snapshots at the beginning or end of each layer exclusively, check out the 'Classic' layer/height or any of the Gcode triggers.  I recommend the 'Smart' gcode trigger, as you will get a nice snapshot plan preview and error checking.  Let me know if you need any help setting this up.

FormerLurker, 2020-09-23:
Oh, BTW, thanks for the donation!  They are ALWAYS appreciated.  Please don't hesitate to post again.

buildcomics, 2020-09-23:
No worries. I have set it up with the gcode snaps, and it seems to work. I still have two issues:

1- Exessive strining, I am not sure if this is expected when the snapshot takes about two seconds (see the youtube video), and if it's reven remotely possible to do this without the noozle oozing while it's steady for taking the snapshots. I've played around with the retraction settings on the mini, but no luck so far. Every time a drop of filament is oozing out during stabilization, and it's dropped on the print. It might be because the mini's bowden setup doesn't like to wait for this long, so I might be forced to move away from the DSLR to a webcam. Which would be a shame, but maybe that's the cost of having a cheaper printer without a direct drive... Anyway, tips would be appreciated here, but I don't think this is the fault of octolapse.
[image]

2 - The last layer is not captured. After the second to last layer, it doesn't capture anymore. I have to look if I messed this up in the settings somewhere. If this is to be expected, this issue can be closed.

FormerLurker, 2020-09-23:
**Item 1**

That is a lot of stringing for a mini.  See [this guide](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues) regarding quality issues.  Two seconds is a lot of time for a snapshot, and generally DSLR capture has way worse quality due to the extra time making oozing more likely.  It's worse on bowden setups compared to direct drive. [ Here is one way to reduce the capture time for DSLR](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-No-Download) by eliminating the download step, but you don't get a preview image or a rendered timelapse (images on the SD card).  [Here is another way](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-DSLR-With-Deferred-Download) to reduce the time AND get a rendered timelapse.  Both methods depend on your camera supporting the necessary functionality.

Are you using the built-in prusa mini profile?  Misconfiguration can cause a lot of issues.  Also, are you using the [automatic slicer settings config](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration), and have you turned on [verbose gcode](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Slic3r-Slic3r-PE-and-PrusaSlicer) (only important if you are using either the smart layer or smart 'snap-to-print' profile)?

So, I would recommend checking out the above guides and then trying a test with a webcam to see how much of an effect oozing is having.  If you get great quality with a webcam and poor quality with dslr, you probably need to work on your retraction/temps or change filament.  If you get poor quality with a webcam, it's probably some printer profile config issue.  Based on the image of your first layer above, I'd guess something isn't configured correctly.

**Item 2**

Hmm.  Check to make sure the snapshot command is being put at the end of the last layer.  If not, you could put it in your end gcode.  I bet this is a slicer issue.  The 'Classic' layer trigger would not have this problem.  

Honestly, if I were you I'd be using the smart trigger and just ignore that first frame.  You usually can't tell the shot was in the middle of a layer except for that one frame.  Also, I'm working to add the ability to use the Gcode trigger and the smart layer trigger at the same time (and some other similar tricks without gcode) to ensure a snap before the print starts (empty build plate) and after the print has completed (though that will be more tricky).  The If you have verbose gcode enabled in your slicer, Octolapse will take snapshots over infill if possible, which will hide most/all artifacts (provided your printer config is correct).

Let me know how it goes!

buildcomics, 2020-09-23:
Cool, I've set up smart trigger - snap to print and disabled, and the quality looks MUCH better. Not sure how, but maybe the counter pressure of the nozzle resting against the already printed parts stops the oozing. I made you a video of the result here (sorry for setting up the camera correctly in the first few frames) : https://youtu.be/dbkVGhdzx8U

There is still some artifacts on the cube:
[image]
But I am not sure if this is caused by Octolapse. I'm running the same gcode without Octolapse to see what happens.

FormerLurker, 2020-09-23:
Ok, excellent. With some effort, you should be able to achieve very good quality when running Octolapse.  Let's keep this issue open and I will continue to help if I can.  I have seen fantastic results with the mini, and I am confident you can too.

Snap to print is usually the best, mostly because the snapshots are so fast.  Retraction is automatically done by default, which will slow oozing.  Since the snapshot is so fast (no travel), there just isnt enough time for any oozing (or very little).  Similarly, using the 'center' stabilization with any layer trigger is usually faster, and any artifacts are pulled into the model instead of away (but not better than snap to print).  If you want the extruder out of the way, your best bet is to position your camera so that you don't have to move the extruder too far (customize the stabilization profile).

There are SO many ways to tweak octolapse, your printer, your slicer, and your filament, that it is difficult to say what will work best.  Hopefully the sweet timelapses are worth the effort!

buildcomics, 2020-09-23:
I have printed another cube (same g-code) without octolapse, same artifacts. This means the octolapse settings I got are now not interfering with the print, which is great :-).

Btw, I will one day or another post the python api I wrote for the camera using gphoto2 on github as well, together with the full build for my 3d printing setup. The advantage of using the Python API was mostly speed. Having the opportunity to have the camera connected and initialized before triggering the snapshot cut down on the time about two seconds I had previously. I also must note that the camera is connected to another raspberry pi (just like the printer), and I am running octoprint on a fast little server. This way I can use a raspberry pi v1 B without any performance issues.

[image]

That's enough cube's for today however, it's way passed bedtime on this side of the water.

closed 2020-09-23 by buildcomics