FormerLurker/Octolapse #599: Octolapse generated stringing during taking snapshot

alshaerahmed, 2020-09-26:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.0

#### Version of OctoPrint

OctoPrint Version: Version 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

1.caliprated 100% my ender 3 pro printer 
2.activated the cgode snap mode in octolapse
3. activated all retraction settings in camera profile in octolapse
4. still getting stringing while taking snapshot

#### What should have happened?
no stringing should have been generated

#### What happened instead?
stringing generated

#### Operating System running OctoPrint and Octolapse

OS Name: rasbian
Os Version: Octopi image 0.17 downloaded from https://octoprint.org/download/
#### Printer model & used firmware incl. version

Printer Model:Ender 3 pro
Printer Firmware Version: Stock Marlin 1.0.1

#### Browser and version of browser, operating system running browser

Browser: Firefox 
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: https://www.thingiverse.com/thing:3510948

#### Link to settings.json

Link to settings.json with all passwords removed: https://drive.google.com/file/d/1aXbFdb79k3FPaNXfBHBgMZ_6nZL2l2vL/view?usp=sharing

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://drive.google.com/file/d/1EnNZUX5Vgg1APHYM4XUv1nm04kAGJbkw/view?usp=sharing

#### Link to octoprint.log

Link to octoprint.log:  https://drive.google.com/file/d/1msWumDilpr5WXJIlCS_Bv8xOplgTE1b8/view?usp=sharing

#### Link to contents of Javascript console in the browser

Link to javascript console output: couldnot locate it

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: https://www.youtube.com/watch?v=5E4Jo22ejxs (you can see the stringing from the point the head leaves the model to go back to take photo)

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

spanzetta, 2020-09-26:
Hello @alshaerahmed 
this look similar to the issue I have reported at the end of July
https://github.com/FormerLurker/Octolapse/issues/570

See my video at https://www.youtube.com/watch?v=HU7WurIzkGc (after 00:30 sec)

alphaaGithub, 2020-11-09:
I have the same issue..... Small Blop appearing in the direction where the print head part when taking the photo for the Octolapse.  This is a massive problem.  We need to fix it.

alphaaGithub, 2020-11-09:
I Hope some one can find a solution for this problem.

FormerLurker, 2020-11-09:
Sorry I haven't commented on this yet, it slipped through the cracks.  

So, there is a ton of functionality in Octolapse to deal with this.  The first step is to run a stringing test and minimize stringing without Octolapse first.

Check out [this wiki](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues) if you haven't seen it yet.

I'll go through some of the main points in that wiki entry and will elaborate a bit.

First, your printer profile in Octolapse must be correctly configured.  Common problems are an incorrect G90/G91 influences extruder setting.  This should be true for Marlin 2.x and Smoothieware and false for RepRap and Prusa firmware.  If you are unsure what setting to use here, it will be the one that doesn't produce terrible stringing and blobs :)  There are tests you can run to determine this setting for some printers.  I'm working on a 'firmware check' button to fill that setting in correctly.  Next, you need to have the slicer settings in the printer profile set EXACTLY to what is used in your slicer.  The guide discusses enabling 'Automatic Slicer Settings' detection, which is a sure fire way to ensure all the settings are correct.

Next The first thing you will want to try if you are not already is using the 'Smart Triggers'.  However, unless certain information exists within your gcode, this may result in *WORSE* quality than using the classic triggers.  The smart triggers all try to minimize stabilization time, and if you are stabilizing away from your part, this means Octolapse might take snapshots while over exterior perimeters, which will result in artifacts like those shown in the photos above.  For Cura and Simplify 3d this is not an issue, since gcode produced from those slicers always has feature information within comment sections.  It IS a problem for PrusaSlicer and Slic3r variants.  If you are using one of these, make sure 'verbose gcode' is enabled.  This tells Octolapse what features are printing (exterior/interior perimeters, infill, bridges, etc).  Octolapse prefers to take snapshot over certain features, like ooze shields and infill, and really hates taking snapshots over exterior perimeters.  If the feature type information is not available, Octolapse tries to divine what feature is printing based on the feed rate.  This is not always accurate depending on your slicer settings.

The next thing you can do is to minimize the stabilization time.  This might mean increasing travel time, reducing snapshot acquisition time (not using a DSLR, streaming at a higher FPS, Stabilize over the part, etc..).  The longer your printer takes to complete a snapshot, the more time there will be for filament to ooze from the nozzle.

You can also avoid using Octolapse if you are printing very fine layer heights.  The thinner each layer is, the more sensitive it is to oozing.  0.2mm works pretty well and 0.05 oozes like hell in my own testing.

Finally, there are a few hardware modifications that might reduce the image acquisition time.  Use the very fastest SD memory you can get your hands on.  This will lessen the image write time substantially.  You will also find that OctoPrint will boot faster and run better.  I've tracked SEVERAL mysterious stringing/blobbing issues down to very slow SD cards (some as slow as 2MB/s!).  Class 10 SD is an absolute minimum, and can write at around 10MB/s.  That means if your snapshot is 1MB it will take 100MS to download, which is actually a long time.  Faster memory is available, but it is real world write speed that matters here.  As usual, make sure you are running a reasonably fast pi, preferably a Pi3/4, with cooling and an adequate power supply.  Not doing this will affect all aspects of octoprint negatively.

Lastly, you might want to examine your gcode.  If your printer is  struggling to print a huge number of tiny gcodes, all quality will suffer, including when using Octolapse.   You might try 'Arc Welder' if your firmware is compatible (Marlin 2.0.6 minimum recommended, Prusa Firmware needs some help, but will work too).  Octolapse should work fine with Arc Welder.

Summary:  There is a LOT you can do to improve quality with Octolapse.  I'm at the point where it's very difficult to tell prints with Octolapse from prints without.  I spent over a year working only of improving quality with Octolapse, but there is no magic bullet.  Things can be improved, but there will always be some impact.  It can only be minimized.

Please let me know if any of that helped at all.