FormerLurker/Octolapse #600: (Large GCode file) Preview popup delay larger than countdown timer - preview disappears before being shown

war4peace, 2020-09-27:
#### If this is a feature request describe it here

No new feature requested

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. I have a GCode with over 4 million lines, using Octolapse with timed preview window with 10 seconds timer until auto-accept. The reason for the timed window is I sometimes start a print from my mobile phone using another app such as Printoid.
2. With the GCode being so large, after scanning the whole file, there is a delay between scanning being finished and Octolapse preview window showing up. It seems that the Octolapse timer countdown starts when scanning is finished but before the preview popup. The delay is incidentally just as large as the countdown timer, making it impossible to use the Octolapse preview functionality

#### What should have happened?
The Octolapse preview timer should start when the pop-up window is displayed, instead of when scanning finished.

#### What happened instead?

The Octolapse preview timer started when scanning finished, instead of when the pop-up window is displayed.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version:  0.17.0

#### Printer model & used firmware incl. version

Printer Model: Creality CR 10 MAX
Printer Firmware Version: Marlin 2.0.5 modified by TM3D

#### Browser and version of browser, operating system running browser

Browser: Chrome 85
Browser OS: Windows 10 X64 1909

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: Any large GCode file with more than 1.5M lines exhibit the issue, however it seems that the 10-second delay is reached at about 4M lines of code, depending on CPU speed.

#### Link to settings.json

Link to settings.json with all passwords removed: Not applicable

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  Not applicable

#### Link to octoprint.log

Link to octoprint.log:  Not applicable

#### Link to contents of Javascript console in the browser

Link to javascript console output: I unfortunately closed the web interface after starting the print, and the print takes over 24h to finish, so that's gone.

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: Not applicable. I can capture video of the behavior using OBS but the issue can easily be replicated based on the information above.

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-09-27:
Interesting.  My guess is there is some delay sending the snapshot plan to the browser.  I will take a look.

Regarding your use case (starting a print remotely), that is exactly why i added the countdown!  Awesome

FormerLurker, 2020-09-27:
Oh, if you can link your gcode file, that may be helpful.

war4peace, 2020-09-27:
Sure thing, here's the code:
https://drive.google.com/file/d/1HAqCpv3XfW3XcItcV_d1V39Ce6N3FlLl/view?usp=sharing
FYI the pop-up does show up for small GCodes, but in this case the delay is so large, the pop-up barely shows, or not at all.

FormerLurker, 2020-09-27:
Thanks!  I believe I understand the problem.  The snapshot plan is likely taking a while to download.  As a workaround, I suggest increasing to maybe 20 seconds while I figure out a solution.

war4peace, 2020-09-27:
Thank you, no rush, I was considering the workaround but thought you might want to know about it. Maybe start the counter when plan finishes downloading instead of when it starts, if possible, of course? Not sure if possible, just throwing ideas around.

FormerLurker, 2020-09-27:
Nice idea, and yes I am glad you reported it.  However, if you start with printoid or another remote app, no browser will be connected to receive the plan.  I may have to check for connected clients, and wait for a reply with some timeout IF one is connected.  Oh the joys of development, lol!

war4peace, 2020-09-27:
That's right, I haven't thought about it. I usually have my web browser open all the time on my main PC, but the printer is in another room (and floor), so sometimes I am next to the printer and initiate a print through Printoid/OctoRemote, so the plan downloads upstairs anyway.
But I understand what you are saying. Maybe add a 2-second delay for each million lines of code? OctoLapse is parsing the code anyway, so getting a counter out of it is easy. On the other hand it might confuse customers.