FormerLurker/Octolapse #602: Update raspi camera documentation and fix broken links in wiki

calebmah, 2020-09-28:
#### If this is a feature request describe it here

This is a follow up issue to issue https://github.com/FormerLurker/Octolapse/issues/592. It affects [this wiki page](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera) on setting up the raspberry pi camera. Hope it helps!

1. Step numbering skips from step 1 to step 3
[image]

2. Instructions for changing `octopi.txt` to `camera="usb"` should be clearer and more explicit that it should NOT be `camera="raspi"` like you might expect.
[image]

3. Broken links
[SSH guide in step 1](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Connecting-To-Terminal-Remotely.md)
[image]

[Snapshot troubleshooting guide in step 7](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting.md#my-camera-fails-the-snapshot-address-template-test)
[image]

[Custom image preference troubleshooting guide in step 8](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting.md#i-cant-enable-custom-image-preferences)
[image]

[Camera not loading troubleshooting guide in step 8](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting.md#i-can-enable-custom-image-preferences-but-my-camera-stream-will-not-load)
[image]

#### Version of Octolapse
 
Octolapse Version: NA

#### Version of OctoPrint

OctoPrint Version: NA

FormerLurker, 2020-09-28:
Thanks!  My markdown editor (PyCharm) does NOT correctly handle github links, so I need to pretty much code them manually and check them all.  This should help fix a few that I didn't catch.

calebmah, 2020-10-02:
I'd be happy to try help update the links for you if you don't have time, the problem is that I can't actually find the parts of the wiki they are supposed to be referencing. If you are able to provide the relevant links to the right pages, I can try give it a go!