FormerLurker/Octolapse #603: Octolapse connect to DSLR camera but give Error

Moisuki, 2020-10-01:
**Camera model**
Panasonic Lumix GX80 (DSLR)

**What is the problem?**
Give me Error but it connect and work (so the Problem is that I can´t start a print because the Error message)

**I use 2 Scripts**

**This to connect to the Camera** ( I enter that by Octolapse -> Cameras -> Before Print Start Script)

#!/bin/sh
curl -X Get -G http://192.168.54.1//cam.cgi?mode=setsetting
-d type=device_name
-d value=SM-G903F \

sleep 1

curl -X Get -G http://192.168.54.1//cam.cgi?mode=accctrl
-d type=req_acc
-d value=4D454930-0100-1000-8001-024500021C98
-d value2=SM-G903 \

sleep 1

curl -X Get -G http://192.168.54.1//cam.cgi?mode=setsetting
-d type=device_name
-d value=SM-G903F \

**And this to make a photo** ( I enter that by Octolapse -> Cameras -> Before Snapshot Script)

#!/bin/sh
curl -X Get -G http://192.168.54.1/cam.cgi?mode=camcmd
-d value=capture \

Both work it connect and make a Picture when I click on "Test Script".
It take a Photo but give me the Error

**Additional information about your setup**
Octoprint 1.4.2
Octolapse v0.4.0
Raspberry PI 3 B+

**Log:**

2020-09-29 21:23:19,060 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Print Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   7090      0 --:--:-- --:--:-- --:--:--  8666
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    29    0    29    0     0   4833      0 --:--:-- --:--:-- --:--:--  5800
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0  15600      0 --:--:-- --:--:-- --:--:-- 19500
2020-09-29 21:23:21,488 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Snapshot Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   6000      0 --:--:-- --:--:-- --:--:--  6000
2020-09-29 21:30:04,852 - octolapse.__init__ - INFO - Started logging to file.
2020-09-29 21:30:04,859 - octolapse.settings - INFO - Loading existing settings file from: /home/pi/.octoprint/data/octolapse/settings.json.
2020-09-29 21:30:04,986 - octolapse.settings - INFO - Creating settings from iterable.
2020-09-29 21:30:05,208 - octolapse.settings - INFO - Settings created from iterable.
2020-09-29 21:30:05,209 - octolapse.settings - INFO - Settings file loaded.
2020-09-29 21:31:27,962 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Print Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   6500      0 --:--:-- --:--:-- --:--:--  6500
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    29    0    29    0     0   1933      0 --:--:-- --:--:-- --:--:--  1933
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   4875      0 --:--:-- --:--:-- --:--:--  4875
2020-09-29 21:31:40,125 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Snapshot Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0  13000      0 --:--:-- --:--:-- --:--:-- 13000
2020-09-29 21:31:42,986 - octolapse.script - ERROR - Error output (stderr) for 'DSLR - Before Snapshot Camera Script':
	  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
	                                 Dload  Upload   Total   Spent    Left  Speed
	
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100    78    0    78    0     0   7090      0 --:--:-- --:--:-- --:--:--  7800

FormerLurker, 2020-10-08:
I'm not sure why, but the output you see is coming from the stderr stream.  This is reserved for errors.

Also, why are you using the before snapshot script and not this:
[image]

I'll see if I can replicate your error, but I unfortunately do not have a DSLR that can be activated via HTTP.  I may be able to simulate it somehow.

Moisuki, 2020-10-11:
Or its there any easier way to connect the DSLR to Octolapse.
I tryed with GPhoto but the GX80 is not supported.