FormerLurker/Octolapse #604: Overlay problem in Python 3.8.2

teddylarsen, 2020-10-07:
Line 658 in utility.py needs to be changed to something like

linux_font_paths = str(subprocess.check_output(["fc-list", "--format", "%{file}\\n"]), 'UTF-8').split('\n')

for the list of fonts to be correctly read when running Octolapse under Python 3.8.2.
Otherwise an error message about a missing font-config is shown.

FormerLurker, 2020-10-07:
Thanks for pointing it out.  Do you know if that edit works in previous versions of python?  I will still test, but it would be good to know.

teddylarsen, 2020-10-07:
Unfortunately I don't know if it does. I've only tested on Ubuntu 20.04 with Python 3.8.2.

Also, I'm afraid I may have been a bit too quick in posting as the change only seems to be working on the preview page and not in actual timelapses. Will do some more testing and report back.

FormerLurker, 2020-10-07:
Do you have fontconfig installed?

teddylarsen, 2020-10-07:
I have fontconfig version 2.13.1 installed.
My test print just finished and while the overlay data is not shown in the temporary snapshots, it **is** shown in the final timelapse. So I believe it is only a matter of testing whether the proposed solution works in previous versions of Python as well.

FormerLurker, 2020-10-07:
OK, thanks!  Hopefully I'll have some time to implement this over the next couple weeks.

ahnm, 2020-10-21:
With regards to this fix, the variable `linux_font_paths` had an empty string appended to the end which results in the same "font-config error". Trimming this resolved the issue for me.

FormerLurker, 2020-10-21:
@ahnm, nice.  That should work!

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker