FormerLurker/Octolapse #606: Octolapse Error: Camera Script Failed Errors were detected - The 'RP - Snapshot Camera Script' process returned errors. See plugin_octolapse.log for details

decod3dotcom, 2020-10-10:
#### Version of Octolapse 
Octolapse Version: Octolapse 0.4.0

#### Version of OctoPrint
OctoPrint Version: OctoPrint 1.4.2 
Python 2.7.16 

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: 
2020-10-10 01:36:39,810 - octolapse.script - ERROR - The 'RP - Snapshot Camera Script' process has timed out before completing.  Attempting to kill the process.
2020-10-10 01:36:39,825 - octolapse.script - ERROR - An error occurred while killing the 'RP - Snapshot Camera Script' process.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/script.py", line 103, in kill
    proc.kill()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/psutil/__init__.py", line 289, in wrapper
    return fun(self, *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/psutil/__init__.py", line 1259, in kill
    self._send_signal(signal.SIGKILL)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/psutil/__init__.py", line 1206, in _send_signal
    raise AccessDenied(self.pid, self._name)
AccessDenied: psutil.AccessDenied (pid=1182)
2020-10-10 01:36:40,275 - octolapse.script - ERROR - Error output (stderr) for 'RP - Snapshot Camera Script':
	ERROR: Could not get image.
2020-10-10 01:37:00,738 - octolapse.script - ERROR - Error output (stderr) for 'RP - Snapshot Camera Script':
	ERROR: Could not get image.

#### What were you doing when the problem occurred
1. I was able to complete the remote trigger in putty
2. It successfully save the image
3. The script in octoprint triggers the camera, and then I get an error "Camera Script Failed Errors were detected - The 'RP - Snapshot Camera Script' process returned errors. See plugin_octolapse.log for details"

#### What should have happened?
Captured an image

#### What happened instead?
Errored out

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi 
Os Version: 0.17.0

#### Printer model & used firmware incl. version
Printer Model: Ender 5 Pro
Printer Firmware Version: Marlin 1.3.1

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Win10

I have edited the suders correctly, looked up with gphoto2 was located, and everything is write with that. 

I don't know if it's possible that my camera isn't on the compatibility list, but it seems to be working for the most part. Any ideas on if I can get this to work? I have a Canon EOS RP.

Let me know if you need any other information.

decod3dotcom, 2020-10-18:
Anything?

FormerLurker, 2020-10-18:
Sorry I missed this issue.  Been working every spare minute on another plugin release.

This error in your log:

> The 'RP - Snapshot Camera Script' process has timed out before completing. Attempting to kill the process.

suggests the script timed out.  Try increasing the timeout in your camera profile here:

[image]

However, the default timeout is pretty large.  You may want to ensure your camera is only producing .jpg images and not both jpg and raw, and that your resolution/compression levels on your camera are reasonable.  The longer your script takes to complete, the larger the quality impact.

Let me know if that helps, and again, sorry for the delay!

decod3dotcom, 2020-10-19:
That makes sense. So the in-camera settings affect the script? I know I have mine set to JPEG/RAW, so that's probably it. 

Already tried increased the timeout.

decod3dotcom, 2020-10-19:
Think that might've been the issue. It's running right now with zero errors. I'll let you know tomorrow, when the print finishes! Thanks for all of your help, so far!

FormerLurker, 2020-10-19:
Of course!  Looking forward to hearing how it goes.  Good luck!

decod3dotcom, 2020-10-19:
So far so good. The video rendered fine. 

I'm now working through another issue, that seems to just happen with Octolapse. I keep getting blobs from the movement made to take each snapshot. It appears this is common and due to slicer settings, so I'm tweaking those as we speak. Hoping to iron all this out so I can start adding Octolapses to my YT videos! 👍🏻

FormerLurker, 2020-10-19:
Well, the longer the snapshot takes the worse the impact.  Have you seen [this wiki entry](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Print-Quality-Issues)?  Might be worth going through it.  I'm happy to help in any way I can!

decod3dotcom, 2020-10-22:
Okay I got it working 100%!

https://youtu.be/UjlS-5DmZ8o