FormerLurker/Octolapse #607: [Request] Send custom system command before snapshot

Ruko2010, 2020-10-14:
#### If this is a feature request describe it here

It would be cool to execute a system command before the snapshot is triggered. It would be even more cool, if this command could be executed a a little time (say 1-2 seconds) before the snapshot. The Background is, that I'm mostly printing in dark areas, but I have a gcode controled light strip around my printer. it is controled via the gpio ports of my pi. Today i have to leave on the strip for the whole print to capture a timelapse, but it would be cool if the strip will be on, when photos are taken (and 2 seconds before that, because the camera has to adjust to the light when switching the strip on).

#### Version of Octolapse
 
Octolapse Version: v 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-19:
so, adding a command X seconds before would be extremely difficult, since there is no good way to predict when that would be.  The printer buffers commands, and it's impossible to tell exactly where it is without waiting for a move to complete.  I have seen other people do exactly what you are wanting with the before snapshot script.  You could add a delay if necessary (see the snapshot delay in the camera profile) to allow your camera time to adjust.  I can see if I can find a similar script.  How are you controlling the light? GPIO, http?  Something else?

Ruko2010, 2020-10-21:
I‘m controlling the led-strip with GPIO, so a before and after script would be cool.
A delay would be ok, but it will stop the printer for the delayed time, right? So there may be a risk of affecting the print.