FormerLurker/Octolapse #609: raspberry pi camera high quality

mikgangal, 2020-10-19:
#### If this is a feature request describe it here

full resolution on the new raspi camera high quality

just watched a video of somebody who used the new raspberry camera (high quality) and the results were not much better than with a webcam; seems like the guy couldn't increase the resolution past 2 MP while the new camera can go all the way to 12MP
is this a true limitation? and if so any chance it could be improved? (before i shell out the cash on the module, adapters, new lenses ettcccc..)
Thanks so much

#### Version of Octolapse
 
Octolapse Version: 0.4

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-19:
I'm going to grab one of these and see what can be done.  12MP will be difficult to achieve using mjpgstreamer, but I'll see what can be done.  I know a capture routine could be created via a custom script, but it would only work if the camera isn't streaming since exclusive access is required, so I hope that's ok.

carcamerarig, 2020-10-19:
> I'm going to grab one of these and see what can be done.  12MP will be difficult to achieve using mjpgstreamer, but I'll see what can be done.  I know a capture routine could be created via a custom [...]

Aah excellent, I've been waiting for this. Do you think you could implement the camera only service (max sensor size) and a frame rate that will reasonably transfer? Even 1fps is adequate for framing and focus right?

FormerLurker, 2020-10-19:
> Do you think you could implement the camera only service (max sensor size) and a frame rate that will reasonably transfer? Even 1fps is adequate for framing and focus right?

At 1FPS, on average 500MS will be added to each snapshot, which is a lot of time.  That would impact quality quite a bit, especially for the 'snap to print' trigger.  Is focus automatic for the HQ camera?  For my logitech 920, an extremely short delay (30-60MS depending on framerate) is adequate when using auto-focus.  I've never thought about why that is honestly.

> Do you want me to send you one to work with?

That is a very generous offer!  Let me consider that.  In the meantime, I'll try to come up with a script you can use for now.

carcamerarig, 2020-10-19:
Well, I can't write the script myself so it's a small price to pay and everyone benefits with HQ timelapse. No one else has done it apart from the external dslr script. 

What about a 2nd preview port/mode image for octolapse preview and the full res image straight to file or does just add even more time?

FormerLurker, 2020-10-19:
Well, for the preview image I create a thumbnail from the original, and that is done on a background thread.  All focus is on minimizing the snapshot acquisition time.

I am going to try to do this on my picam V2 first.  In theory, it's still capability is pretty high too, so it should be a useful test case.  Once I have the script I'll send it to you for testing.  If it doesn't work on the HQ, I will take you up on your offer.

FormerLurker, 2020-10-19:
I also wanted to mention that I'm finishing up another plugin, and will be starting on overdue Octolapse issues right after that.

carcamerarig, 2020-10-19:
Excellent news 👍🏼

carcamerarig, 2020-10-19:
> I also wanted to mention that I'm finishing up another plugin, and will be starting on overdue Octolapse issues right after that.

I that plug-in will be a game changer in terms of how smooth it'll run.

mikgangal, 2020-10-20:
> At 1FPS, on average 500MS will be added to each snapshot, which is a lot of time. That would impact quality quite a bit, especially for the 'snap to print' trigger. Is focus automatic for the HQ [...]

i believe the picam high quality is fixed focus.. just like in the original one; more over i think the aperture is also manual (at least in the videos i see on the web)

since i know nothing about coding.. i can ask a dumb question.. could you have the camera stream in really crappy resolution.. just so that we can monitor the progress etc... and cut the feed when it's time to take the shot.. for however many seconds it's needed (i understand this can have an impact on print quality.. but i find the small plastic bits fairly easy to remove).. and then restart the feed until the next shot needs to be taken?

thanks for reading my nonsense :D

carcamerarig, 2020-10-29:
> I also wanted to mention that I'm finishing up another plugin, and will be starting on overdue Octolapse issues right after that.

Did you ha e a chance to look into this?

FormerLurker, 2020-10-29:
Not yet, been hammering out aec welder features based on user request.  I promise i will get back to Octolapse soon!