FormerLurker/Octolapse #610: Flashforge Creator Pro printing in incorrect location of build plate when Octolapse enabled

DaveTSG, 2020-10-19:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0.

#### Version of OctoPrint

OctoPrint Version: 1.4.2.

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES
("Log Everything" preset was selected in the OctoLapse tab)

#### What were you doing when the problem occurred

1. Tried to print a test cube in the centre of the 3D printer build plate with Octolapse's "test mode" enabled. [This gcode file](https://www.dropbox.com/s/brtmw17lr0h31u2/xyz%20cube.gcode?dl=0).
2. Accepted the snapshot plan preview. It looked ok.
3. The extruders move around as if they are printing the cube at the front of the build plate (and at times completely off the build plate).

#### What should have happened?
The cube should've been printed in the centre of the build plate.

#### What happened instead?
The cube would have been printed at the front of the build plate.

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Flashforge Creator Pro (2016 model, non-IDEX kit, bought in July 2020 - there are slight differences with earlier models e.g. those manufactured in 2016-2019)
Printer Firmware Version: Sailfish v7.8 r0f02f

#### Browser and version of browser, operating system running browser

Browser: Google Chrome v86.0.4240.80
Browser OS: MacOS Catalina 10.15.7
(also happens on Windows via Edge browser, and various iOS devices with Chrome or Safari browser. Seems to not be a browser issue.)

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: [Here](https://www.dropbox.com/s/brtmw17lr0h31u2/xyz%20cube.gcode?dl=0). I have also tried a dozen other different files/designs. I don't think it's the GCode file itself (unless Simplify3D is screwing them all up...). That said, all these files work perfectly fine when the Octolapse plugin is disabled.

#### Link to settings.json

Link to settings.json with all passwords removed: [here](https://www.dropbox.com/s/2e99byfv76jmkzy/settings.json?dl=0).

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [here](https://www.dropbox.com/s/1upnvv69rjhpcdu/plugin_octolapse.log?dl=0).

#### Link to octoprint.log

Link to octoprint.log:  [here](https://www.dropbox.com/s/bcgbz98m7dbgsqm/octoprint.log?dl=0).

#### Link to contents of Javascript console in the browser

Link to javascript console output: not needed. Not a browser problem.

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

It appears that Octolapse thinks the centre of the build plate is somewhere else (just beyond the front of the build plate).

I tried to set the firmware settings to "require explicit G90/G91/M82/M83" as I can see those in my gcode files, but that didn't fix the issue.

I noticed there have been similar issues in the Octoprint forum, but they all seem to have been resolved by an update made to Octolapse back in 2018. I found a more recent forum post (June 2020) where someone else had the same problem. I [replied to that post ten days ago with a lot of detail](https://community.octoprint.org/t/octolapse-0-4-0-not-working-with-flashforge/20863/7?u=davetsg), but haven't received a reply, so I am posting here in case you didn't see my reply in the forum post. I'm not sure if this is a bug or a configuration issue or ...?

I found [this](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Troubleshooting-Printer-Profile-Issues#the-extruder-isnt-in-the-proper-position-in-the-snapshot) troubleshooting guide.
I manually specified the printer volume in the Octolapse printer profile - identical to what I had set in OctoPrint itself. However, when selecting the origin, I ran the following commands (as per the linked-to guide):

```
G28 X Y; Home X and Y axes
G0 X0 Y0; Move to the origin: 0,0
```
The G28 command moved the extruders to the back-right of the printer enclosure (beyond the build plate itself).
The G0 X0 Y0 command moved the extruders to the very front of the build plate, but in the centre. e.g. my print build area in the Octolapse settings is x225mm * y 145mm * z 150mm. It looks like the extruders have been moved to x=112.5mm (roughly), y=0mm (assuming y=0 at the front of the build plate)

(these aren't exact measurements but it looks to be pretty much centred by-eye so I'm assuming this x=112.5 is the case)

Here's a diagram...:

[image]

The blue X is where the extruders end up after running the G28 command.
The green X is where the extruders end up after running the G0 command.

As such, neither of the only available options "front left" or "centre" are appropriate as the Origin Type in Octolapse's printer profile/settings. Perhaps we need a new "origin type" called "front-centre"?

I'm not 100% sure if this is related to [this issue](https://community.octoprint.org/t/octolapse-0-4-0-not-working-with-flashforge/20863?u=davetsg), but if Octolapse is incorrectly setting the origin point, that may explain both our issues...

If I disable Octolapse and then re-issue the command G0 X0 Y0, the extruders end up in the back-right corner, where the blue X is in my diagram. I don't know what to make of this (this is my first 3D printer and I am very much still stumbling around in the dark).

[Here's a fun video showing what happens during a test print](https://www.dropbox.com/s/g92i0hrkvfxzor2/IMG_9703.mp4?dl=0). I am guessing that the extruder assembly is being instructed (by Octolapse) to move all the way forward, beyond the front of the build plate, such that the assembly ends up hitting the edge of the axes themselves, very close to hitting the enclosure wall. When Octolapse is disabled completely, this same gcode file prints perfectly in the centre of the build plate.

The video starts about a tenth of a second after the print starts in OctoPrint, so while it looks like the video starts when the extruders are already in motion, you haven't missed anything.

Please let me know if you need further information to help diagnose this.

Thanks in advance for your help.
David

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-19:
Thank you for such a detailed report!  That is going to really help me figure out what is going on.  Also, sorry about my lack of response on the Octoprint forum.  I took a break from Octolapse for a bit to finish up another plugin (arc welder), and stopped monitoring the forum as often as I usually do.  It's easier to track these things from Github though.  I have started to get back into things more recently, and have been working on things for the last several weekends.

So, before I have a chance to dive in, perhaps take a look at this setting in the printer profile:

[image]

If your printer reliably goes to approximately the same spot after a home, you can enter the coordinates manually.

Regarding the origin...  Geesh.  Every printer needs to do things a bit different eh?  I'm thinking this could be causing the issue depending on your settings, but I need to look at your settings more closely.  I'll try to get to this as soon as possible.

DaveTSG, 2020-10-19:
Thanks for the reply. 
I don't have anything set for the home position coordinates. I'm uncertain as to what I should put in there.

I noticed in Simplify 3D that at the start of a print, the XYZ axes are all homed, but they don't specify the coordinates. 

```
; **** Replicator 1 dual start.gcode ****
M73 P0 ; enable build progress
G162 X Y F3000 ; home XY maximum
G161 Z F1200 ; home Z minimum
G92 Z-5 ; set Z to -5
G1 Z0 ; move Z to 0
G161 Z F100 ; home Z slowly
```
Those G161/162 commands [just tell the printer to go home](https://reprap.org/wiki/G-code#G161:_Home_axes_to_minimum), but no specific coordinates are given, so I'm not sure what I should enter in for the home position coordinates in Octolapse.

Ah, I just found the printer settings in Simplify 3D and saw this:

[image]
[image]

I entered the following into Octolapse's home position settings: X 112.5 Y 72.5 Z 0.
I figure this means (based on the homing directions in the Simplify 3D screenshots) that its starting at the front-left corner, then moving 112.5mm right and 72.5mm. This would be the centre of the build plate, so I changed the origin type in Octolapse to "centre".

That said, when I click the "home" buttons in the "control" tab of Octoprint, the extruders go to the back-right corner of the printer, beyond the build plate. So I'm not sure if these settings are correct or not...

Anyway, I tried again with the same gcode, but the printer did exactly the same thing as in the OP video.

I then changed the origin type in Octolapse back to front left just to see if that'd help, but it made no difference.

Finally, I set the home position coordinates back to zero and the origin type to centre. Again, no dice.

Should I instead enter the XY coordinates of the back-right corner of the printer as the home position in Octolapse? Is there some way of finding out the exact coordinates? (or should I just get a measuring tape and measure the coordinates of where the extruders are when I press the home button in Octoprint - in the back-right corner?)

Thanks again for your help. :)

FormerLurker, 2020-10-19:
Thanks for trying.  I will dive into this soon.  The noise your printer made as it tried to move past the endstops will haunt my dreams until this is resolved...

DaveTSG, 2020-10-20:
Haha, it definitely didn't sound healthy!

DaveTSG, 2020-10-26:
By the way, I've just backed you on Patreon. Looks like a great plugin and you clearly have put in a great deal of effort. :)

FormerLurker, 2020-10-26:
I really appreciate that @DaveTSG!  I've got a few more things to deal with regarding my new plugin (arc welder), and will be getting back to the Octolapse issues soon.

zymbiosis, 2021-01-17:
@DaveTSG I also have a FFCP and do not appear to have this issue (though I'm still working through other tweaks with octolapse).  It has been a while but I slightly remember having to make changes to octoprint for the origins to work correctly.  In your Octoprint Settings -> Printer Profiles -> Print Bed & Volume settings, do your settings match mine?

[image]

Also, slight side note, you shouldn't have to have the toolhead offset configured in S3D for Tool 1, the printer firmware handles the offset during tool changes.  Maybe it's working for you but just wanted to point that out as well.

DaveTSG, 2021-01-17:
Hi @zymbiosis. Thanks for all that. However, I have since replaced the FFCP with a Prusa i3 Mk3S (there were many problems with the FFCP, and FF refused to even respond to my emails for technical support.... ditto the retailer I bought it from... it ended its short life [in this manner](https://youtu.be/GPVwK-MpiSw)... except with less baseball bat and more axe/block splitter).

I should've bought the Prusa from the start. That thing just bloody-well works. :-)

In any case, I no longer have any of the Octoprint/Octolapse/S3D settings I had for the FFCP so can't comment on your suggestions. Thank you though for your thoughts.

FormerLurker, 2021-01-18:
@DaveTSG, I think @zymbiosis, suggestion would have probably worked for you.  Since you now have an Mk3, can I close this issue?

DaveTSG, 2021-01-18:
> can I close this issue?

Yep, thank you.

closed 2021-01-18 by DaveTSG

Palyzoa, 2022-02-27:
Hi

Am i able to re-open this please.

I have octolapse running on mmy FFCP though I cannot get it to take snapshots from the rear right.

Rear right is the only mode that doesn't try to fly off the front of the build plate. though I have to set fixed 

I have tried multiple custom home points which don't seam to take effect. 

Thanks for any help. 

[image]

[image]

[image]

[image]