FormerLurker/Octolapse #611: Raspberry Pi HQ Camera timeouts after a few shots

starlord1980, 2020-10-24:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: Octolapse 0.4.0

#### Version of OctoPrint

OctoPrint Version: OctoPrint 1.4.2 (Python 3.7.3 )

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: Yes

#### What were you doing when the problem occurred

Print started, Pi HQ camera snapped a few shots but then received an error:
Camera Error: An timeout occurred downloading a snapshot for the Default Webcam camera in 5.000 seconds.

#### What should have happened?
All snapshots taken successfully.

#### What happened instead?
Webcam stream then went blank and only comes back when you reboot the Pi

For example:
Snapshot Count:17
Failed Snapshots:270

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi 1.4.2 (under python3)
Os Version: pi@octopi:~ $ uname -a
Linux octopi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

#### Printer model & used firmware incl. version

Printer Model: Ender 3
Printer Firmware Version: 2.0.7.2

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: macOS Catalina

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  will update and provide this shortly

#### Link to octoprint.log

Link to octoprint.log: will update and provide this shortly

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: will do a screencapture next and update here

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-24:
I haven't gotten hold of an HQ camera yet, so I'm not sure what is going on.  If it's taking over 5 seconds (according to the timeout), that is WAY too long.  I suspect the resolution is simply too high and/or there was some kind of lockup in mjpgstreamer.

There are other issues that refer to the HQ pi cam as well, fyi.

starlord1980, 2020-10-24:
ok, thanks.
I also tried upping the gpu_mem=256 also no effect.

Also if useful my camera settings in /boot/octopi.txt:
camera_raspi_options="-x 1920 -y 1080 -fps 10"

FormerLurker, 2020-10-24:
@starlord1980, I would need to see your entire octopi config file to verify that it is correct.  If you want to post it, please use gist.github.com to make it easier for me to use.  Also, I don't believe a resolution of 1080P will result in a full field of view.  At least that's what happened in previous cameras.  I was able to find resolutions that supported a full field of view on the raspberry pi site.

starlord1980, 2020-10-24:
Certainly..
https://gist.github.com/starlord1980/635c431b00e0939a8792d77bb65e4d89

FormerLurker, 2020-10-24:
The HQ camera doesn't appear to have very good documentation.  I could only find the following reasonable full FOV resolutions in the [official camera guide](https://magazines-attachments.raspberrypi.org/books/full_pdfs/000/000/036/original/Camera-Guide.pdf):

1012 × 760
2028 × 1520

Hopefully that gets updated too.  I've also seen that there is a linux v4l driver, which should allow you to add the driver and run the camera like a USB cam, which may help.  If you can find the official driver for the HQ camera, I can show you how to install it.  If it's the same as the other pi cameras, I have a video and guide on the wiki that shows how to do this.

FormerLurker, 2020-10-24:
OH, and your config file looks correct..  Are you able to grab an image via http?  Something like this should work:

```
http://{IP_OF_PI_GOES_HERE}:8080/?action=snapshot
```
or
```
http://{IP_OF_PI_GOES_HERE}/webcam?action=snapshot
```
If that takes over 5 seconds, something is wrong.  It should take a few 100 milliseconds max.

Also, does the 'Test' button work in octolapse:

[image]

starlord1980, 2020-10-24:
I have a print running at the moment and the camera is no longer responding, it will work again once I reboot the Pi. At the moment this is what I get from ./.octoprint/logs/plugin_octolapse.log:

`2020-10-24 21:48:44,578 - octolapse.camera - ERROR - Unable to connect to the provided snapshot URL of http://127.0.0.1:8080/?action=snapshot for the 'Default Webcam' camera profile.
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 677, in urlopen
    chunked=chunked,
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 426, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/home/pi/oprint/lib/python3.7/site-packages/urllib3/connectionpool.py", line 421, in _make_request
    httplib_response = conn.getresponse()
  File "/home/pi/oprint/lib/python3.7/site-packages/raven/breadcrumbs.py", line 390, in getresponse
    rv = real_getresponse(self, *args, **kwargs)
  File "/usr/lib/python3.7/http/client.py", line 1336, in getresponse
    response.begin()
  File "/usr/lib/python3.7/http/client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.7/http/client.py", line 275, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
`

FormerLurker, 2020-10-24:
Hmm.  The camera not responding is not a good sign.  It may mean that your Octoprint settings are off, or there is an issue with the camera setup.   When your next print completes, perhaps switch the snapshot url to:

```
http://127.0.0.1/webcam?action=snapshot
```
and see if that helps.

You might also consider updating your raspberry pi.  See [this url](https://www.raspberrypi.org/documentation/raspbian/updating.md) for details and instructions.

starlord1980, 2020-10-24:
All the updates have been done, I also tried rpi-update but that didnt help.

I've rebooted the Pi now and have picture again. I've changed the camera resolution to default, so now its just:
camera_raspi_options="-fps 10"
Test button is successful.

This print has 175 layers so I'll leave it overnight and see if its still capturing tomorrow.
I'm logging this in debug mode as well so i'll post the completed log.

FormerLurker, 2020-10-24:
Thank you for the update.  I may have (fingers crossed) just released a new plugin depending on how the pull request goes, so I should have much more time to start tacking some Octolapse issues in the next couple weeks.  I am looking into getting an HQ camera, so hopefully I can solve your issue, as well as other HQ camera issues.

starlord1980, 2020-10-25:
Morning!
Well that Octolapse was successful, picture looks nice, but the only downside was it captured at 640x480.

I think I'll try those resolutions you mentioned and work my way up. I'll start with this and let you know what happens.
camera_raspi_options="-x 1012 -y 760 -fps 10"

starlord1980, 2020-10-25:
Well, seems that didnt work, the camera feed stops immediately after starting the print, I've uploaded the logs from that print here.

https://gist.github.com/starlord1980/3498837ef2cfe97184eb22fd78b5d813

FormerLurker, 2020-10-25:
Did you try switching

```http://127.0.0.1:8080/?action=snapshot```

to

```http://127.0.0.1/webcam?action=snapshot```

??  Check in the Octolapse camera profile and change the base address from ```http://127.0.0.1:8080``` to ```http://127.0.0.1/webcam```.  You may also test that out in the OctoPrint webcam settings.  Otherwise there must be some reason the HQ cam is dying.  I can't see how it could be related to Octolapse.  Maybe the ribbon is not firmly attached, or there is not enough power to run it?

starlord1980, 2020-10-25:
Tried that, I got "Could not retrieve snapshot URL, please double check the URL"
At the moment its being powered by a POE HAT, and have tried another Pi 4 board entirely to the same problem.
I do have an official rPi 5.1v 3a PSU I'll try next and a spare ribbon cable so I'll try swapping those out next.

FormerLurker, 2020-10-25:
Ok, it was worth a shot.  Am interested to know if POE is responsible. If a regular PSU works, let me know.  Another option would be to setup a new pi that only runs a camera server, if you happen to have a spare pi around.  I have considered that since I run 3 cameras (pi3).

starlord1980, 2020-10-26:
Thanks for your help on this...
I've set up a new overnight print, swapped ribbon, now powered from rPi PSU and I also just ran a sudo apt full-upgrade and got a new kernel version:
pi@octopi:~ $ uname -a
Linux octopi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l GNU/Linux

If it works now I wont know which of those three things solved it but I'll swap stuff around until I figure it out.

starlord1980, 2020-10-28:
Starting to think it was the ribbon cable all along, I'm having consistently good lapses now, current settings are:
`camera_raspi_options="-x 2028 -y 1520 -fps 10"
`
Picture has gone blank, but only once so far, it seems to be behaving a lot better.
I've put the POE HAT back on and so far its ok.

FormerLurker, 2020-10-28:
Oh great!  Those cables are touchy and tricky to insert all the way.

nibblerrick, 2020-11-09:
I didn't want to create an extra issue for that, it relates a little bit to this: I tried the camera, and with "raspicam"-setting in octoprint.txt it works. Then I tried to do the steps [from the wiki](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Configuring-a-Raspberry-Pi-Camera) and that didn't seem to work. I had a /dev/video0-device and could list possible resolutions and webcamd seemed to start fine.
Tried that with octopi 0.17 and 0.18.
Maybe somebody else can test this or is this just an unsupported case?

FormerLurker, 2020-11-09:
@nibblerrick, I think it may be a driver issue.  The hq cam may not be fully backwards compatible.  I will watch for an updated driver, and will try it out.