FormerLurker/Octolapse #612: Focus resetting after reboot on Logitech C925e

biga888, 2020-10-26:
#### Version of Octolapse
Octolapse Version: 0.4.0

#### Version of OctoPrint
OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. set auto focus off
2. set focus to 25
3. reboot

#### What should have happened?
the focus should retain.

#### What happened instead?
the focus did not retain

#### Operating System running OctoPrint and Octolapse
OS Name: OctoPi
Os Version: 0.17.0

#### Printer model & used firmware incl. version
Printer Model: Ender 5
Printer Firmware Version: Marlin 2.0

#### Browser and version of browser, operating system running browser
Browser: Chrome
Browser OS: Window 10

I was digging around with different setting, did a fresh install and still had the issue. Then I noticed that in the log file that the focus was being set before auto focus was being disabled.  I changed the order in the setting.json file. /home/pi/.octoprint/data/octolapse/settings.json
I change the Focus Auto to Order 14 and the Focus (Absolute) to 15 and this solved my problem.

I am using a Logitech  046d:085b | Logitech | Webcam C925e

I took a quick look to see if I could change a template or something on where the order was being generated from.

FormerLurker, 2020-10-26:
That is unfortunate.  I may need to write a custom settings page for the C925e.  Hopefully it wasn't detected as a 920...  It is a bit irritating that the manual focus value isn't being saved when the camera is set to automatic focus.  

Can you see if you can grab input.json from your camera?  you should be able to access it via http like so:
```
http://{IP_OF_PI_HERE}:8080/input.json
```
or 
```
http://{IP_OF_PI_HERE}/webcam/input.json
```

I will look into it, thanks for reporting!

biga888, 2020-10-26:
here is the pastebin link

https://pastebin.com/iebDkMh2