FormerLurker/Octolapse #614: Feature Request: camera controls to match Min/Max values from "v4l2-ctl -d <deviceid> --all"

mordhau5, 2020-10-30:
#### If this is a feature request describe it here

I was testing out a few different webcams, namely the Razer Kiyo and a generic USB Live Cam and noticed the values in the Camera Controls that seemed to either have too much range that had no change or not support the full range of values. I realize these will change from camera to camera, but fortunately, the range of values for a given camera is easily found via the webcam driver control application. I could be mistaken but when loading this form it seems like a trivial task to quickly fire off a few calls to this app and parse out the min/max values of each attribute.

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-10-30:
Check http://{IP_OF_PI_HERE}:8080/input.json

That is where I get the available settings, min/max range, steps and options.

mordhau5, 2020-10-30:
> Check http://{IP_OF_PI_HERE}:8080/input.json
> 
> That is where I get the available settings, min/max range, steps and options.

I see, well it seems then like there could be a potential bug as here is the range for focus:

> {
> "name": "Focus (absolute)",
> "id": "10094858",
> "type": "1",
> "min": "1",
> "max": "1023",
> "step": "1",
> "default": "240",
> "value": "240",
> "dest": "0",
> "flags": "16",
> "group": "1"
> },

And here is how the form for this specific webcam shows:

[image]

FormerLurker, 2020-10-30:
Hmm maybe the switch from one camera to the other isn't working for some reason.  Try creating a new profile and see if that works.

mordhau5, 2020-10-30:
I only have one currently registered to the PI, the other is a raspi-HQ I'm triggering raspistill from a script.

FormerLurker, 2020-10-30:
Can you try a new profile anyways?  It could help debug since i don't have one of those cameras to test.  Also, can you post the entire input.json file?  I might be able to replicate it.

mordhau5, 2020-10-30:
Yes of course, I'm in the process of grabbing it all now.

mordhau5, 2020-10-30:
Ok interesting, re-creating the profile populated the correct settings!

Input.Json (renamed as .txt, GH wouldn't accept the original file format):
[input.txt](https://github.com/FormerLurker/Octolapse/files/5468176/input.txt)

FormerLurker, 2020-10-30:
Use gist.github.com to post.  try going to the original profile and enable/disable custom image preferences.  That should trigger a re-fetch of the json file.  If that fails, I think I know where to look.  Did you ever have another usb cam or picam plugged in?

mordhau5, 2020-10-31:
Sorry for delay. Yes, today I was swapping USB cams to see their comparative image quality. With each one I did a webcamd restart and octoprint restart.

Just tried out your recommendation to swap custom image prefs, and that did trigger the fetch and settings show up appropriately.

FormerLurker, 2020-10-31:
Ok, cool.  I will see what I can do.  In the meantime, just use a single profile per cam, and disable the one you are not using.

mordhau5, 2020-10-31:
Will do. Also, hopefully not too annoying an ask to put it here, but I was wondering if I could be a guinea pig for testing h.264/5 encoding (iirc one of them is buggy). I thought I might be able to be helpful in that way since I have a pi4 4gb and possibly enough resources to render videos with it.

FormerLurker, 2020-10-31:
Sure, both encoders are available within the rendering profile.  If it fails, you can choose new settings and rerender in the 'failed renderings' popup (a red button will appear on the tab if any rendering fails).

mordhau5, 2020-11-03:
Where would you like me to post findings about h.264 rendering? I've started testing it with my pi4.