FormerLurker/Octolapse #615: Octolapse successfully renders but then immediately deletes the result.

mankyd, 2020-10-31:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?
Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred
1. Enable Octolapse
2. Enable "Octoslack" plugin and have it upload timelapses (not sure if this matters or not)
3. Look for timelapses on Octoprint server after rendering

#### What should have happened?
Timelapses should exist and be downloadable.

#### What happened instead?
No timelapses are listed. I know it rendered successfully because the timelapse was successfully uploaded to a slack channel and can be viewed there.

#### Operating System running OctoPrint and Octolapse
OS Name: Debian/OctoPi
Os Version: OctoPi 0.17.0

#### Link to settings.json
Link to settings.json with all passwords removed: https://gist.github.com/mankyd/b0c80a581b7ee7aabc4fba418d722db0

#### Link to plugin_octolapse.log
Link to plugin_octolapse.log: https://github.com/mankyd/dump/blob/master/plugin_octolapse.log?raw=true

That's a big file. I think it contains multiple prints maybe? Anyways, I think the most interesting bits are at the very end:

```
...
2020-10-31 11:28:32,469 - octolapse.script - INFO - stderr: [libx264 @ 0x2038600] 264 - core 155 r2917 0a84d98 - H.264/MPEG-4 AVC codec - Copyleft 2003-2018 - http://www.videolan.org/x264.html - options: cabac=1 ref=3 deblock=1:0:0 analyse=0x3:0x113 me=hex subme=7 psy=1 psy_rd=1.00:0.00 mixed_ref=1 me_range=16 chroma_me=1 trellis=1 8x8dct=1 cqm=0 deadzone=21,11 fast_pskip=1 chroma_qp_offset=-2 threads=1 lookahead_threads=1 sliced_threads=0 nr=0 decimate=1 interlaced=0 bluray_compat=0 constrained_intra=0 bframes=3 b_pyramid=2 b_adapt=1 b_bias=0 direct=1 weightb=1 open_gop=0 weightp=2 keyint=250 keyint_min=25 scenecut=40 intra_refresh=0 rc_lookahead=40 rc=abr mbtree=1 bitrate=6000 ratetol=1.0 qcomp=0.60 qpmin=0 qpmax=69 qpstep=4 ip_ratio=1.40 aq=1:1.00
2020-10-31 11:28:32,470 - octolapse.script - INFO - stderr: Output #0, mp4, to '/home/pi/.octoprint/data/octolapse/tmp/octolapse_rendering_tmp/afd56f54-421d-4265-9ca6-74f9d8994e6d.tmp':
 ...
2020-10-31 11:37:48,637 - octolapse.render - DEBUG - Deleting all snapshot images at: /home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp/a8199043-31da-4baa-8d66-1747a85afa92/354def78-9eea-409a-ad23-ee966dfff4ba
2020-10-31 11:37:48,763 - octolapse.render - DEBUG - Cleaning all temporary rendering files.
2020-10-31 11:37:48,960 - octolapse.render - INFO - Sending render complete message
2020-10-31 11:37:51,756 - octolapse.render - INFO - Cleaning temporary snapshot folders at /home/pi/.octoprint/data/octolapse/tmp.
2020-10-31 11:37:51,763 - octolapse.render - INFO - Cleaning temporary snapshot folders at /home/pi/.octoprint/data/octolapse/tmp.
2020-10-31 11:37:51,765 - octolapse.render - INFO - Sending render end message
(END)
```

#### Link to octoprint.log
Link to octoprint.log:  https://gist.github.com/mankyd/dd0bb95f5979a5bfcfacd327f5878f9a

FormerLurker, 2020-10-31:
Where are you looking for the timelapse?  Are you looking here:  
[image]

Also, I dug into the OctoSlack code, and it does look like it can [delete the source file](https://github.com/fraschetti/Octoslack/blob/1119425accdf17444a679a23ea6f5d401d4b6e0b/octoprint_Octoslack/__init__.py#L5045-L5051).  You could enable the *Archive Snapshots After Rendering* option in the rendering profile so that your snapshots are saved after rendering:

[image]

That way you could disable OctoSlack uploads, render via the *Videos and Images->Saved Snapshots*, and have a local copy.  

Also, you might want to check the OctoSlack settings to see if there is an option for deleting the source video that you can disable (I can't tell for sure, but I don't see this in the code based on my cursory look).  If not, you may want to [post a feature request for OctoSlack](https://github.com/fraschetti/Octoslack/issues).

FormerLurker, 2021-01-03:
This issue has not been updated in a while.  Closing it out.  If you want it reopened, let me know.

closed 2021-01-03 by FormerLurker