FormerLurker/Octolapse #616: [Request] Filename in the overlay

Prutsium, 2020-11-02:
Octolapse version: 0.40.0

Hi all not sure if its implemented already but {file_name} or {GCODEFILENAME} dont work to be used in the Overlay settings to have the filename imprinted in the overlay.

Is there a way or a token that already does this?

FormerLurker, 2020-11-23:
Hey, that should work!  Was this related to font selection issues?  Here are all the tokens:

[image]

You can find those in the help (blue question mark next to each setting).

MAN I'm behind in Octolapse issues.  Over Christmas break I hope to have time to start getting caught up.  The fix for that annoying 'None' error from the python extension I wrote has finally allowed me to start looking at these...

Prutsium, 2020-11-24:
Hi Thanks for starting to look into this.

Not sure if it was clear what i planned to do so here's a screenshot:
[image]

FormerLurker, 2020-11-24:
OH, duh!!  Whoops..  I was looking at the filename token replacement.  You are correct in that there is not currently a replacement token for that.  I'm going to be working on adding date/time formatting options soon, so I'll add some file name tokens too.  Marking this as an enhancement.

FormerLurker, 2020-11-26:
see #591

FormerLurker, 2021-01-05:
Ok, this is finally completed.  You can install via the plugin manager from this url:

```https://github.com/FormerLurker/Octolapse/archive/4b2ce4c8a61cb704fd9cb08528a6ab72cc9a28c4.zip```

The new tokens are as follows:  ```{gcode_file}```, ```{gcode_file_name}```, and ```{gcode_file_extension}```

I've also added a few other tokens.  The documentation isn't 100% completed yet, but let me know if you see any help (blue question marks) that need enhancements.

Thanks, and enjoy!

Prutsium, 2021-01-06:
Thanks will try it !

Prutsium, 2021-01-06:
https://user-images.githubusercontent.com/42968695/103747954-b06d2900-5003-11eb-84c5-4a871c8597fc.mp4

Seems to be working! Great! and thanks again

FormerLurker, 2021-01-06:
Awesome :)  Going to close this out, thank you for your suggestion!

closed 2021-01-06 by FormerLurker