FormerLurker/Octolapse #617: add setting for camera position for snap to print

TomW1605, 2020-11-02:
#### If this is a feature request describe it here

i have recently installed a camera mounted on my z axis that goes up and down with the hot end. something about my setup has made it so that when I use the snap to print trigger the hot end is always on the same side of the print as the camera and hiding most of the print. it would be great to add a camera coordinates setting that the smart trigger uses to try and take photos when the hot end is opposite the camera and not in the way.

#### Version of Octolapse
 
Octolapse Version: v0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

FormerLurker, 2020-11-23:
Hey, I thought I responded to this already, but I never actually clicked 'comment'!  Very sorry about that!!!

So, Octolapse can already do this.  You have two options:

1. Stabilize to the closest point to the 'Stabilization' position.
2. Stabilize first to the closest point, then to the nearest previous stabilization position (the 'smooth' option).

Here is where you set the stabilization position:

[image]

That will cause Octolapse to choose the closest endpoint in your gcode to the back left of your printer.  If you want to try the 'smooth' option (you may not since there is no telling where the extruder will wander to, though it will show you this in the preview):

[image]

If you want to customize anything, edit the trigger profile, mark it as a custom profile, and look here:

[image]

Let me know if that answers your question, and again I am VERY sorry about the delay!

FormerLurker, 2021-01-03:
Going to close this out since it hasn't been updated in a while.  Please let me know if you want it reopened.

closed 2021-01-03 by FormerLurker