FormerLurker/Octolapse #619: Info Request - Developing Daemon to Capture Full Resolution RPI HQ Cam Images in < 150ms

mordhau5, 2020-11-02:
#### If this is a feature request describe it here

Hi FormerLurker,

As the title implies, I'm working on a daemon that would allow the user to easily pull images from their RPI cameras at full resolution AND high speed. There's basically no point in using the R-PI HQ camera if you're just going to squish it through the pinhole that is MJPG streamer and so I'd rather use it like a "DSLR". This was easily done by replacing the "gphoto" command with "raspistill" in the script in your DSLR guide, but I found that in order to get a decent image you have to let raspistill run for a couple seconds to get decent exposure and white-balance. Those are precious seconds that you're losing filament pressure away from the print and that's unacceptable to me and many others I'd imagine. It was enough to cause one of my prints that had several very narrow layers to fail. But even after hard-coding the various mmal values like exposure, white balance, and shutter speed to reduce processing time, I was only able to get the raspistill response time down to 1.6 seconds, still rather long. I think it's at least partly because simply starting the camera up takes time.

So I built a simple python daemon that opens a local port, and starts up Picamera with some various image capture parameters, then waits for a request from a client to tell it when to snap and where to put it. It's not perfect, but I have gotten VERY nice looking images in as little as 100ms, from the rpi terminal. This means minimal impact to print quality. As long as it's running, the daemon is waiting for a request, and Picam will have already done whatever metering it needs for a good photo prior to the request. 

Currently, the client (which is just a couple lines of code) is written in Python, so it's currently a middle-man between the shell script that Octoprint requires. This can probably be simplified to just the shell script once I learn to deal with sockets in bash. I'd like to flesh all this out a bit more before I would offer it up, but is this something you might be interested in adding to the Wiki if I got much farther with it? Integration would look something like
-pip install the picamera package (or add a new virtualenv)
-copy scripts and test from terminal
-set up daemon with your preferred camera params (picamera is well documented)
-in octolapse paste in script path to take snapshot, script to verify daemon running before print, and optionally script to close daemon after print

Thought you might find it interesting as the R-PI HQ camera is likely to gain popularity. No worries if not, it was a fun project :)

Here's a quick sample of the results I was able to achieve with literally zero extra retraction, z-hop, or acceleration/speed tuning special for octolapse: https://youtu.be/ExoDvFKGc2c

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

FormerLurker, 2020-11-02:
Cool!  I'd love to see what you come up with.  I have actually been toying around with something similar, and have received numerous requests for this.  Will post more in a week or two, after I finish with some arc welder changes.

mordhau5, 2020-11-04:
I'm still working on making it nicer to read, more robust, and with more of a walkthrough but the basics are working: https://github.com/mordhau5/PiCam-Daemon

One snag I've hit is trying to run 'octolapse-before-print.py' from within Octolapse gives me this odd "Failed to connect to bus: No such file or directory" error. I'm certain it's because it contains os.system() commands, but it's odd because I can run that script from the terminal as the same user without elevation, no problem. The only downside to this right now is that Octolapse can't hook into any sort of efforts to verify the daemon's service is running or attempt to start it.

mordhau5, 2020-11-04:
I think this might be a docker issue

carcamerarig, 2020-11-14:
How have you got on with this, I'm eagerly following your efforts.

mordhau5, 2020-11-14:
> How have you got on with this, I'm eagerly following your efforts.

I haven't touched it in a week or so, but last I had, the proof of concept works quite well. I need to do two things for it to be wiki-friendly really:

1. write up a tutorial on setting up a service the octolapse-client scripts can access
2. re-work the daemon to remain in an always-on state but also takes socket requests that turn the camera on/off

If you'd like to test it out, I might be willing to walk you through setting it up via discord since I haven't written up a read-me yet. Could probably use a guinea pig to give me a better idea of where people might get tripped up.

carcamerarig, 2020-11-15:
> I haven't touched it in a week or so, but last I had, the proof of concept works quite well. I need to do two things for it to be wiki-friendly really:
>
>
>
> 1. write up a tutorial on setting up a [...]

Sure, I'd be happy to test it out with you.

mordhau5, 2020-11-15:
@carcamerarig Are you on discord? If so you can find me on there: mordhau5#6801 I'll help get you set up.

kot0005, 2021-04-09:
> @carcamerarig Are you on discord? If so you can find me on there: mordhau5#6801 I'll help get you set up.

hey I have added you on discord, would love to get some help! thanks!

carcamerarig, 2021-04-09:
hey, sent you a request
Many Thanks

Justin Edgington.
Visit us: Car camera Rig <http://www.carcamerarig.com>
Email us: ***@***.***
Follow us at: Facebook <http://www.facebook.com/carcamerarig>
<https://t.yesware.com/tl/dad779841ffb3b81069864ab6998c13136d2450e/dcc8913a0f61ad65d65ebdfcba0ea0cc/6500a1c8ee38c6efce36ee12187ff131?ytl=http%3A%2F%2Fwww.facebook.com%2Fcarcamerarig>
Call us: +44 (0) 7973 746192

kot0005, 2021-04-14:
I have got this to work with the help of Original poster on discord. Am compiling the steps into a guide.

AubreyKilian, 2021-06-20:
I'd be keen to get a copy of this guide please @kot0005 
I'm currently using the DSLR script with raspistill replaced in to take high res images for the raspicamv2, but as @mordhau5 pointed out, this takes several seconds per snapshot to complete.  I've not had any issues with this, other than just the time it takes.
@mordhau5 - does your daemon still allow the webcam to display in the Octoprint UI?

mordhau5, 2021-06-24:
> @mordhau5 - does your daemon still allow the webcam to display in the Octoprint UI?

@AubreyKilian As a standard webcam stream visible from the Octoprint control tab, no. Because only one service can access a webcam at a time, the daemon requires exclusive access to the PiCam for Octolapse's external scripts. Theoretically you would be able to see the Octolapse previews in the Octolapse tab, which is oftentimes enough of a preview to know how your print is doing, but most recently I have not been able to get the images to show up there. When I was first developing the daemon I had previews coming in just fine, but now they do not, even though the end result renders just fine. Maybe @FormerLurker  could speak to that point.

FormerLurker, 2022-01-13:
@mordhau5, are there updated instructions for your process here, or perhaps some sample .sh scripts?  I'm in the mood to get this working, and have gotten some requests for this functionality.  I may write up a guide and put it on github with your permission.

kot0005, 2022-01-22:
> @mordhau5, are there updated instructions for your process here, or perhaps some sample .sh scripts? I'm in the mood to get this working, and have gotten some requests for this functionality. I may write up a guide and put it on github with your permission.

He said he is busy and will get back here later.