FormerLurker/Octolapse #620: Octolapse hangs at start of the print.

rizwansarwar, 2020-11-04:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: YES

#### What were you doing when the problem occurred

1. Starting a print with Octolapse enabled

#### What should have happened?
Octolapse should have processed the gcode and given a snapshot plan to click on Agree. If I run the same gcode to Marlin with same version of Octolapse, everything works fine.

#### What happened instead?
Octolapse start the gcode processing, the gcode processor dialogue disappears and printer goes into starting state. Nothing happens after that.

#### Operating System running OctoPrint and Octolapse

OS Name: Ubuntu 20.04
Os Version: 20.04

#### Printer model & used firmware incl. version

Printer Model: ANET ET4 (SKR 1.3)
Printer Firmware Version: Klipper Latest.

#### Browser and version of browser, operating system running browser

Browser: Firefox & Chrome
Browser OS: Mac OS

#### Link to the gcode file you were printing when the problem occurred

Same Gcode prints on Marlin

#### Link to settings.json
Same settings work on Marlin

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/6RcRAtad

#### Link to octoprint.log

Link to octoprint.log:  

#### Link to contents of Javascript console in the browser

#### Screenshots and/or videos of the problem:

## Please consider becoming a patron

rizwansarwar, 2020-11-05:
@FormerLurker No love for this one? Not even a comment?

FormerLurker, 2020-11-05:
Hey, It's been less than 24 hours :)  I've seen this before, and I think it has something to do with compilation.  The same thing seems to happen for MacOs as well, so I believe it is related.  The gcode conversion appears to complete normally, then the results are not received correctly after they are sent from the Python extension (C++) to python (Octoprint).

Ubuntu seems to have this issue more often than other flavors, though not always.  Might try making sure your c++ compiler is up-to-date.  It would be interesting to know what version of python you are running as well, though I'm not sure it's related.

This is one of those issues that I can't solve quickly.  I've got to recreate this issue first, and might need to run it through a debugger.

If you want to help out, you could turn on verbose debugging (see the debugging profile), clear the existing log, and print from a very small file (maybe just a single layer), else the log will be unmanageable large.  Maybe I'll spot something in there that will give me a hint as to what is going on.  The log you posted is not detailed enough for me to learn much.

rizwansarwar, 2020-11-05:
@FormerLurker thanks for getting back quickly :). And thanks for the pointer, I have just re-imaged another instance using Octopi and started a print. It has started to print indicating something wrong/weird with the installation of Octoprint/Octolapse. I will monitor the progress but it is looking like rouge installation is the cause. Thank you, I will close this one for now.

closed 2020-11-05 by rizwansarwar