FormerLurker/Octolapse #621: [Request] Custom Pre&Post Stabilization Customization

mordhau5, 2020-11-04:
#### If this is a feature request describe it here

Best I can tell, Octolapse doesn't provide an opportunity to fill in g-code to augment the pre/post stabilization process (not just pre/post snapshot). Where I see this as a missed opportunity is tuning retraction, z-hop, and movement speed. Octolapse adopts all of these intelligently from the g-code, but the way I see it a stabilization move is different from a regular travel. It's a travel with a known pause in between. 

Ex/ Perhaps 1mm of retraction and no z-hop is perfect for my printer during normal printing, but that extra second or so away from the print is just enough to yield a tiny blob on the surface and I might want to tune that out. Or maybe I'm concerned about ringing during normal travel moves so I keep acceleration down, but during stabilization, I'm less concerned.

I could see this quickly leading to novice users breaking things if given carte-blanche to write their own g-code so perhaps just a few fields to tune these features is given. Or if an un-supervised G-Code Textbox is provided, it's provided with a **strong** warning about breaking stabilization.

Let me know what you think or if I've missed the spot where this is already possible. 

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

FormerLurker, 2020-11-04:
I've been planning to do this for a long while.  I'm surprised there isn't an open issue about this already, in fact.

I already have some gcode text boxes around, so I'm not particularly worried about that.  However, Octolapse tracks so much stuff that it makes more sense to provide override settings.  I think that should probably go in the printer profile, don't you think?

So something like an 'Override Slicer Settings' check box that contains the following:

Override Retraction Feedrate
Override Retraction Length (0 to disable)
Override Deretraction Feedrate
Override Deretraction Length (0 to disable)
Override Z-Lift Height (0 to disable)
Override Z-Life Speed
Override Travel Speed

Anything else?  Not sure about acceleration (it would also take extra work and even more settings, like the original acceleration so it can be reset after the snapshot).  Imho if your acceleration is off it is just off.  Same is actually true most likely for travel in the vast majority of cases (move as fast as possible without ringing).

mordhau5, 2020-11-05:
@FormerLurker 
All of those settings look great. I'm glad you differentiated deretraction so that lost filament can potentially be measured and account for on-return. I do still think acceleration would make sense for configuration because you're not really worried about ringing during stabilization. You're worried in the sense that you don't want your machine to skip steps, and you don't want your camera to shake out of position, but ringing is unlikely to be of any concern. Especially since smart triggers will likely return the head to a place where ringing wouldn't be noticed. I'd probably run my acceleration for stabilization moves much higher. But...is it worth the work idk, that's your call. You might only shave a couple ms off.

The argument could be made for printer profile, stabilization profile. Personally, printer profile makes sense to me, but as these settings directly affect stabilization, the Stabilization profile could seem logical. I would say Camera profile since script-based cameras would absolutely require more tuning than mjpg-streamer-cams, but since multiple cameras could be used, it would get tricky prioritizing which overrides to respect in that case.

I was also surprised nobody else explicitly requested it, I searched several pages back to see if my request was a duplicate. Maybe people probably just see diminished print quality/artifacts and give up on it? I'd hope not.

mordhau5, 2020-12-05:
@FormerLurker any chance of seeing this in any builds in the near future?

FormerLurker, 2020-12-05:
Yes, very.  I just released an update that closed the most serious issues I was seeing, so I'm going to be adding some new features.  This will be one.  I've still not figured out exactly where it should go.  I keep thit it needs its own profile.

FormerLurker, 2020-12-05:
Oh, gotta finish the next arc welder release first.