FormerLurker/Octolapse #622: Bug: Snapshots Stabilization error

Prutsium, 2020-11-06:
When running a print session i get every time a Stabilization error.
The cam (Raspi cam on pi4) works fine also Octolapse shows image 

[image]

#### Version of Octolapse
 
Octolapse Version: 
Octolapse 0.4.0
Settings Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Pi4 with Raspi Cam (1.2)
2. Octopi OS (0.18)

#### What should have happened?
Working Snapshot :)

#### What happened instead?
Error popped up

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi OS
Os Version: 0.18
Octoprint 1.4.2
Python 3.7.3

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S
Printer Firmware Version: 3.9.1

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.txt](https://github.com/FormerLurker/Octolapse/files/5503444/settings.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-11:
See Issue #566

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped.  Might take a while to show up in the plugin manager.  Reopen a new issue if you continue to have problems.  Thanks!

closed 2020-12-05 by FormerLurker