FormerLurker/Octolapse #623: Bug: Stabilization error every snapshot

Prutsium, 2020-11-06:
When running a print session i get every time a Stabilization error.
The cam (Raspi cam on pi4) works fine also Octolapse shows image 

[image]

#### Version of Octolapse
 
Octolapse Version: 
Octolapse 0.4.0
Settings Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: ___REPLACE_THIS__OCTOPRINT_VERSION_GOES_HERE

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. Pi4 with Raspi Cam (1.2)
2. Octopi OS (0.18)

#### What should have happened?
Working Snapshot :)

#### What happened instead?
Error popped up

#### Operating System running OctoPrint and Octolapse

OS Name: Octopi OS
Os Version: 0.18
Octoprint 1.4.2
Python 3.7.3

#### Printer model & used firmware incl. version

Printer Model: Prusa MK3S
Printer Firmware Version: 3.9.1

#### Link to settings.json

Link to settings.json with all passwords removed: 
[settings.txt](https://github.com/FormerLurker/Octolapse/files/5503444/settings.txt)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

Prutsium, 2020-11-06:
Seems to be simular to: https://github.com/FormerLurker/Octolapse/issues/566

FormerLurker, 2020-11-11:
I believe it is.  Try the classic layer trigger and see if that works until i figure this out.

FormerLurker, 2020-11-17:
@Prutsium , can you try installing from the plugin manager using this link in the From Url... box: https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip

See if that fixes it.

Prutsium, 2020-11-17:
Will try in a bit. Printer is running right now :)

Prutsium, 2020-11-17:
Seems to be working (at least it grabbed it's first snapshot so looks promising)
[image]

Prutsium, 2020-11-17:
So went true a few layers and looks like issue is resolved.

Thanks for the great work as i love the plugin (and will make an happy donation to you)

Prutsium, 2020-11-17:
> 
> 
> So went true a few layers and looks like issue is resolved.
> 
> Thanks for the great work as i love the plugin (and will make an happy donation to you)

Donation send (Paypal)

FormerLurker, 2020-11-17:
Oh boy!  Thanks for the donation, but I really can't take credit for the fix. Another user figured it out (a cross platform c++ guru apparently).  I am so glad it worked, thank you for testing!  This has been one of those rare issues that has taken months and hours to solve, and in the end all i did was copy in code and push to github.  I did add credits to the about page for the fix.  Gonna sleep much better tonight.

Enjoy octolapse, and please post again if the need arises!

closed 2020-11-17 by FormerLurker