FormerLurker/Octolapse #624: Ultimaker 2 - Octolapse not working (Stabilisation EROOR + G21 ERROR)

TomyVertigo, 2020-11-11:
#### Octolapse v0.4.0
#### OctoPrint Version 1.4.2
[plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/5525276/plugin_octolapse.log)]

I have Ultimaker 2 and slicing with Cura 4.8.0

When I try to make octolapse I keep getting stabilisation error and G21 error as mentioned in some other posts, but did not get solution how to fix it anywhere. :(

Thank you.

FormerLurker, 2020-11-11:
I need your settings json and other info that was in the issue template.  Are you running on a raspberry pi or some other OS?  If so, which one and which version.  Are you using OctoPi?  If so, which version?

However, it does look like the same error I'm seeing in other places and is likely caused by some bad compilation.  Try the classic layer trigger and see if that works.

FormerLurker, 2020-11-11:
See Issue #566

TomyVertigo, 2020-11-11:
Here you go: 
[settings.zip](https://github.com/FormerLurker/Octolapse/files/5526037/settings.zip)

I am running RaspBerry Pi4 8GB. Yes, I am using OctoPi v0.18.

Layer triggering not working. The only thing that is working (with error, tough) is:
Stabilization: Disabled
Trigger: SMART - Snap to print

Whan can I do?

FormerLurker, 2020-11-11:
Are you running python 3 or python 2?

You could try to update your distro, then uninstall and re-install Octolapse (you don't have to delete your settings).  I suspect this issue is compiler related.

[Here is a guide for updating installed packages.](https://www.raspberrypi.org/documentation/raspbian/updating.md).  Just run the first two apt updates: ```sudo apt update``` and ```sudo apt full-upgrade```.

FYI, this appears to be happening only for some ppl using the PI4, for only some flavors of linux, and on some instances running on MacOs.  I've got a pi4 on order, so hopefully I will finally be able to replicate this.

I'll leave this issue open until I hear if the above fixed things for you, then you can head over to Issue #566 for updates.  Sound good?

TomyVertigo, 2020-11-11:
FormerLurker, I did as you suggested and it looks like the G21 error is gone, but still facing 
**Stabilization Error**
_An unexpected error was encountered while running the timelapse snapshot procedure._

Octolapse is not working.

TomyVertigo, 2020-11-11:
Well G21 Error is not gone all the way: 
**Octolapse Timelapse Stopped**
_The printer profile requires an explicit G21 command before any position altering/setting commands, including any home commands. Stopping timelapse, but continuing the print._

Octolapse settings:
Stabilization: Centered
Trigger: Classic - GCODE

FormerLurker, 2020-11-11:
Yeah, so I think you're still hitting the same issue, which is a problem communicating to the compiled python extension.  Maybe I can send an SO file you can drop into the installation folder that I know works.  I can grab one off my pi.  Will send it soon.

FormerLurker, 2020-11-11:
Hey, are you using python 2 or 3?  It makes a difference, because I need to link to a different library.

TomyVertigo, 2020-11-11:
That would be great. 

I am running python v2.7.16.

FormerLurker, 2020-11-11:
[GcodePositionProcessor.zip](https://github.com/FormerLurker/Octolapse/files/5526898/GcodePositionProcessor.zip)

put it here:  home\pi\oprint\lib\python2.7\site-packages\GcodePositionProcessor.so

Let me know if that fixes it or if your pi catches fire and explodes :)

FormerLurker, 2020-11-11:
OH, you will need to completely reboot your pi for that to take affect.

TomyVertigo, 2020-11-11:
So... I do completely fresh install of the OctoPi?

FormerLurker, 2020-11-11:
No, you just copy that file to the path (i recommend sftp, like filezilla), reboot (not just restartin octoprint) and try again

FormerLurker, 2020-11-11:
If it works, at least it will narrow down what is causing this..  if the same issue happens... Not sure

TomyVertigo, 2020-11-11:
OK. Will do that and let you know. 

I really hope this fixes all. :)

FormerLurker, 2020-11-11:
You and me both...  Been working on my new plugin, arc welder, and have been swamped with new issues since the pi 4 and hd picam came out.  Would be happy to figure out what is going on

TomyVertigo, 2020-11-11:
I have only home\pi\oprint\lib\python3.7\site-packages\ and there is none file named GcodePositionProcessor.so
Is that OK? Should I put it here?

FormerLurker, 2020-11-11:
I thought you said you were running python 2???  I'm really confused.  you shouldn't even have a 3.7 directory!

Can you search for GcodePositionProcessor.so and tell me where you find it?  FYI, something must be up with the permissions because I've not been able to find my copy via the find / command.....  As if things aren't already complicated enough :)

FormerLurker, 2020-11-11:
Ok..  find is ..  odd.  Try running this from the home folder:

find . -name GcodePositionProcessor.so -type f

That should tell you where the current file is.  Then put the file I attached in there and overwrite it.

TomyVertigo, 2020-11-12:
So... there was only file GcodePositionProcessor.cpython-37m-aarch64-linux-gnu so I replace this file. 
But,,, when I reboot my OctoPrint there was no Octolapse installed anymore...

TomyVertigo, 2020-11-12:
And now I am getting an Internal server error in Cura.

FormerLurker, 2020-11-12:
Yeah, that shouldn't have been overwritten i don't think.  Not sure what the cura error is.

FormerLurker, 2020-11-12:
Will get back to yot tomorrow after I sleep and have time to think.

TomyVertigo, 2020-11-12:
No improvement in Octolapse, still facing the same errors.
[plugin_octolapse (1).log](https://github.com/FormerLurker/Octolapse/files/5529126/plugin_octolapse.1.log)

FormerLurker, 2020-11-12:
What steps did you perform since last time??

TomyVertigo, 2020-11-12:
_there was only file GcodePositionProcessor.cpython-37m-aarch64-linux-gnu so I replace this file._ With the file you send me. I also copied an unchanged file into this folder. 

Any other suggestions?

TomyVertigo, 2020-11-12:
@FormerLurker do you think that could be related to 64bit structure of (RaspBerry Pi4) 8GB version?

FormerLurker, 2020-11-12:
Good question.  It is entirely possible.  Maybe try a 32 bit version as a test if you have a spare sd?

TomyVertigo, 2020-11-12:
I also ordered 4GB version of RaspBerry Pi4 and will be testing it for the weekend. I want to come to the ground to this problem.

FormerLurker, 2020-11-17:
@TomyVertigo, can you try installing from the plugin manager using this link in the *From Url...* box:  ```https://github.com/FormerLurker/Octolapse/archive/88aa71f0fbac306b0d9acf19dff7388a30b29b64.zip```

See if that fixes it.

TomyVertigo, 2020-11-18:
@FormerLurker this feels like a great improvement, (at least) for now G21 is gone, but full functionality is still not there. Just tried orbit mode and it is not working correctly.

FormerLurker, 2020-11-18:
I need more info to help you with that.  How exactly is it not working?

TomyVertigo, 2020-11-18:
Here you go. Some files regarding the latest Octolapse.
[OctoLapse.zip](https://github.com/FormerLurker/Octolapse/files/5560312/OctoLapse.zip)

Octolapse settings:
[image]

FormerLurker, 2020-11-18:
Are you familiar with what the 'snap to print' trigger actually does?  It doesn't allow the printhead to move off of the part or away from the tool paths at all, so it creates a time lapse that is as close as possible.  Try using a fixed position stabilization for that, like back left or something.  If you want a true orbit, try using ng the smart layer trigger.

FormerLurker, 2020-11-18:
I should have mentioned this sooner, but check in the trigger profile setting for the little blue question marks.  I have added a lot of info that should explain what each trigger does and how the settings affect them.  If i misunderstood your issue, please let me know

TomyVertigo, 2020-11-18:
> Are you familiar with what the 'snap to print' trigger actually does? It doesn't allow the printhead to move off of the part or away from the tool paths at all, so it creates a time lapse that is as [...]

OK. I left it from before when nothing else didn't work. I will try now and let you know. 

I still get stabilization errors tough.

FormerLurker, 2020-11-18:
If the stabilization errors keep occurring, I recommend creating a new issue.  I recommend running a very small print (calibration cube or something similar) and logging with the 'debug' profile until the error occurs, then you can stop the print and upload the log.  [Here are some instructions](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Reporting-An-Issue#creating-an-octolapse-log-file) for accomplishing this that should yield a clean log file.

TomyVertigo, 2020-11-18:
Run a few test prints now and no error accourd. Great job @FormerLurker 💯

FormerLurker, 2020-11-18:
Awesome!  Going to close this.  If you have any other issues, please let me know!

commit 88aa71f0 (2020-11-17): Add fixes from @tech2077 to fix faulty conversion from bool to long int.  Potentially fixes #624,#623, #622, #589, #581, #580, #579, #578, #577, #566, #560, and #545.
closed 2020-11-18 by FormerLurker