FormerLurker/Octolapse #625: [Request] pre and post print gcode

TomW1605, 2020-11-13:
#### If this is a feature request describe it here

basically what the title says. add an option to run custom gcode before and after a print that is being recorded with octolapse. this would be separate from octoprint's own pre/post gcode and would only run if the print is being recorded. this would allow for gcode that prepares the printer for recording (eg turns on lights).

this is different to the other requests i have seen that centre around turning the light on/off immediately before/after a snapshot like a camera flash. in my case my printer is in my bedroom and I can sleep with the noise and light but not if the light keeps flashing so i want it on the whole print but only turn on automatically if it is being recorded.

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: ___REPLACE_THIS__YES_OR_NO

#### What were you doing when the problem occurred

1. ___REPLACE_THIS__STEP_ONE_GOES_HERE
2. ___REPLACE_THIS__STEP_TWO_GOES_HERE
3. ___REPLACE_THIS__STEP_...

#### What should have happened?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### What happened instead?
___REPLACE_THIS__PUT_YOUR_DESCRIPTION_HERE

#### Operating System running OctoPrint and Octolapse

OS Name: ___REPLACE_THIS__OS_NAME_GOES_HERE
Os Version: ___REPLACE_THIS__OS_VERSION_GOES_HERE

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: ___REPLACE_THIS__BROWSER_VERSION_GOES_HERE
Browser OS: ___REPLACE_THIS__BROWSER_OS_GOES_HERE

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  LINK_GOES_HERE

#### Link to octoprint.log

Link to octoprint.log:  ___REPLACE_THIS__LINK_GOES_HERE

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-13:
I will consider this.  In the meanwhile, you could create a before print start script to handle this:

[image]

Not super easy to send gcode that way, but it's possible.  You could easily turn on a light via a raspberry pi pin though.

TomW1605, 2020-11-13:
I currently have the light setup with the enclosure plugin so I can toggle it via a push button on a GPIO pin. it should be posable to have another input pin directly connected to an output pin to achieve the same effect with a script that toggles the output pin. that is definitely doable, however, "octolapse->script->GPIO out->GPIO in->enclosure plugin->gcode" is quite a long control chain with lots to go wrong so direct gcode control would be preferred

FormerLurker, 2020-11-13:
You could script a call to the api command to send the gcode.  Not saying it is a great way to do it, buy it is a reasonable workaround until this is built in.

You could also send the gcode to turn on the light before the snapshot and one after the snapshot to turn it off.  That is possible with the current interface, but i haven't tried it myself.

TomW1605, 2020-11-13:
i will try the api method for now. as mentioned in OP the printer is in my bedroom so i don't want the light flashing all night so the "led flash" method is not an option.

FormerLurker, 2020-11-13:
Lol, whoops.  Forgot about OP in my last response.  The before/after thing won't work for you.

The script needs to be directly executed, so a bash or batch script.  You can call a python script from that, but you should be able to send the request without python if you want

TomW1605, 2020-11-13:
thanks i will try that now.

FormerLurker, 2020-11-13:
Be sure to click on the blue help icon next to those scripts in the camera profile.  There should be (hopefully) useful info there.

TomW1605, 2020-11-13:
i have made a simple script but am running into an error. i have made a new issue for it #626