FormerLurker/Octolapse #626: OSError: [Errno 8] Exec format error on pre print script test

TomW1605, 2020-11-13:
#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: yes

#### What were you doing when the problem occurred

1. create simple script
2. make it runable
3. put the path in the camera settings
4. click test

#### What should have happened?
the script gets run

#### What happened instead?
there was an error

#### Operating System running OctoPrint and Octolapse

OS Name: OctoPi
Os Version: 0.16.0

#### Printer model & used firmware incl. version

Printer Model: ___REPLACE_THIS__PRINTER_MODEL_GOES_HERE
Printer Firmware Version: ___REPLACE_THIS__PRINTER_FIRMWARE_VERSION_GOES_HERE

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: ___REPLACE_THIS__SETTINGS_JSON_LINK_GOES_HERE

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  https://pastebin.com/KPvBEbHr

#### Link to octoprint.log

Link to octoprint.log:  https://pastebin.com/L8Zrp6BN

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

FormerLurker, 2020-11-13:
Sorry Tom, the script doesn't work like that.  It won't call arbitrary commands.  you need to create a script file on your pi, grant it execute permissions, then run that.  I don't want the security implications of allowing arbitrary command execution in Octolapse.

Check out these sample scripts:  https://github.com/FormerLurker/Octolapse/tree/master/octoprint_octolapse/data/scripts/linux

TomW1605, 2020-11-13:
ignore the first line of the log. that was just me being hopeful i did not expect it to work. the actual error is at the end of the log.

TomW1605, 2020-11-13:
ok i forgot to add `#!/bin/sh` to the beginning of the file

unfortunately now it is throwing another error but the script runs. i think it is timing out but still running?

FormerLurker, 2020-11-13:
Does the script run from the console?

TomW1605, 2020-11-13:
yes. and it runs from the test button as well but only after the timeout is reached (it is almost instant from the console)

FormerLurker, 2020-11-13:
can I see the script and what you've entered into the settings in the camera profile?

FormerLurker, 2020-11-13:
You didn't include the script setting in the snap of your camera settings.

TomW1605, 2020-11-13:
sorry took the screenshot but forgot to add it
[image]

FormerLurker, 2020-11-13:
You are redirecting the output of the curl operation:

> /dev/null

Try pulling that out.  Octolapse needs access to stdout and stderr

TomW1605, 2020-11-13:
i have removed that and the silent flag from the curl. same error

FormerLurker, 2020-11-13:
Ok, I'll have to take a look at it when I'm back in front of my PC.  Very strange though.

TomW1605, 2020-11-14:
ok let me know if you need more information

TomW1605, 2020-11-24:
no rush. i recently destroyed my camera light (4V led in a mostly 12V system was bouned to go wrong (i pluged it in the wrong place)) so dont have a need for this till i get that fixed.

FormerLurker, 2020-11-29:
OK, finally figured this out.  Curl is... Odd.  It outputs status updates via stderr, which is NOT the standard way of doing business.  Fortunately it offers flags to change this behavior.  You were SO close with the -s flag.  You want to do this instead:

```curl -s -S "url goes here"```

That should fix it!

TomW1605, 2020-11-29:
thanks for finding that, non standard programs are a pain. i can't test it now because i have a ~12 hour print going but will test in the morning when it is done

FormerLurker, 2020-11-29:
Hey, no rush.  I may need an option on those scripts to ignore errors for situations like this.  I hesitate somewhat because 99% of the time an error is an error.  However, what is up with curl, lol!

TomW1605, 2020-11-29:
you could do a grep of the script file for commands that are known to cause errors that are not actual errors and only show the ignore error option if they are present

FormerLurker, 2020-11-29:
Since Octolapse is multi-platform this is harder than it seems :)  I can detect the OS it's running on, so definitely a good idea.  I would just add these to the help file, but it seems like very few ppl look at those :)  I will think about this.

TomW1605, 2020-11-29:
ah multi platform is great but inconvenient sometimes.

FormerLurker, 2021-01-03:
Going to close this out since it seems mostly solved.  I left the other issue open and marked it as an enhancement.

closed 2021-01-03 by FormerLurker