FormerLurker/Octolapse #627: Error messages when used in an offline installation & Error Loading In Progress Renderings

ventilator, 2020-11-16:
#### If this is a feature request describe it here

___REPLACE_THIS__FEATURE_REQUEST_DESCRIPTION_GOES_HERE

#### Version of Octolapse
 
Octolapse Version: 0.4.0

#### Version of OctoPrint

OctoPrint Version: 1.4.2

#### When you ran into the problem, did you have diagnostic logging enabled?

Diagnostic Logging was Enabled: NO

#### What were you doing when the problem occurred

1. Installed Octopi
2. Updated to the recent version 1.4.2
3. Installed Octolapse via the plugin manager
3.b from this step on: The SD card was used in another raspberry Pi without Internet connection. There is no internet connection available at the productive site.
4. Rebooted
5. Error Message appears in webinterface
[image]

as well as error in log files regarding "_get_profiles_from_server"

#### What should have happened?
Plugin should show no error messages if there is an offline usage of octolapse

#### What happened instead?
Error message and no timelapses are recoreded

#### Operating System running OctoPrint and Octolapse

OS Name: octopi
Os Version: 0.17.0

#### Printer model & used firmware incl. version

Printer Model: Prusa Mini
Printer Firmware Version: recent as of 2020-11-16

#### Browser and version of browser, operating system running browser

Browser: Chrome
Browser OS: Windows 10

#### Link to the gcode file you were printing when the problem occurred

Link to Gcode File: ___REPLACE_THIS__GCODE_FILE_LINK_GOES_HERE

#### Link to settings.json

Link to settings.json with all passwords removed: attached
[settings.json.log](https://github.com/FormerLurker/Octolapse/files/5548270/settings.json.log)

#### Link to plugin_octolapse.log

Link to plugin_octolapse.log:  [plugin_octolapse.log](https://github.com/FormerLurker/Octolapse/files/5548276/plugin_octolapse.log)

#### Link to octoprint.log

Link to octoprint.log:  attached
[octoprint (2).log](https://github.com/FormerLurker/Octolapse/files/5548286/octoprint.2.log)

#### Link to contents of Javascript console in the browser

Link to javascript console output: ___REPLACE_THIS__LINK_GOES_HERE

#### Screenshots and/or videos of the problem:

Screenshot/Video Links: ___REPLACE_THIS__LINKs_GO_HERE

## Please consider becoming a patron
If you like this project, [please support my work by becoming a patron,](https://www.patreon.com/bePatron?u=9588101) and consider adding a 'star' to the repository.  It takes a lot of time and effort to maintain the project and respond to issues.  The cost of test prints, software, cameras, printer parts, etc. can quickly add up, so every bit helps.

You can find various [videos and tutorials](https://www.youtube.com/channel/UCnnTLQetNw7FqfQOLJCMqoQ) by subscribing to my Youtube channel.  You can also [follow me on Twitter](https://twitter.com/FormerLurker1).

FormerLurker, 2020-11-17:
You are correct.  I will fix this, thanks for reporting!  In looking at this I see that the next version of Arc Welder would have also had this problem (it loads settings from a github repository too), so I'm very glad you mentioned this!

FormerLurker, 2020-11-23:
I see you had 'Automatic Updates' disabled and am wondering what is going on there.  I don't see any errors in the log, though I do see messages from OctoPrint about being offline.  Not saying you didn't have some kind of problem there, I just don't see any evidence.

However, regarding the popup notification, the problem there is that you have no ffmpeg path entered here or Octolapse can't read it for whatever reason:

[image]

This killed the rendering processor thread (though it definitely should not die, but rather just wait for an update to the ffmpeg diretory, will look into that).  See if entering the ffmpeg folder and rebooting fixes this, because Octolapse cannot render without it.

FormerLurker, 2020-11-23:
I added the enhancement tag by mistake earlier.  whoops.

ventilator, 2020-11-27:
Strange, if I look into my config, there is a ffmpeg path configured, it comes back with a valid test and also can be called from command line

[image]

Rebooted a couple of times in between; but anyway thanks for investing your time to look into into it

FormerLurker, 2020-11-27:
Hmm, that is very strange.  I will think on this and will get back to you

ventilator, 2020-11-27:
I tested this right now: I have two raspberry pis here:
One has internet access via Wifi, the other does not, because the printer is at a location where no internet is available (but wired LAN connection in a protected net work)

If I move the SD card to the online RPi, Octolapse loads and I can e.g. render the timelapses from earlier printing sessions.

But once I move to the offline location the webinterface brings the above error message referring to the internal error and three exceptions in the log file:

`2020-11-27 16:17:40,511 - octoprint.plugin - ERROR - Error while calling plugin octolapse
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 230, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 2358, in on_startup
    raise e
ReadTimeout: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Read timed out. (read timeout=10.0)`

`2020-11-27 16:20:32,509 - octoprint - ERROR - Exception on /plugin/octolapse/loadFailedRenderings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1296, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1315, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1715, in load_failed_renderings_request
    failed_jobs = self._rendering_processor.get_failed()
AttributeError: 'NoneType' object has no attribute 'get_failed'
2020-11-27 16:20:32,520 - tornado.access - ERROR - 500 POST /plugin/octolapse/loadFailedRenderings (::ffff:10.169.131.94) 67.74ms
2020-11-27 16:20:32,556 - octoprint - ERROR - Exception on /plugin/octolapse/loadInProcessRenderings [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1296, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1315, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login/utils.py", line 272, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/__init__.py", line 1727, in load_in_process_renderings_request
    in_process_jobs = self._rendering_processor.get_in_process()
AttributeError: 'NoneType' object has no attribute 'get_in_process'`

FormerLurker, 2020-11-27:
Ok, i think that nails the issue.  This is also killing the UI in some cases.  There are two other very similar issues all having to do with an offline instance (something I forgot to check).  Will take a look this weekend for sure.  Keep an eye out for a devel rc dropping soon.

ventilator, 2020-11-27:
Nice! Just ask if you need more log files :-)

FormerLurker, 2020-11-29:
OK, after wading through all the issues I received over the last week, I have finally gotten to this.  The error you are seeing is quite unexpected (You got a read timeout but not a connection timeout!  I would NOT have expected your pi to connect then not be able to read).  I have handled the exception, and the update process should (hopefully) exit gracefully.  I believe the rendering processor issues (unable to fetch failed renderings from server, and thus really unable to do anything because the rendering processor was not started) is because the 'check for updates' call crashed the initialization routine that created the rendering processor thread.

I've tested this by killing my internet connection, and though I do not receive a read timeout (I get a connection timeout), that is now handled too.  Since your machine is behaving differently than mine, it is possible some other exception is thrown, but just post any errors in the plugin_octolapse.log file in here and I'll handle those too.

You can apply the fix by installing via the plugin manager from this link:  ```https://github.com/FormerLurker/Octolapse/archive/8f552307aa3e710b374b58fc7caf7c577c877e5d.zip```

Please let me know how it goes!  I have added a few other fixes in preparation for OctoPrint v1.5, so if you see anything off in the UI (a massive number of files changed), please post those here too and I'll fix them.

Thanks!

FormerLurker, 2020-12-05:
This should be fixed in V0.4.1, which just dropped. Might take a while to show up in the plugin manager. Reopen a new issue if you continue to have problems. Thanks!

closed 2020-12-05 by FormerLurker

ventilator, 2020-12-09:
Seems to work.
Now the plugin detects the offline condition and uses a cached version of the settings file.

`
2020-12-09 15:20:55,456 - octolapse.__init__ - INFO - Started logging to file.
2020-12-09 15:20:55,518 - octolapse.settings - INFO - Loading existing settings file from: /home/pi/.octoprint/data/octolapse/settings.json.
2020-12-09 15:20:55,719 - octolapse.settings - INFO - Creating settings from iterable.
2020-12-09 15:20:56,582 - octolapse.settings - INFO - Settings created from iterable.
2020-12-09 15:20:56,699 - octolapse.settings - INFO - Settings file loaded.
2020-12-09 15:20:57,602 - octolapse.__init__ - INFO - Setting octolapse camera address to http://127.0.0.1:8080/.
2020-12-09 15:20:57,602 - octolapse.__init__ - INFO - Setting octolapse camera snapshot template to {{camera_address}}?action=snapshot.
2020-12-09 15:20:57,603 - octolapse.__init__ - INFO - Setting octolapse defalt streaming url to /webcam/?action=stream.
2020-12-09 15:20:57,608 - octolapse.__init__ - INFO - Setting the default octolapse rendering bitrate to the Octoprint default of 10000k.
2020-12-09 15:20:57,625 - octolapse.settings - INFO - Saving settings to: /home/pi/.octoprint/data/octolapse/settings.json.
2020-12-09 15:20:57,714 - octolapse.settings - INFO - Settings saved.
2020-12-09 15:20:57,724 - octolapse.camera - INFO - Applying all webcam image settings to the Webcam - Default OctoPi 0.16.0 camera.
2020-12-09 15:21:19,084 - octolapse.settings_external - ERROR - Could not retrieve the settings file from the server!  Using cached version.
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings_external.py", line 43, in get_available_profiles
    profiles_dict = ExternalSettings._get_profiles_from_server(current_octolapse_version)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings_external.py", line 83, in _get_profiles_from_server
    settings_version = ExternalSettings._get_best_settings_version(current_octolapse_version)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings_external.py", line 217, in _get_best_settings_version
    versions = ExternalSettings._get_versions()["versions"]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octolapse/settings_external.py", line 260, in _get_versions
    raise ExternalSettingsError('profiles-retrieval-error', message, cause=e)
ExternalSettingsError: profiles-retrieval-error: An error occurred while retrieving profiles from the server. - Inner Exception: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Read timed out. (read timeout=10.0)
2020-12-09 15:21:19,105 - octolapse.render - INFO - Cleaning temporary snapshot folders at /home/pi/.octoprint/data/octolapse/tmp.
2020-12-09 15:21:19,108 - octolapse.render - INFO - Fetching all unfinished renderings and metadata at '/home/pi/.octoprint/data/octolapse/tmp/octolapse_snapshots_tmp'.
2020-12-09 15:21:19,109 - octolapse.render - INFO - Snapshot folder cleaned.
2020-12-09 15:21:19,111 - octolapse.__init__ - INFO - Octolapse - loaded and active.
2020-12-09 15:21:30,097 - octolapse.__init__ - INFO - Octolapse - is loading template configurations.
2020-12-09 15:21:30,101 - octolapse.__init__ - INFO - Octolapse - is loading template configurations.

`

FormerLurker, 2020-12-09:
That's awesome!  I see that maybe I should switch that 'error' to a warning, or maybe even an 'info' message.  It's not REALLY an error.

Thanks for the update as well as closing the issue.  It's a pretty rare thing, and is most appreciated :)

ventilator, 2020-12-09:
Thanks for fixing it :-)