FormerLurker/Octolapse #628: Fonts for OSD not found (Pi)

Prutsium, 2020-11-21:
Octoprint version: 1.4.2
Octopi version: 0.18
Octolapse version: v0.4.1+u.88aa71f
Python version: 3.7.3

I made recently a fresh install of OctoPi and noticed some issues with fonts as Octolapse dont see any fonts anymore.
Font-Config is installed and fc-list give this result:
pi@octopi:/ $ fc-list
/usr/share/fonts/truetype/dejavu/DejaVuSerif-Bold.ttf: DejaVu Serif:style=Bold
/usr/share/fonts/truetype/dejavu/DejaVuSansMono.ttf: DejaVu Sans Mono:style=Book
/usr/share/fonts/truetype/dejavu/DejaVuSans.ttf: DejaVu Sans:style=Book
/usr/share/fonts/truetype/dejavu/DejaVuSans-Bold.ttf: DejaVu Sans:style=Bold
/usr/share/fonts/truetype/dejavu/DejaVuSansMono-Bold.ttf: DejaVu Sans Mono:style=Bold
/usr/share/fonts/truetype/dejavu/DejaVuSerif.ttf: DejaVu Serif:style=Book

The log of Octolapse reports an error for a font but i cant select or deselect any font:
2020-11-21 15:33:28,677 - octolapse.render - ERROR - An error occurred while opening the selected font
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 115, in preview_overlay
    font = ImageFont.truetype(rendering_profile.overlay_font_path, size=50)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 836, in truetype
    return freetype(font)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 833, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 194, in __init__
    font, size, index, encoding, layout_engine=layout_engine
OSError: cannot open resource
2020-11-21 15:33:55,868 - octolapse.render - ERROR - An error occurred while opening the selected font
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 115, in preview_overlay
    font = ImageFont.truetype(rendering_profile.overlay_font_path, size=50)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 836, in truetype
    return freetype(font)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 833, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 194, in __init__
    font, size, index, encoding, layout_engine=layout_engine
OSError: cannot open resource
2020-11-21 15:34:17,048 - octolapse.render - ERROR - An error occurred while opening the selected font
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 115, in preview_overlay
    font = ImageFont.truetype(rendering_profile.overlay_font_path, size=50)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 836, in truetype
    return freetype(font)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 833, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 194, in __init__
    font, size, index, encoding, layout_engine=layout_engine
OSError: cannot open resource
2020-11-21 15:55:42,957 - octolapse.render - ERROR - An error occurred while opening the selected font
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse/render.py", line 115, in preview_overlay
    font = ImageFont.truetype(rendering_profile.overlay_font_path, size=50)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 836, in truetype
    return freetype(font)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 833, in freetype
    return FreeTypeFont(font, size, index, encoding, layout_engine)
  File "/home/pi/oprint/lib/python3.7/site-packages/PIL/ImageFont.py", line 194, in __init__
    font, size, index, encoding, layout_engine=layout_engine
OSError: cannot open resource

How could i fix this?

FormerLurker, 2020-11-21:
Try installing from this URL in the plugin manager:  ```https://github.com/FormerLurker/Octolapse/archive/devel.zip```

Also, what version of python are you running?  I need to test this in both versions (2.7 and 3+)

Prutsium, 2020-11-21:
As usual printer is running now but will do when it finishes.

Python version: 3.7.3

FormerLurker, 2020-11-21:
Sounds good.  And use this url instead:  ```https://github.com/FormerLurker/Octolapse/archive/5a4b744dbf6105d426c35eb449076c9c101e9dc2.zip```

It's safer since I may be pushing more to devel.

Prutsium, 2020-11-21:
Yeah seems to fix it :)
[image]

Thanks for the quick reaction and solution.

FormerLurker, 2020-11-21:
Awesome!  If you have a python 2.7 instance, I'd love to see if it works there too.  I will eventually get around to it :)

Prutsium, 2020-11-21:
Pitty enough my both systems are Python 3 already :)

FormerLurker, 2020-11-29:
Tested in python 2.7, and is working!  Closed :)

closed 2020-11-29 by FormerLurker