FormerLurker/Octolapse #630: [Request] Overlay content from external file

MarcinOrlowski, 2020-11-24:
It'd be extremely handy if there's a possibility to use content of external file as source for overlay text. For example

    {file:/tmp/status.txt}

would overlay content of `/tmp/status.txt`. That opens endless possibilities - I can have external tools run by cron periodically storing i.e. cabinet temperature, power consumption, anything and then overlaid on my octolapse.

FormerLurker, 2020-11-26:
So, how were you thinking this would behave?  Do you just want all the text in the file to be dumped into the overlay where that tag is placed?

MarcinOrlowski, 2020-11-26:
>  Do you just want all the text in the file to be dumped into the overlay where that tag is placed?

Yes, I think that it would be simplest approach. 

BTW: I looked at the sources and noticed that current implementation would require some refactoring to allow dynamic content in the placeholder. So hile at it, what about making this more generic and allow any of the placeholder to have optional arguments (let's say `:` be the separator `key:arg`. That would allow ie. to have custom output from any placeholder i.e. `{current_time}` - I'd like to get rid of seconds - now I have to modify octolapse myself, but with the above change I just just write `{current_time|%Y-%m-%d %H:%M|}` and voila. What you think?

FormerLurker, 2020-11-26:
Yes, that is another feature request :) will do this at the same time

MarcinOrlowski, 2020-11-26:
Yet one more step f- what about `{cmd:exec}` which run given executable `exec` and capture stdout and then use it for overlay?